ros2/ros1_bridge #17: ROS 2 listener does not respond to ROS 1 talker

jacquelinekay, 2015-11-04:
As of Alpha 2, a ROS 2 listener does not appear to trigger callbacks when subscribed to a ROS 1 talker, even though the dynamic bridge registers the topic and prints out the message "Passing message from ROS 1 to ROS 2". Communication does succeed in the other direction, however (A ROS 1 listener receiving messages from a ROS 2 talker.

I used the instructions in the README for this repo to reproduce this error:

https://github.com/ros2/ros1_bridge/blob/master/README.md

dirk-thomas, 2015-11-04:
For ROS 2 did you invoke `ros1_bridge_listener` (as mentioned in the instructions) or ` listener`?

jacquelinekay, 2015-11-04:
sorry, didn't catch that was the fix!

closed 2015-11-04 by jacquelinekay