ros2/ros1_bridge #22: [dynamic_bridge] could not get topic names and types: not implemented

Karsten1987, 2016-03-30:
I am following [this](https://github.com/ros2/ros2/wiki/Linux-Install-Binary#ros-1-bridge) tutorial for launching the ros1 bridge. It terminates with a RMW error:
``` 
~/ros2_ws$ dynamic_bridge
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not get topic names and types: not implemented, at /home/karsten/ros2_ws/src/eProsima/ROS-RMW-Fast-RTPS-cpp/rmw_fastrtps_cpp/src/functions.cpp:1775
Aborted (core dumped)
```

Am I doing something wrong?

jacquelinekay, 2016-03-30:
The error is not your fault, it's because rmw_fastrtps_cpp doesn't have a complete implementation of the rmw standard yet.

If you have compiled ROS 2 with another DDS implementation, e.g. Opensplice or Connext, you can run the bridge with a different middleware, e.g. `dynamic_bridge__rmw_opensplice_cpp`, and it should work.

If that executable is not available then make sure you've followed the instructions to install Opensplice: https://github.com/ros2/ros2/wiki/Linux-Development-Setup#install-one-or-more-dds-implementations and try rebuilding your workspace.

dirk-thomas, 2016-03-30:
Closing as duplicate of eProsima/ROS-RMW-Fast-RTPS-cpp#14.

closed 2016-03-30 by dirk-thomas