ros2/ros1_bridge #26: Error on start of ros1_bridge_listener

rohbotics, 2016-07-15:
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not create subscription: type support implementation 'introspection_cpp'(0x7f59fceae669) does not match rmw implementation 'opensplice_static'(0x7f59fb535659), at /home/rohan/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_subscription.cpp:66, at /home/rohan/ros2_ws/src/ros2/rcl/rcl/src/rcl/subscription.c:76
Aborted (core dumped)

Running on xenial with ROS kinetic, and ROS2 alpha7.
I compiled ROS2 from source, and disabled fastrtps, so I only have opensplice.

Am I doing something wrong?

wjwwood, 2016-07-16:
How did you disable Fast-RTPS? Did you build with Fast-RTPS first, then disable and rebuild?

gerkey, 2016-07-16:
We saw that earlier this week. I suspect that if you run `ldd` on the executable that's throwing that error, you'll find that it's incorrectly linked against fastrtps. I don't know what's going on under the hood, but it seems to have something to do with our over-linking strategy. You can try to work around it by invoking the executable that's explicitly named for one rmw implementation (e.g., instead of running `foo`, run `foo__rmw_opensplice_cpp`).

rohbotics, 2016-07-16:
I put an AMENT_IGNORE in the ePromisa folder, deleted build and install,
then rebuilt everything.

I tried running both the opensplice and the default(also opensplice because
fastrtps is disabled).

dirk-thomas, 2016-07-17:
There is no difference in the libraries / executables with and without the rmw implementation suffix. Both are built and linked the same way (with overlinking) so they will have the same problem.

Since the problem happens with only a single rmw implementation in the workspace the reason must be that a target is linked against message packages before the client library or rmw implementation. This is the same problem as ros2/ros2#253 ros2/examples#115

PR #27 (2016-07-18): fix link order
Fixes #26. Connect to #26.

Please review.
closed 2016-07-18 by dirk-thomas