ros2/ros1_bridge #28: Problem with custom messages

haueck, 2016-07-18:
Hi,

I have three types of custom messages and I am trying send them between ROS1 and ROS2 nodes. However, I keep on getting those errors:
```
failed to create 1to2 bridge for topic 'robot_robot_alarm' with ROS 1 type 'messages/RobotAlarm' and ROS 2 type 'messages/RobotAlarm': No template specialization for the pair
failed to create 1to2 bridge for topic 'robot_robot_control' with ROS 1 type 'messages/RobotControl' and ROS 2 type 'messages/RobotControl': No template specialization for the pair
failed to create 2to1 bridge for topic 'robot_robot_sensor' with ROS 2 type 'messages/RobotSensor' and ROS 1 type '': No template specialization for the pair
```
I suspect that ros2/ros1_bridge cannot see ros1 messages. Before building ros2 I build and install ros messages like this:
```
source /opt/ros/kinetic/setup.bash
cmake -DCMAKE_INSTALL_PREFIX:PATH=/opt/ros/kinetic/ ..
make
make install
```
And all files seem to be correctly installed in /opt/ros/kinetic. What do I do incorrectly?

Messages for ros: https://github.com/piappl/ros2_benchmarking/tree/master/comm/ros1node/messages
Messages for ros2 https://github.com/piappl/ros2_benchmarking/tree/master/comm/ros2node/messages
Docker file for ros1_bridge: https://github.com/piappl/ros2_benchmarking/blob/master/docker/ros2_bridge/Dockerfile

Thanks!
Rafał

dirk-thomas, 2016-07-18:
The bridge can only relay messages which are known at build-time. From reading your Dockerfile it doesn't seem to build the ROS 2 message package before it builds the bridge.

haueck, 2016-07-18:
ROS2 messages are added in the parent Dockerfile on line 24: https://github.com/piappl/ros2_benchmarking/blob/master/docker/ros2_base/Dockerfile

dirk-thomas, 2016-07-18:
Please check the content of generated `generated_factories.cpp` file if it contains the template functions for your custom messages.

haueck, 2016-07-18:
No, it doesn't. Only standard messages.

dirk-thomas, 2016-07-18:
Then you message package was not found when the bridge is built and therefore it can't handle those types. You might want to go through the build output and check that it was built and installed correctly and then also sourced the right environment when building the bridge.

haueck, 2016-07-18:
Ok, I will double check it. Thanks for your help!

haueck, 2016-07-19:
After reading [__init__.py](https://github.com/ros2/ros1_bridge/blob/master/ros1_bridge/__init__.py) I found out that a message package name has to end with _msgs.

closed 2016-07-19 by haueck

dirk-thomas, 2016-07-19:
The `_msgs` suffix is mentioned in this document: https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst

If your package doesn't follow that naming schema you could also define some custom mapping rules as described there.