ros2/ros1_bridge #29: Bug in create_bridge_from_*_to_* functions

haueck, 2016-07-19:
Hi,

I think there might be a bug in [bridge.hpp](https://github.com/ros2/ros1_bridge/blob/master/include/ros1_bridge/bridge.hpp) in functions create_bridge_from_2_to_1 and create_bridge_from_1_to_2. The function create_bridge_from_2_to_1 is called when there is a ROS2 publisher and ROS1 subscriber (as the name suggests) and it is also called in my scenario where ROS2 node only publishes and ROS1 node only receives. However, this functions creates ROS2 subscriber and ROS1 publisher. Shouldn't it be the other way around?

I run into this problem, because I keep on getting this error, when ros1_bridge creates publishers/subscribers:
```
type support implementation 'introspection_cpp'(0x7f5103ec7669) does not match rmw implementation 'opensplice_static'(0x7f5104fca659)
```
I build ROS2 only with OpenSplice implementation so I don't understand why there is a mismatch. Do you have any ideas?

Dockerfiles:
- [ros2:base](https://github.com/piappl/ros2_benchmarking/blob/master/docker/ros2_base/Dockerfile)
- [ros2:bridge](https://github.com/piappl/ros2_benchmarking/blob/master/docker/ros2_bridge/Dockerfile)

Thanks and regards,
Rafał

dirk-thomas, 2016-07-19:
To fix the type support problem you reported in #26 you will need the patch from #27 (using alpha 7 is not sufficient).

The function `create_bridge_from_1_to_2` creates a ROS 2 publisher and a ROS 1 subscriber. The subscriber receives the messages in the ROS 1 system, converts it, and republishes it in the ROS 2 system.

The function `create_bridge_from_2_to_1` creates a ROS 1 publisher and a ROS 2 subscriber. The subscriber receives the messages in the ROS 2 system, converts it, and republishes it in the ROS 1 system.

Both directions are working fine with the tutorials as far as I can tell.

haueck, 2016-07-20:
It all makes sense, thank you for your help!

closed 2016-07-20 by haueck