ros2/ros1_bridge #30: terminate called after throwing an instance of 'std::runtime_error'

shengwen1997, 2016-09-10:
Hello,

I am now testing the ros1_bridge, I compiled the project by myself.

However, after I type the command "ros1_bridge_listener"  the following message just shows up:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not create subscription: type support implementation 'introspection_cpp'(0x7fcbc1e2d669) does not match rmw implementation 'opensplice_static'(0x7fcbc04b4659), at /home/shengwen-asus/ros2_ws/src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_subscription.cpp:66, at /home/shengwen-asus/ros2_ws/src/ros2/rcl/rcl/src/rcl/subscription.c:76
Aborted
```

Environment:

* ROS2 alpha7

* Ubuntu 16.04 (x64) 

Could you guys give me any help?

dirk-thomas, 2016-09-10:
The error message says that the type support is not the right one to be used with OpenSplice. Can you please provide more information how you built it, what rmw implementation you installed (OpenSplice I assume), and if the problem still happens if you do a clean rebuild of the workspace (delete build and install spaces and build again).

shengwen1997, 2016-09-12:
I tried the prebuild version and my own compiled version, both will report the same message.

My DDS implementation is "OpenSlpice", and I installed the implementation by typing the following command:

```
sudo apt install libopensplice64
``` 

thanks

dirk-thomas, 2016-09-14:
I can confirm the problem for the `ros1_bridge_talker` and `ros1_bridge_listener` for the packages binaries using OpenSplice.

dirk-thomas, 2016-09-14:
A local build as well as the nightly package seem to work fine for me. Can you please try the latest nightly package (http://ci.ros2.org/view/packaging/job/packaging_linux/383/) and comment if that works for you.

dirk-thomas, 2016-09-14:
Closing as duplicate of #30. The fix has landed after the alpha 7 tags. That's why the released binaries don't work.

closed 2016-09-14 by dirk-thomas

dhood, 2016-09-14:
@dirk-thomas would have meant to refer to #29

shengwen1997, 2016-09-15:
Thanks, I will try this again and report to you guys

shengwen1997, 2016-09-15:
The prebuild version work fine now, but the one I compiled from source failed.
Where can I get the newest source code?  my source code was generate by **vcs** command

dirk-thomas, 2016-09-15:
I assume you have cloned the repositories using the `ros2.repos` file. The tutorial recommends to download that file from the latest release tag. As an alternative you could get the latest version of that file from the master branch.

A simpler approach could be to only switch your already cloned repository of `ros1_bridge` to the master branch and update to the current state. I haven't tried that combination myself though.

dhood, 2016-09-15:
@shengwen1997 You may find https://github.com/ros2/ros2/wiki/Maintaining-a-Source-Checkout-in-Linux-or-OS-X#development-branches useful for updating to the newest source code.

shengwen1997, 2016-09-17:
I git pull the latest commit at master branch and everything works perfect now, thanks for helping!