ros2/ros1_bridge #31: ros1_bridge is not stable on rasperry pi 3

shengwen1997, 2016-09-17:
Hello,

I sucessfully compile the ros1_bridge on my raspberry pi3 (I tried both cross compiling and set swap memory),

Now, the problem is the dynamic bridge can not connect to the** ros1 chatter** or **ros1 listener** very well,

the program didn't crash, but just not working, I saw nothing after launch the "ros1_bridge" (but sometimes it still working!)

Environment

* Ubuntu Mate 16.04 (raspberry pi3)
*  ROS2 alpha7

Could you give me any help, thanks!

shengwen1997, 2016-09-17:
I can privode the ros2_ws for **armhf** if it is needed

dirk-thomas, 2016-09-17:
Can you reproduce the same "unreliability" with just ROS 2 talkers and listeners (try stopping and restarting one side repeatedly)?

Which rmw implementation are you using? Which version of ROS 2 are using?

If you use FastRTPS in version Alpha 7 you might want to try updating to master since eProsima has fixed a few issues related to the discovery since the last alpha.

dirk-thomas, 2016-10-17:
I will close this for now since without the requested additional information there is nothing we can do about this. Please feel free to provide more information and the ticket can be reopened.

closed 2016-10-17 by dirk-thomas