ros2/ros1_bridge #33: support for request / reply

dirk-thomas, 2016-10-17:


haueck, 2016-10-25:
I am working on this, it should be ready by the end of this week.

codebot, 2016-10-25:
@haueck, for the benefit of others, can you please briefly describe here how you'll implement this feature?

haueck, 2016-10-26:
Here you can find a working implementation, please have a look at it: https://github.com/haueck/ros1_bridge/commit/5ee6ff4ae42c7870f4ad123e3e0fc61e00a62fdb

I wanted to use the existing logic so some functions got quite long and have many parameters, it can be refactored in the next step.

There is a problem with removing service bridges though. To find a ROS2 service I look for topics that end with "Reply" and "Request" and if there is a pair of such topics, I assume that it is a ROS2 service. But when I shutdown the service, function get_topic_names_and_types() keeps on listing those topics, even though the service does not exist. Do you know why it is like this?

adamdbrw, 2016-10-26:
I will add a couple of things that we discussed when reviewing the change:
- what should be the policy of handling undesirable circumstances in ros2 service callback function? Currently there is a std exception thrown.
- unified logging (ROS::Info etc. ?). Currently printfs as elsewhere in the code.
- The get_ros1_service_info function is kinda rough right now and hopefully we can agree on how to make it less error prone (see Rafał comment).
- Currently, services in namespaces are not supported, which looks like the next thing to do.

Let us know what you think!

haueck, 2016-10-26:
I also created those two patches while working on this issue, please have a look: https://github.com/haueck/rclcpp/commit/5445ddf2a6138e3afe6ac02d118cd111a42dde02 https://github.com/haueck/examples/commit/0819f7d6680812164a4299cb187358a8f0c6e942

dirk-thomas, 2016-10-26:
Can you please create pull requests for any changes you make in any of the repos. That will help in tracking all patches as they evolve as well as provide a place to discuss the patches.

@haueck 

* The function `get_topic_names_and_types` should only return the current topics. If removed service topics are still reported that sounds like a bug to me. Can you provide step-by-step instructions and clarify which rmw implementation you are using?
* Can you please also update the README to mentioned a simple service example.

@adamdbrw 

* Currently if a service throws an exception the client never gets a response. In the future that error case needs to be communicated to the client (as it is done in ROS 1). For now raising any kind of exception is "ok". If you want your service to be robust you have to catch any exception within the service callback and set appropriate values on the response yourself (if the response message allows that).
* ROS 2 does not yet have any logging API. So for now it will be printf only.
* Regarding the `get_ros1_service_info` function. Currently you connect to the server directly. Couldn't you just query these information from the ros master?
* We haven't settled on how to handle service names (namespaced by mode vs. global). See last question in the first comment in https://github.com/ros2/design/pull/90

dirk-thomas, 2016-10-26:
Can you please create pull requests for any changes you make in any of the repos. That will help in tracking all patches as they evolve as well as provide a place to discuss the patches.

@haueck 

* The function `get_topic_names_and_types` should only return the current topics. If removed service topics are still reported that sounds like a bug to me. Can you provide step-by-step instructions and clarify which rmw implementation you are using?
* Can you please also update the README to mentioned a simple service example.

@adamdbrw 

* Currently if a service throws an exception the client never gets a response. In the future that error case needs to be communicated to the client (as it is done in ROS 1). For now raising any kind of exception is "ok". If you want your service to be robust you have to catch any exception within the service callback and set appropriate values on the response yourself (if the response message allows that).
* ROS 2 does not yet have any logging API. So for now it will be printf only.
* Regarding the `get_ros1_service_info` function: ROS 1 doesn't declare the service types to the master so yes, you have to do the "ugly" part of talking to the server directly to retrieve that information.
* We haven't settled on how to handle service names (namespaced by mode vs. global). See last question in the first comment in https://github.com/ros2/design/pull/90

haueck, 2016-10-27:
I added instructions to README.md on how to create a bridge for example_interfaces/AddTwoInts service.

Pull requests: https://github.com/ros2/ros1_bridge/pull/34 https://github.com/ros2/rclcpp/pull/261 https://github.com/ros2/examples/pull/130

haueck, 2016-10-27:
Regarding get_topic_names_and_types(), launch add_two_ints_server and this simple program below. It keeps on listing topics even after the server is stopped.
```
add_two_intsReply: example_interfaces::srv::dds_::AddTwoInts_Response_
add_two_intsRequest: example_interfaces::srv::dds_::AddTwoInts_Request_
```

```
#include <iostream>
#include <chrono>
#include <thread>
#include "rclcpp/rclcpp.hpp"

void poll(rclcpp::node::Node::SharedPtr node)
{
    auto topics = node->get_topic_names_and_types();
    for (auto it : topics)
    {
      std::cout << it.first << ": " << it.second << std::endl;
    }
    std::this_thread::sleep_for(std::chrono::seconds(2));
}

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  auto node = rclcpp::node::Node::make_shared("service_test");
  rclcpp::executors::SingleThreadedExecutor executor;
  std::function<void()> f = std::bind(poll, node);
  auto timer = node->create_wall_timer(std::chrono::seconds(2), f);
  while (rclcpp::utilities::ok())
  {
    executor.spin_node_once(node);
  }
  return 0;
}
```

wjwwood, 2016-10-27:
This is a little off-topic, but I think the `std::this_thread::sleep_for(std::chrono::seconds(2));` does not need to be in your callback since the timer will ensure that it is called as close to every two seconds as possible.

haueck, 2016-10-28:
You're right... I forgot to delete it, first I implemented it differently.

haueck, 2016-10-28:
I will be off work for the next two weeks, Adam will take this over.

haueck, 2016-11-14:
Hi,

I'm back to work and I would like to continue working on this issue. Have you reviewed this change? Is the logic correct? Now I will create system tests to cover it.

haueck, 2016-11-14:
To create those tests I need to generate a header from a srv file for ROS. I know that I can do it using catkin's add_service_files() and generate_messages(). But ROS2 uses ament so I cannot use those macros. Is there any way to generate such header so I can build ROS service server and client and create automatic system tests?

haueck, 2016-11-14:
Or I can simply add that generated header to the project...

dirk-thomas, 2016-11-14:
In ROS 2 the package `rosidl_cmake` provides the CMake API to perform code generation for messages and services. You can look at the [test_rclcpp](https://github.com/ros2/system_tests/blob/master/test_rclcpp/CMakeLists.txt) package as an example how to use services in tests.

haueck, 2016-11-14:
Are those generated headers compatible with ROS1? I thought that rosidl_generate_interfaces() generates headers that can be only used with ROS2.

dirk-thomas, 2016-11-14:
I didn't get that you are referring to ROS 1 service files. While you call the ROS 1 message generation (using `gencpp` / `genpy`) I would suggest to just depend on a package which already provides this, e.g. `roscpp_tutorials`.

haueck, 2016-11-15:
I've changed the concept - the tests for ros1_bridge will only compile if diangostic_msgs package is available for ROS and ROS2. It is difficult to use a custom service because ros1_bridge would need to see it during compilation. Then I can use SelfTest service to test communication between ROS and ROS2.

haueck, 2016-11-22:
The fixes after the review and tests are in the new pull request https://github.com/ros2/ros1_bridge/pull/36. There are two commits so the history is preserved.

Four tests fail due two issues:
- a bug described above that makes get_topic_names_and_types() return stopped ROS2 services 
- there is no way to return an error from ROS2 service request handler. I thought that I should throw an error but this error is not caught anywhere and the application crashes. I think that this call should be in a try/catch block: https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/service.hpp#L142

I added the support for a simple manual mapping between services. Now it is possible to specify corresponding package names or package names and service names. Request and response messages need to be exactly the same though. If this feature to map fields is necessary, it can be created separately.

haueck, 2016-11-22:
To run an example from README this change in ros2/examples is required: https://github.com/ros2/examples/pull/136

haueck, 2016-12-01:
Could you please have a look on those changes :-)

dirk-thomas, 2016-12-01:
I will try to test / review / merge this PR in the next few days. Sorry that it takes so long - we are focusing on same really big things before the first beta release.

haueck, 2016-12-02:
In this case I keep my fingers crossed for the beta release!

dirk-thomas, 2016-12-14:
Finished by #48. Thank you for the contribution!

closed 2016-12-14 by dirk-thomas

haueck, 2016-12-15:
You are welcome, I hope I can contribute more in the future :)