ros2/ros1_bridge #38: Bridges to ROS 2 publishers/subscribers not removed when using Fast RTPS

dhood, 2016-12-04:
Connects to ros2/rmw_fastrtps#64 - opening this ticket here for visibility.

Bridges aren't always removed when expected (when ROS 2 nodes shut down) if `rmw_fastrtps_cpp` is used as the rmw implementation

Kukanani, 2017-06-02:
I just confirmed that this is still an issue, at least when the publishing ROS2 node is killed via Ctrl-C.

Karsten1987, 2017-06-02:
The same goes for count subscriber / count publisher btw

dirk-thomas, 2017-06-02:
@Kukanani @Karsten1987 have you both used the latest version from the default branches (including FastRTPS and rmw_fastrtps_cpp)?

Kukanani, 2017-06-02:
Updating appears to have fixed the issue on my machine.

Karsten1987, 2017-06-05:
bridge works for me. The output of "--show-introspection" shows that the ros2 node correctly disappears.

dirk-thomas, 2017-06-05:
Based on the latest reports confirming that the current state works I will close this ticket for now. Please feel free to comment here if you still experience the problem and the ticket can be reopened.

closed 2017-06-05 by dirk-thomas