ros2/ros1_bridge #43: Bidirectional bridge duplicating messages with FastRTPS

vilhjalmur89, 2016-12-06:
I've been trying out ROS 2 and I'm seeing unexpected behavior when using the bridge. 

The following image shows three messages being sent; First without any bridge, then with a FastRTPS-bridge and finally with a OpenSplice-bridge. 

The static_bridge with FastRTPS passes a message from Ros 1 to Ros 2, but also back from ROS 2 to ROS 1, even though there was no ROS 2 process running (except for the bridge).
The same scenario with OpenSplice only passes the message from ROS 1 to ROS 2, as expected. 

[image]

dhood, 2016-12-07:
Thank you for the issue report @vilhjalmur89 

I can confirm this issue happens for both `static_bridge` and `simple_bridge`. With the `simple_bridge` you get some additional logging that shows this in action:

```
dhood@osrf-esteve:~ [~]$ simple_bridge__rmw_opensplice_cpp 
I heard from ROS 1 from: [/rostopic_17524_1481068220810]
I heard from ROS 1: [static_bridge__rmw_opensplice_cpp is running]
Passing along to ROS 2: [static_bridge__rmw_opensplice_cpp is running]
^Csignal_handler(2)
dhood@osrf-esteve:~ [~]$ simple_bridge__rmw_fastrtps_cpp
I heard from ROS 1 from: [/rostopic_17588_1481068239197]
I heard from ROS 1: [static_bridge__rmw_fastrtps_cpp is running]
Passing along to ROS 2: [static_bridge__rmw_fastrtps_cpp is running]
  I heard from ROS 2: [static_bridge__rmw_fastrtps_cpp is running]
  Passing along to ROS 1: [static_bridge__rmw_fastrtps_cpp is running]
    I heard from ROS 1 from myself
```

This doesn't seem to be an issue for the `dynamic_bridge`. Hopefully you can find that useful until we get around to looking into this. If you are interested in investigating yourself, we would gladly welcome a pull request. My current suspicion is that the logic to ignore publishers/subscribers from the bridge itself that exists in the dynamic bridge (e.g. https://github.com/ros2/ros1_bridge/blob/cc1479b7b832c18d18a4a93523ea04deee54418b/src/dynamic_bridge.cpp#L380) does not seem to be implemented in the static bridge.

dirk-thomas, 2018-01-02:
This is also a problem with the `parameter_bridge`. Additional filtering based on the node name of the bridge etc. is only implemented in the `dynamic_bridge` atm.

Ekci, 2018-04-06:
Ok, so the problem is that ```ros2_callback()``` sends messages back to ROS1 without any filtering:
https://github.com/ros2/ros1_bridge/blob/d7954c901d3ff941dd00ab68d791df76763223f6/include/ros1_bridge/factory.hpp#L145-L159

```ros1_callback``` does it perfectly but I don't think that there's currently a way to get ```connection_header``` or ```caller_id``` in ROS2 ```rclcpp```. There's also no way to check by timestamp as a lot of ```msgs``` do not have ```header``` included.

I have implemented simple logic that checks if current ROS2 message equals to the last message published by ```ros1_callback``` but it's not allowing to send two exactly equal messages in a row.

https://github.com/ros2/ros1_bridge/blob/d7954c901d3ff941dd00ab68d791df76763223f6/include/ros1_bridge/factory.hpp#L104-L143

dirk-thomas, 2018-04-06:
> I have implemented simple logic that checks if current ROS2 message equals to the last message published by `ros1_callback` but it's not allowing to send two exactly equal messages in a row.

That approach wouldn't be acceptable since dropping "duplicate" messages would be changing the data flow and is therefore not desired.

> Please tell me if you have an idea how to solve it, maybe I missed something and there's a way to get some metadata about the message?

As mentioned above the `dynamic_bridge` already implements the filtering based on the node name of the source of the message (see e.g. https://github.com/ros2/ros1_bridge/blob/master/src/dynamic_bridge.cpp#L504-L507). This logic should be moved into a function which can then be reused by the different variants of the bridge.

ArkadiuszN, 2018-04-06:
The lines that you linked are filtering messages that were published on ROS1 by the bridge. ROS2 part is here: https://github.com/ros2/ros1_bridge/blob/48a373f66cdf3dc9a6f0be055374ccb096387ae5/src/dynamic_bridge.cpp#L658-L672 but to be honest I don't quite get it how to translate that to https://github.com/ros2/ros1_bridge/blob/d7954c901d3ff941dd00ab68d791df76763223f6/include/ros1_bridge/factory.hpp#L145-L159

wjwwood, 2018-04-06:
> but I don't think that there's currently a way to get connection_header or caller_id in ROS2 rclcpp.

Not sure this is exactly what you're looking for, but you can get the message info from our C api and that contains the publisher GUID. You can also get the GUID for any publishers that you have a reference to, so you could compare if messages are from a publisher in the same node/process.

This is the function that returns the message info:

https://github.com/ros2/rcl/blob/e48a445f46f1da21ff0c4bf3d52bec9120ab7b57/rcl/include/rcl/subscription.h#L254-L257

This is the message info structure:

https://github.com/ros2/rmw/blob/a82424cfc32312182a775d8607d15c7af71e170f/rmw/include/rmw/types.h#L227

These are the functions that which allow you to get a GUID for a publisher and then compare two GUID:

https://github.com/ros2/rmw/blob/a82424cfc32312182a775d8607d15c7af71e170f/rmw/include/rmw/rmw.h#L368-L376

`rclcpp` uses it to find intra process pairs:

https://github.com/ros2/rclcpp/blob/45dcd0c6e58bf54d771ae8635d30cc91f8a08e2d/rclcpp/include/rclcpp/subscription.hpp#L186-L192

And then allows you to get it in a callback:

https://github.com/ros2/rclcpp/blob/45dcd0c6e58bf54d771ae8635d30cc91f8a08e2d/rclcpp/include/rclcpp/any_subscription_callback.hpp#L154-L177

I didn't follow this issue very closely, but maybe it would help.

ArkadiuszN, 2018-04-07:
@wjwwood Good to know, thank you! I was hoping to access GUID in this callback:
https://github.com/ros2/ros1_bridge/blob/d7954c901d3ff941dd00ab68d791df76763223f6/include/ros1_bridge/factory.hpp#L146 
but after hours of untangling this code I still don't quite understand how. And it looks like I can't use  ```dispatch()``` either. Sorry :(

wjwwood, 2018-04-07:
@ArkadiuszN this callback here needs to have the message info argument as well:

https://github.com/ros2/ros1_bridge/blob/d7954c901d3ff941dd00ab68d791df76763223f6/include/ros1_bridge/factory.hpp#L95-L99

So that the `create_subscription` call right after that:

https://github.com/ros2/ros1_bridge/blob/d7954c901d3ff941dd00ab68d791df76763223f6/include/ros1_bridge/factory.hpp#L100

Knows to pass it into the callback function. You can then pass the message info from the above linked lambda into the `ros2_callback` function you referenced just now.

ArkadiuszN, 2018-04-08:
Thank you for explaining this. I have just created a pull request.

ArkadiuszNiemiec, 2018-04-12:
This issue should be now fixed by the #113.

wjwwood, 2018-04-12:
I'll close, please comment if the problem isn't fixed by #113.

closed 2018-04-12 by wjwwood