ros2/ros1_bridge #56: dynamic_bridge crashes when trying to send PointCloud2 Messages

jhoare, 2017-02-13:
I am trying to transmit PointCloud2 messages from a ROS system (currently using a kinect) using a ros1_bridge, and have a ROS2 subscriber on a different machine. 

When I try to run a ROS2 subscriber, the dynamic_bridge will crash. I've compiled ros2 from source (using fastRTPS), and I am using ros kinetic on ubuntu 16.04.

dirk-thomas, 2017-02-13:
Can you please note information. Have you built ROS 2 yourself or are you using prebuild binaries? Which version? Is there any error output? Have you tried running the bridge with `gdb` to get more information what happened?

jhoare, 2017-02-13:
``dynamic_bridge`` will crash with the following error:

```
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
Aborted
```

I ran it in GDB, and here is the stacktrace:

```
#0  0x00007ffff1979428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff197b02a in __GI_abort () at abort.c:89
#2  0x00007ffff1fb284d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff1fb06b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff1fb0701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff1fb0919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff1fd926f in std::__throw_length_error(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff2042099 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x0000000000460b1a in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::forward_iterator_tag) ()
#9  0x000000000045cc08 in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct_aux<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__false_type) ()
#10 0x00000000004579ee in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >) ()
#11 0x0000000000453b22 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, void>(__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, __gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<char> const&) ()
#12 0x000000000044d4cf in main::{lambda()#2}::operator()() const ()
#13 0x0000000000450284 in void rclcpp::timer::GenericTimer<main::{lambda()#2}, std::chrono::_V2::steady_clock, (void*)0>::execute_callback_delegate<{lambda()#2}, (std::chrono::_V2::steady_clock)0>() ()
#14 0x0000000000450133 in rclcpp::timer::GenericTimer<main::{lambda()#2}, std::chrono::_V2::steady_clock, (void*)0>::execute_callback() ()
#15 0x00007ffff7476acf in rclcpp::executor::Executor::execute_timer(std::shared_ptr<rclcpp::timer::TimerBase>) () from /home/jhoare/projects/remotework/ros2_ws/install/lib/librclcpp.so
#16 0x00007ffff74765ce in rclcpp::executor::Executor::execute_any_executable(std::shared_ptr<rclcpp::executor::AnyExecutable>) () from /home/jhoare/projects/remotework/ros2_ws/install/lib/librclcpp.so
#17 0x00007ffff747635c in rclcpp::executor::Executor::spin_once(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/jhoare/projects/remotework/ros2_ws/install/lib/librclcpp.so
#18 0x00007ffff7475db6 in rclcpp::executor::Executor::spin_node_once_nanoseconds(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/jhoare/projects/remotework/ros2_ws/install/lib/librclcpp.so
#19 0x0000000000455019 in void rclcpp::executor::Executor::spin_node_once<rclcpp::node::Node, std::ratio<1l, 1000l> >(std::shared_ptr<rclcpp::node::Node>, std::chrono::duration<long, std::ratio<1l, 1000l> >) ()
#20 0x000000000044e841 in main ()
```

I built from source following the instructions here: https://github.com/ros2/ros2/wiki/Linux-Development-Setup it looks like it has checked out the release-beta1 branch/tag on all of the repositories. 

The image example works correctly on my machine.

Karsten1987, 2017-02-16:
I get the same error when publishing laserscan msgs. Using the latest version built from source.

Karsten1987, 2017-02-16:
the bug was introduced in here: https://github.com/ros2/ros1_bridge/pull/36
I'll follow up the discussion on the referenced PR.

wjwwood, 2017-02-18:
Should be fixed by https://github.com/ros2/ros1_bridge/pull/58, @Karsten1987 tested it.

closed 2017-02-18 by wjwwood