ros2/ros1_bridge #60: segfault on dynamic_bridge

gerkey, 2017-03-10:
When following [](this tutorial), I get a reliable segfault. Here's the backtrace:

~~~
#0  0x00007fc1a073523a in eprosima::fastrtps::rtps::ReaderProxy::set_change_to_status(eprosima::fastrtps::rtps::CacheChange_t const*, eprosima::fastrtps::rtps::ChangeForReaderStatus_t) () from /home/gerkey/ros2_ws/install/lib/libfastrtps.so
[Current thread is 1 (Thread 0x7fc190ff9700 (LWP 8184))]
(gdb) bt
#0  0x00007fc1a073523a in eprosima::fastrtps::rtps::ReaderProxy::set_change_to_status(eprosima::fastrtps::rtps::CacheChange_t const*, eprosima::fastrtps::rtps::ChangeForReaderStatus_t) () from /home/gerkey/ros2_ws/install/lib/libfastrtps.so
#1  0x00007fc1a0731d25 in eprosima::fastrtps::rtps::StatefulWriter::send_any_unsent_changes() ()
   from /home/gerkey/ros2_ws/install/lib/libfastrtps.so
#2  0x00007fc1a072a888 in eprosima::fastrtps::rtps::AsyncWriterThread::run() ()
   from /home/gerkey/ros2_ws/install/lib/libfastrtps.so
#3  0x00007fc1a40fdc80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fc1aa01d70a in start_thread (arg=0x7fc190ff9700) at pthread_create.c:333
#5  0x00007fc1a3b6c82d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
~~~

I'll play with it more, including trying to get a backtrace in debug mode.

wjwwood, 2017-03-11:
I think I reported the same issue upstream in https://github.com/eProsima/Fast-RTPS/issues/78 (but it would be good to track it here as well). I just wanted to cross-reference them.

dirk-thomas, 2017-05-19:
I will close this since the upstream ticket has been resolved. Please comment if the problem still exists with the current version.

closed 2017-05-19 by dirk-thomas