ros2/ros1_bridge #61: crash in vector lookup

BrannonKing, 2017-04-11:
I get the following crash in dynamic_bridge when I shut down my sender:
```
failed to create 2to1 bridge for topic 'map' with ROS 2 type '����' and ROS 1 type 'nav_msgs/OccupancyGrid': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option
created 2to1 bridge for topic 'map' with ROS 2 type 'nav_msgs/OccupancyGrid' and ROS 1 type 'nav_msgs/OccupancyGrid'
```
OccupancyGrid works fine for some period of time before the failure. The memory for the string used as the map key is somehow getting trampled. Hence the weird characters in the type name. Note: I've been using the opensplice middleware (so that could be the trampler).

dirk-thomas, 2017-04-11:
Can you reproduce the same problem when using a different rmw implementation (e.g. FastRTPS)? We currently don't support OpenSplice so if the problem limited to that rmw implementation we might not troubleshoot the problem.

mikepurvis, 2017-04-13:
I might be getting this crash too, see:

```
Failed to read request_type from a headerFailed to read request_type from a headerFailed to read request_type from a headerterminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
```

Backtrace:

```
(gdb) bt
#0  0x00007ffff51b8418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff51ba01a in __GI_abort () at abort.c:89
#2  0x00007ffff57f184d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff57ef6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff57ef701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff57ef919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff58182bf in std::__throw_out_of_range(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x000000000040d67a in std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >::at (
    __k="name", this=0x7fffa80f43b0) at /usr/include/c++/5/bits/stl_map.h:532
#8  update_bridge (ros1_node=..., ros2_node=std::shared_ptr (count 8, weak 1) 0x741550, ros1_publishers=std::map with 391 elements = {...}, 
    ros1_subscribers=std::map with 390 elements = {...}, ros2_publishers=std::map with 0 elements, ros2_subscribers=std::map with 0 elements, 
    ros1_services=std::map with 20 elements = {...}, ros2_services=std::map with 0 elements, bridges_1to2=std::map with 0 elements, 
    bridges_2to1=std::map with 0 elements, service_bridges_1_to_2=std::map with 0 elements, service_bridges_2_to_1=std::map with 0 elements, 
    bridge_all_1to2_topics=false, bridge_all_2to1_topics=false) at /home/rosbuild/ci_scripts/ws/src/ros2/ros1_bridge/src/dynamic_bridge.cpp:288
#9  0x00000000004104bc in <lambda(const ros::TimerEvent&)>::operator()(const ros::TimerEvent &) const (__closure=0xccc9c0)
    at /home/rosbuild/ci_scripts/ws/src/ros2/ros1_bridge/src/dynamic_bridge.cpp:583
#10 0x00007ffff72244eb in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() () from /opt/ros/kinetic/lib/libroscpp.so
#11 0x00007ffff7249cf0 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/kinetic/lib/libroscpp.so
#12 0x00007ffff724b0f3 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/kinetic/lib/libroscpp.so
#13 0x00007ffff72a2af4 in ros::AsyncSpinnerImpl::threadFunc() () from /opt/ros/kinetic/lib/libroscpp.so
#14 0x00007ffff44075d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#15 0x00007ffff79be6fa in start_thread (arg=0x7fffaef9b700) at pthread_create.c:333
#16 0x00007ffff5289b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

BrannonKing, 2017-04-17:
I don't think your crash is the same thing. Mine is clearly memory corruption (and very likely related to OpenSplice.). I also often get this additional error with it: 
```
terminate called after throwing an instance of 'std::bad_alloc`
```

mikepurvis, 2017-04-17:
Okay, I'll move my report to a new issue.

dirk-thomas, 2017-05-30:
@BrannonKing I think #68 addresses the race condition which leads to the crash described in this ticket. It would be great if you could try the patch and report back if you still see the problem or if it seems to be fixed.

PR #68 (2017-05-31): fix race conditions
Before the polling lambda modified the maps while the `update_bridge` is iterating over them.

Now these changes are build up in a separate variable before the variables passed by reference to `update_bridge`. When they are swapped the global mutex is being locked.

Fixes #61.
closed 2017-05-31 by dirk-thomas