ros2/ros1_bridge #62: Expose throughput and heartbeat settings

jespersmith, 2017-04-12:
We use the ros1 to ros2 bridge to send large point clouds around over WiFi. Because of the data size, the flood of UDP packets results in dropped packets. With the default heartbeat settings, this results in quite a big delay using RELIABLE and using BEST_EFFORT we would lose a large number of packets.

Is it possible to set the maximum throughput and configure the hearbeat settings?

mikaelarguedas, 2017-04-12:
This is unfortunately not possible on a per node / per topic basis at the moment. This is on the wishlist though.
Right now these settings are using the default provided by Fast-RTPS and can be overriden [here](https://github.com/ros2/rmw_fastrtps/blob/63742a48d809ca1ae54fb93bd5c31cb35b441114/rmw_fastrtps_cpp/src/functions.cpp#L846-L852).
Note: modifying these values in rmw_fastrtps_cpp will impact all created topics.

Hope this helps

jespersmith, 2017-04-12:
That could get us working, as we do not use ROS2 directly (we are working on a Java RTPS implementation with support for ROS2 message format). Is it a possibility to make these settings user changeable after compilation so we don't have to recompile but can just set ENV variables or edit a config file?

mikaelarguedas, 2017-04-12:
Follow-up of this discussion on rmw_fastrtps https://github.com/ros2/rmw_fastrtps/issues/95

mikaelarguedas, 2017-10-16:
@jespersmith FYI since https://github.com/ros2/rmw_fastrtps/pull/153 it is possible to override vendor's default QoS settings using xml files

jespersmith, 2017-10-16:
That sounds like a good solution to us. I'm closing this issue.

closed 2017-10-16 by jespersmith