ros2/ros1_bridge #63: Crash in std::map when initially connecting

mikepurvis, 2017-04-17:
I get this crash when I run `dynamic_bridge` on our robots. I assume it's a particular node or topic that triggers it, because when I only run a subset of the system, the bridge _does_ seem to work.

```
Failed to read request_type from a headerFailed to read request_type from a headerFailed to read request_type from a headerterminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
```

Backtrace:

```
(gdb) bt
#0  0x00007ffff51b8418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff51ba01a in __GI_abort () at abort.c:89
#2  0x00007ffff57f184d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff57ef6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff57ef701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff57ef919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff58182bf in std::__throw_out_of_range(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x000000000040d67a in std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >::at (
    __k="name", this=0x7fffa80f43b0) at /usr/include/c++/5/bits/stl_map.h:532
#8  update_bridge (ros1_node=..., ros2_node=std::shared_ptr (count 8, weak 1) 0x741550, ros1_publishers=std::map with 391 elements = {...}, 
    ros1_subscribers=std::map with 390 elements = {...}, ros2_publishers=std::map with 0 elements, ros2_subscribers=std::map with 0 elements, 
    ros1_services=std::map with 20 elements = {...}, ros2_services=std::map with 0 elements, bridges_1to2=std::map with 0 elements, 
    bridges_2to1=std::map with 0 elements, service_bridges_1_to_2=std::map with 0 elements, service_bridges_2_to_1=std::map with 0 elements, 
    bridge_all_1to2_topics=false, bridge_all_2to1_topics=false) at /home/rosbuild/ci_scripts/ws/src/ros2/ros1_bridge/src/dynamic_bridge.cpp:288
#9  0x00000000004104bc in <lambda(const ros::TimerEvent&)>::operator()(const ros::TimerEvent &) const (__closure=0xccc9c0)
    at /home/rosbuild/ci_scripts/ws/src/ros2/ros1_bridge/src/dynamic_bridge.cpp:583
#10 0x00007ffff72244eb in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() () from /opt/ros/kinetic/lib/libroscpp.so
#11 0x00007ffff7249cf0 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/kinetic/lib/libroscpp.so
#12 0x00007ffff724b0f3 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/kinetic/lib/libroscpp.so
#13 0x00007ffff72a2af4 in ros::AsyncSpinnerImpl::threadFunc() () from /opt/ros/kinetic/lib/libroscpp.so
#14 0x00007ffff44075d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#15 0x00007ffff79be6fa in start_thread (arg=0x7fffaef9b700) at pthread_create.c:333
#16 0x00007ffff5289b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

This is using the implementation that is included with the Beta 1 binary download for Linux (so, bundled FreeRTPS, I believe).

dirk-thomas, 2017-05-22:
I looked into this a bit but couldn't find anything wrong. The map `ros1_services` seems to be locked correctly. Since I can't reproduce the problem I can't help much at the moment. If you can't reproduce the problem maybe you can use a modified version and add some debug information around that area. E.g. some information about the [iterator](https://github.com/ros2/ros1_bridge/blob/76010c748a58994d23da38a2b9493493e0d9b1b7/src/dynamic_bridge.cpp#L303) as well as the [package / name fields](https://github.com/ros2/ros1_bridge/blob/76010c748a58994d23da38a2b9493493e0d9b1b7/src/dynamic_bridge.cpp#L309).

A different crash has been fixed in #66 but I don't see it being related.

Regarding the error messages I have added the missing trailing newlines in76010c748a58994d23da38a2b9493493e0d9b1b7.

dirk-thomas, 2017-05-30:
@mikepurvis I just created #68 which is supposed to address #61. Since the problem was insufficient locking the result could be corrupted memory. So it might be possible that this issue is just a different incarnation of the same problem. Maybe the race condition which leads to the crash described in this ticket. It would be great if you could try the patch and report back if you still see the problem or if it seems to be fixed.

dirk-thomas, 2017-07-10:
I will close this for now assuming the referenced fix has also resolved this problem. If that is not the case please feel free to comment here with more information and the ticket can be reopened.

closed 2017-07-10 by dirk-thomas