ros2/ros1_bridge #64: too much console output

BrannonKing, 2017-04-19:
I don't think the dynamic_bridge needs to continually spew "Passing message from ROS2 to ROS1". The message would be more useful if it included the message topic and type. However, I think we would do better to aggregate those messages and output something every five seconds saying how many of each message topic/type went which direction.

wjwwood, 2017-04-19:
Yup, the console output of the bridge could be cleaned up a lot. I think we were waiting on the console logging that @dirk-thomas is currently waiting on. That way we can make use of throttled logging messages and stuff like that. So you get some feedback without an overwhelming amount.

I'd certainly take a pr in the meantime that either disables the output or even better makes an option to enable/disable it.

mikaelarguedas, 2017-08-08:
this can now be addressed given that logging macros are implemented in [rcutils](https://github.com/ros2/rcutils/blob/master/resource/logging_macros.h.em)

allenh1, 2017-08-08:
I think a `--verbose=n` option is appropriate.

mikaelarguedas, 2017-08-08:
What does `n` refer to here ? a period in seconds ?

I think a throttled debug print would be pretty good in this scenario

allenh1, 2017-08-08:
> What does n refer to here ? 

I was referring to some sort of debug level (sorry -- that was rather unclear), but I think that a throttled print would likely be better.

But, more general still, I don't know how necessary it is to say "publishing message from ROS _ to ROS _" every single time it publishes a message.

closed 2017-09-12 by dirk-thomas

dirk-thomas, 2017-09-12:
This should be addressed by #90 and #91.

PR #91 (2017-09-12): remove check for header fields request_type and response_type
This was introduced with the original implementation in #36.

A non C++ client might not even set these header fields. Since the bridge doesn't even use them anywhere this patch removes them (which gets rid of the repeated warning).

This PR together with #90 should resolve #64.