ros2/ros1_bridge #65: dynamic_bridge requires that all topic types include '::'

BrannonKing, 2017-05-18:
If you have a generated reply topic type that lacks the '::' dynamic bridge crashes (because `t.rfind("::")` returns `npos`, but this is not handled). Do we require the middleware to conform to some kind of type name standard? Where is that documented? Please see the full issue here: https://github.com/tocinc/rmw_coredx/issues/10

dirk-thomas, 2017-05-19:
> Do we require the middleware to conform to some kind of type name standard? Where is that documented?

Currently the bridge relies on the type names to match what we use in all current implementations. It is not documented explicitly - only implicitly in the `.msg` to `.idl` generator ([namespaces](https://github.com/ros2/rosidl_dds/blob/3700647737f27a2c4db6ad7641663c026fe4782e/rosidl_generator_dds_idl/resource/msg.idl.em#L46-L53), [struct](https://github.com/ros2/rosidl_dds/blob/3700647737f27a2c4db6ad7641663c026fe4782e/rosidl_generator_dds_idl/resource/msg.idl.em#L86)). All rmw impl. use the same typename pattern in order to be able to interoperate:
* https://github.com/ros2/rmw_connext/blob/34ffeaad30669060a085710727c48eae1b464929/rmw_connext_cpp/src/functions.cpp#L131-L132
* https://github.com/ros2/rmw_connext/blob/34ffeaad30669060a085710727c48eae1b464929/rmw_connext_dynamic_cpp/src/utility_templates.hpp#L64
* https://github.com/ros2/rmw_fastrtps/blob/5ef866fce0e020fc4531897ec2ebad679dbb7480/rmw_fastrtps_cpp/src/functions.cpp#L122

It should certainly be mentioned in the [design doc](http://design.ros2.org/articles/mapping_dds_types.html).

> If you have a generated reply topic type that lacks the '::' dynamic bridge crashes (because t.rfind("::") returns npos, but this is not handled).

The fact that the bridge crashed when a typename doesn't comply is certainly not intended but a bug. Please see #66 which should catch this case and at least prevent the crash. It would be great if you could try the patch and comment if it prevents the crash in your case.

mikaelarguedas, 2017-08-08:
@dirk-thomas what is the action item here? documenting the type naming pattern in the design doc? or is there something that should be done in this repo?

dirk-thomas, 2017-08-08:
Yes, documenting the type naming pattern *somewhere*. Nothing needs to happen in this repo imo.