ros2/ros1_bridge #77: Can't pass the custom message from ROS1 to ROS2.

mc-android-developer, 2017-06-26:
I've created custom message:
<pre>
string name
string description
</pre>
It's the same in ROS1 and ROS2 and remains in the same package 'skynet_msgs'
I've created yaml file as explained and added it to ros1_bridge/package.xml:
<pre>
-
  ros1_package_name: 'skynet_msgs'
  ros2_package_name: 'skynet_msgs'
</pre>
Here is my ros1_bridge/CMakeLists.txt file:
<pre>
cmake_minimum_required(VERSION 2.8.3)

project(ros1_bridge)

if(NOT WIN32)
  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14 -Wall -Wextra")
endif()

find_package(rmw REQUIRED)

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(rmw_implementation_cmake REQUIRED)
find_package(std_msgs REQUIRED)
find_package(skynet_msgs REQUIRED)

# find ROS 1 packages
include(cmake/find_ros1_package.cmake)

find_package(PkgConfig)
if(NOT PKG_CONFIG_FOUND)
  message(STATUS "Failed to find PkgConfig, skipping...")
  # call ament_package() to avoid ament_tools treating this as a plain CMake pkg
  ament_package()
  return()
endif()

find_ros1_package(roscpp)
if(NOT ros1_roscpp_FOUND)
  message(STATUS "Failed to find ROS 1 roscpp, skipping...")
  # call ament_package() to avoid ament_tools treating this as a plain CMake pkg
  ament_package()
  return()
endif()

find_ros1_package(std_msgs REQUIRED)
find_ros1_package(skynet_msgs REQUIRED)

set(TEST_ROS1_BRIDGE FALSE)
if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  find_package(diagnostic_msgs REQUIRED)
  find_ros1_package(diagnostic_msgs)
  find_ros1_package(roslaunch)
  ament_lint_auto_find_test_dependencies()
  if(ros1_diagnostic_msgs_FOUND AND ros1_roslaunch_FOUND)
    set(TEST_ROS1_BRIDGE TRUE)
  endif()
endif()

ament_export_include_directories(include)

ament_python_install_package(${PROJECT_NAME})

ament_package()

set(generated_path "${CMAKE_BINARY_DIR}/generated")
set(generated_files "${generated_path}/get_factory.cpp")
list(APPEND generated_files "${generated_path}/get_mappings.cpp")

# generate per package compilation units to keep the memory usage low
ament_index_get_resources(message_packages "rosidl_interfaces")
foreach(message_package ${message_packages})
  find_package(${message_package} REQUIRED)
  list(APPEND generated_files "${generated_path}/${message_package}_factories.cpp")
endforeach()

add_custom_command(
  OUTPUT ${generated_files}
  COMMAND ${PYTHON_EXECUTABLE} bin/ros1_bridge_generate_factories
    --output-path "${generated_path}" --template-dir resource
  WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR})

if(NOT WIN32)
  # ignore warning in ROS 1 message headers
  set_source_files_properties(${generated_files}
    PROPERTIES COMPILE_FLAGS "-Wno-unused-parameter")
endif()

include_directories(include ${generated_path})

function(custom_executable target)
  cmake_parse_arguments(
    ARG "ROS1_DEPENDENCIES" "" "TARGET_DEPENDENCIES" ${ARGN})

  add_executable(${target}${target_suffix}
    ${ARG_UNPARSED_ARGUMENTS})
  ament_target_dependencies(${target}${target_suffix}
    "rclcpp${target_suffix}"
    ${ARG_TARGET_DEPENDENCIES})
  if(ARG_ROS1_DEPENDENCIES)
    ament_target_dependencies(${target}${target_suffix}
      "ros1_roscpp"
      "ros1_std_msgs"
      "ros1_skynet_msgs")
  endif()
  if(ARG_DEPENDENCIES)
    add_dependencies(${target}${target_suffix} ${ARG_DEPENDENCIES})
  endif()

  install(TARGETS ${target}${target_suffix}
    DESTINATION bin)
endfunction()

macro(targets)
  if(NOT "${target_suffix} " STREQUAL " ")
    get_rclcpp_information("${rmw_implementation}" "rclcpp${target_suffix}")
  endif()

  custom_executable(simple_bridge_1_to_2 "src/simple_bridge_1_to_2.cpp"
    ROS1_DEPENDENCIES
    TARGET_DEPENDENCIES "std_msgs" "skynet_msgs")
  custom_executable(simple_bridge_2_to_1 "src/simple_bridge_2_to_1.cpp"
    ROS1_DEPENDENCIES
    TARGET_DEPENDENCIES "std_msgs" "skynet_msgs")

  custom_executable(simple_bridge "src/simple_bridge.cpp"
    ROS1_DEPENDENCIES
    TARGET_DEPENDENCIES "std_msgs" "skynet_msgs")

  add_library(${PROJECT_NAME}${target_suffix} SHARED
    "src/convert_builtin_interfaces.cpp"
    ${generated_files})
  ament_target_dependencies(${PROJECT_NAME}${target_suffix}
    "rclcpp${target_suffix}"
    ${message_packages}
    "ros1_roscpp"
    "ros1_std_msgs"
    "ros1_skynet_msgs")

  install(TARGETS ${PROJECT_NAME}${target_suffix}
    ARCHIVE DESTINATION lib
    LIBRARY DESTINATION lib
    RUNTIME DESTINATION bin)

  custom_executable(static_bridge
    "src/static_bridge.cpp"
    ROS1_DEPENDENCIES
    TARGET_DEPENDENCIES ${message_packages})
  target_link_libraries(static_bridge${target_suffix}
    ${PROJECT_NAME}${target_suffix})

  custom_executable(dynamic_bridge
    "src/dynamic_bridge.cpp"
    ROS1_DEPENDENCIES
    TARGET_DEPENDENCIES ${message_packages})
  target_link_libraries(dynamic_bridge${target_suffix}
    ${PROJECT_NAME}${target_suffix})

  if(TEST_ROS1_BRIDGE AND NOT "${target_suffix}" STREQUAL "")
    custom_executable(test_ros2_client_cpp "test/test_ros2_client.cpp")
    ament_target_dependencies("test_ros2_client_cpp${target_suffix}" "ros1_roscpp" "diagnostic_msgs")
    custom_executable(test_ros2_server_cpp "test/test_ros2_server.cpp")
    ament_target_dependencies("test_ros2_server_cpp${target_suffix}" "ros1_roscpp" "diagnostic_msgs")

    set(TEST_BRIDGE_DYNAMIC_BRIDGE "${CMAKE_INSTALL_PREFIX}/bin/dynamic_bridge${target_suffix}")
    set(TEST_BRIDGE_ROS2_CLIENT "$<TARGET_FILE:test_ros2_client_cpp${target_suffix}>")
    set(TEST_BRIDGE_ROS2_SERVER "$<TARGET_FILE:test_ros2_server_cpp${target_suffix}>")
    set(TEST_BRIDGE_RMW ${rmw_implementation})

    configure_file(
      test/test_dynamic_bridge.py.in
      test_dynamic_bridge${target_suffix}.py.genexp
      @ONLY
    )
    file(GENERATE
      OUTPUT test_dynamic_bridge${target_suffix}_$<CONFIGURATION>.py
      INPUT test_dynamic_bridge${target_suffix}.py.genexp)
    ament_add_nose_test(test_dynamic_bridge${target_suffix}
      "${CMAKE_CURRENT_BINARY_DIR}/test_dynamic_bridge${target_suffix}_$<CONFIGURATION>.py"
      TIMEOUT 60)
  endif()
endmacro()

if(TEST_ROS1_BRIDGE)
  add_executable(test_ros1_client "test/test_ros1_client.cpp")
  ament_target_dependencies(test_ros1_client "ros1_roscpp")
  add_executable(test_ros1_server "test/test_ros1_server.cpp")
  ament_target_dependencies(test_ros1_server "ros1_roscpp")
  set(TEST_BRIDGE_ROS1_ENV "${ros1_roslaunch_PREFIX}/env.sh")
  set(TEST_BRIDGE_ROSCORE "${ros1_roslaunch_PREFIX}/bin/roscore")
  set(TEST_BRIDGE_ROS1_CLIENT "$<TARGET_FILE:test_ros1_client>")
  set(TEST_BRIDGE_ROS1_SERVER "$<TARGET_FILE:test_ros1_server>")
endif()

call_for_each_rmw_implementation(targets GENERATE_DEFAULT)

install(
  FILES skynet_rules.yaml
  DESTINATION share/${PROJECT_NAME})
install(
  PROGRAMS bin/ros1_bridge_generate_factories
  DESTINATION lib/ros1_bridge_generate_factories
)
install(
  DIRECTORY cmake
  DESTINATION share/${PROJECT_NAME}
)
install(
  DIRECTORY include/
  DESTINATION include
)
install(
  DIRECTORY resource
  DESTINATION share/${PROJECT_NAME}
)
</pre> 
I've created publisher and subscriber nodes for ROS1 and ROS2. They pass the message fine between each other inside when work inside ROS1 or ROS2 only.
When I start roscore and dynamic bridge and I start subscriber on ROS1 and publisher on ROS2.
Message is not passed. Dynamic bridge says:
<pre>
failed to create 2to1 bridge for topic 'description_msgs_test' with ROS 2 type 'skynet_msgs/NodeDescription' and ROS 1 type '': No template specialization for the pair
failed to create 2to1 bridge for topic 'description_msgs_test' with ROS 2 type 'skynet_msgs/NodeDescription' and ROS 1 type '': No template specialization for the pair
failed to create 2to1 bridge for topic 'description_msgs_test' with ROS 2 type 'skynet_msgs/NodeDescription' and ROS 1 type '': No template specialization for the pair
</pre>

Could you hint what I missed?

mikaelarguedas, 2017-06-26:
Hi @mc-android-developer, thanks for reporting the issue.

The way the dynamic_bridge is designed you should not need to specify the mapping in a yaml file if the massage package + name is the same between ros1 and ros2. You should also not need to change anything in the CMakLists or the package.xml of the bridge.

From the error message it looks like your ROS1 workspace with your custom messages was not sourced when you built the bridge hence the (`ROS 1 type ''` message)
Could you give more details about your setup (what version of the code you are using mostly and maybe also the platform and the rmw implementation you are using).

Usually when I use custom messages I have:
- a ros2 workspace with just the core of ROS2
- ros1 installed from binaries
- an overlay catkin workspace with my custom message package
- an overlay ament (ROS2) workspace with my custom message package

The steps I use are:
- in one shell build my ros2 workspace
- in another shell build my catkin workspace
- in another shell: source my ros2 workspace and build my overlay workspace
- finally build the bridge in a new shell:
  - source /opt/ros/<ROSDISTRO>/setup.bash
  - source <MY_ROS2_WORKSPACE>/install/(local_)setup.bash
  - source my catkin workspace
  - source my ros2 overlay workspace
  - rm -rf ros2_ws/build/ros1_bridge
  - build the bridge: `ament build --only ros1_bridge`

If you can give this a try and report here with the outcome that would be awesome,
Thanks!

mc-android-developer, 2017-06-26:
Hi @mikaelarguedas. Thank you for prompt response.

>The way the dynamic_bridge is designed you should not need to specify the mapping in a yaml file if the massage package + name is the same between ros1 and ros2. You should also not need to change anything in the CMakLists or the package.xml of the bridge.

Right. That was my first effort. It didn't worked so I've added yaml file.
 
> From the error message it looks like your ROS1 workspace with your custom messages was not sourced when you built the bridge hence the (ROS 1 type '' message)

I've made sure that ROS1 workspace with custom message is sourced and recompiled bridge again. Here is an output:
<pre>
~/ros2_ws $ src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install -j1 --only ros1_bridge
# Topological order
 - (ament_cppcheck)
 - (ament_package)
 - (ament_cmake_core)
 - (ament_cmake_export_definitions)
 - (ament_cmake_export_dependencies)
 - (ament_cmake_export_include_directories)
 - (ament_cmake_export_interfaces)
 - (ament_cmake_export_libraries)
 - (ament_cmake_export_link_flags)
 - (ament_cmake_gmock)
 - (ament_cmake_gtest)
 - (ament_cmake_include_directories)
 - (ament_cmake_libraries)
 - (ament_cmake)
 - (ament_cmake_nose)
 - (ament_cmake_python)
 - (ament_cmake_target_dependencies)
 - (ament_cmake_test)
 - (ament_cmake_auto)
 - (ament_lint_auto)
 - (ament_pep8)
 - (ament_pep257)
 - (ament_pyflakes)
 - (ament_copyright)
 - (ament_clang_format)
 - (ament_cpplint)
 - (ament_index_python)
 - (ament_lint_cmake)
 - (ament_cmake_lint_cmake)
 - (ament_cmake_copyright)
 - (ament_cmake_clang_format)
 - (ament_cmake_cppcheck)
 - (ament_cmake_cpplint)
 - (ament_cmake_pep257)
 - (ament_cmake_pep8)
 - (ament_cmake_pyflakes)
 - (ament_cmake_uncrustify)
 - (console_bridge)
 - (fastcdr)
 - (fastrtps)
 - (gmock_vendor)
 - (gtest_vendor)
 - (launch)
 - (launch_testing)
 - (osrf_pycommon)
 - (ament_tools)
 - (poco_vendor)
 - (class_loader)
 - (uncrustify)
 - (ament_uncrustify)
 - (ament_lint_common)
 - (ament_index_cpp)
 - (connext_cmake_module)
 - (fastrtps_cmake_module)
 - (opensplice_cmake_module)
 - (python_cmake_module)
 - (rmw)
 - (rmw_implementation_cmake)
[... 348 lines cut ...]
failed to create 1to2 bridge for topic 'description_msgs_test' with ROS 1 type 'skynet_msgs/NodeDescription' and ROS 2 type 'skynet_msgs/NodeDescription': No template specialization for the pair
failed to create 1to2 bridge for topic 'description_msgs_test' with ROS 1 type 'skynet_msgs/NodeDescription' and ROS 2 type 'skynet_msgs/NodeDescription': No template specialization for the pair
failed to create 1to2 bridge for topic 'description_msgs_test' with ROS 1 type 'skynet_msgs/NodeDescription' and ROS 2 type 'skynet_msgs/NodeDescription': No template specialization for the pair
...
</pre>

I use Ubuntu Linux. ROS1 lunar (from binaries) and ROS2 beta1 (compiled from sources) with default rmw:
<pre>
~/ros2_ws $ cat ros2.repos 
repositories:
 ...
  ros2/ros1_bridge:
    type: git
    url: https://github.com/ros2/ros1_bridge.git
    version: release-beta1
 ...
  vendor/console_bridge:
    type: git
    url: https://github.com/ros/console_bridge.git
    version: 148df7e841b91da488ec3bb4abd295a2bccdb728

</pre>

Now I'm trying to do steps you described. I'll post result later.
Thank you!

mikaelarguedas, 2017-06-26:
Yeah this console output indicates that the bridge is not being rebuilt. If you try the same after deleting build/ros1_bridge it should rebuild and the output should contain:
```
-- Checking for module 'my_custom_msg_package'
--   Found my_custom_msg_package, version x.x.x
```

and later something like:
```
Generating generated/get_factory.cpp, generated/get_mappings.cpp, generated/my_custom_msg_package.cpp
```

mc-android-developer, 2017-06-26:
I did as you offered. I reverted all changes with yaml file in ros1_bridge and removed build/ros1_bridge dir.

<pre>
$ src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install -j1 --only ros1_bridge
# Topological order
 - (ament_cppcheck)
 - (ament_package)
 - (ament_cmake_core)
 - (ament_cmake_export_definitions)
 - (ament_cmake_export_dependencies)
 - (ament_cmake_export_include_directories)
 - (ament_cmake_export_interfaces)
 - (ament_cmake_export_libraries)
 - (ament_cmake_export_link_flags)
 - (ament_cmake_gmock)
 - (ament_cmake_gtest)
 - (ament_cmake_include_directories)
 - (ament_cmake_libraries)
 - (ament_cmake)
 - (ament_cmake_nose)
 - (ament_cmake_python)
 - (ament_cmake_target_dependencies)
 - (ament_cmake_test)
 - (ament_cmake_auto)
 - (ament_lint_auto)
 - (ament_pep8)
 - (ament_pep257)
 - (ament_pyflakes)
 - (ament_copyright)
 - (ament_clang_format)
 - (ament_cpplint)
 - (ament_index_python)
 - (ament_lint_cmake)
 - (ament_cmake_lint_cmake)
 - (ament_cmake_copyright)
 - (ament_cmake_clang_format)
 - (ament_cmake_cppcheck)
 - (ament_cmake_cpplint)
 - (ament_cmake_pep257)
 - (ament_cmake_pep8)
 - (ament_cmake_pyflakes)
 - (ament_cmake_uncrustify)
 - (console_bridge)
 - (fastcdr)
 - (fastrtps)
 - (gmock_vendor)
 - (gtest_vendor)
 - (launch)
 - (launch_testing)
 - (osrf_pycommon)
 - (ament_tools)
 - (poco_vendor)
 - (class_loader)
 - (uncrustify)
 - (ament_uncrustify)
 - (ament_lint_common)
 - (ament_index_cpp)
 - (connext_cmake_module)
 - (fastrtps_cmake_module)
 - (opensplice_cmake_module)
 - (python_cmake_module)
 - (rmw)
 - (rmw_implementation_cmake)
 - (rosidl_default_generators)
 - (rosidl_default_runtime)
 - (rosidl_generator_dds_idl)
 - (rosidl_parser)
 - (rosidl_cmake)
 - (rosidl_generator_c)
 - (rmw_connext_shared_cpp)
[... 426 lines cut ...]
-- Up-to-date symlink: /home/mc/ros2_ws/install/share/ros1_bridge/resource/get_mappings.cpp.em
-- Up-to-date symlink: /home/mc/ros2_ws/install/share/ros1_bridge/resource/pkg_factories.cpp.em
-- Up-to-date symlink: /home/mc/ros2_ws/install/share/ros1_bridge/resource/pkg_factories.hpp.em
-- [ament] Deploying: /home/mc/ros2_ws/install/local_setup.bash
-- [ament] Deploying: /home/mc/ros2_ws/install/local_setup.sh
-- [ament] Deploying: /home/mc/ros2_ws/install/local_setup.zsh
-- [ament] Deploying: /home/mc/ros2_ws/install/setup.bash
-- [ament] Deploying: /home/mc/ros2_ws/install/setup.sh
-- [ament] Deploying: /home/mc/ros2_ws/install/setup.zsh
-- [ament] Deploying: /home/mc/ros2_ws/install/_order_packages.py
</pre>

I still can't see: -- Checking for module 'skynet_msgs'

Only these messages:
<pre>
-- Found skynet_msgs: 0.0.0 (/home/mc/skynet-edge/core_ws/install/share/skynet_msgs/cmake)
...
[  1%] Generating generated/get_factory.cpp, generated/get_mappings.cpp, generated/diagnostic_msgs_factories.cpp, generated/stereo_msgs_factories.cpp, generated/rcl_interfaces_factories.cpp, generated/tf2_msgs_factories.cpp, generated/example_interfaces_factories.cpp, generated/builtin_interfaces_factories.cpp, generated/std_srvs_factories.cpp, generated/visualization_msgs_factories.cpp, generated/pendulum_msgs_factories.cpp, generated/lifecycle_msgs_factories.cpp, generated/geometry_msgs_factories.cpp, generated/trajectory_msgs_factories.cpp, generated/composition_factories.cpp, generated/std_msgs_factories.cpp, generated/shape_msgs_factories.cpp, generated/nav_msgs_factories.cpp, generated/actionlib_msgs_factories.cpp, generated/sensor_msgs_factories.cpp, generated/skynet_msgs_factories.cpp
...
[ 30%] Building CXX object CMakeFiles/ros1_bridge.dir/generated/skynet_msgs_factories.cpp.o
[ 81%] Building CXX object CMakeFiles/ros1_bridge__rmw_fastrtps_cpp.dir/generated/skynet_msgs_factories.cpp.o
</pre>

I'm sure all warkspaces (ros2 core, ros2 skynet_msgs, ros1 core, ros1 skynet_msgs) are sourced at this point. It still doesn't work this way. dynamic_bridge output is the same. Does it look like incorrect output as well?

mikaelarguedas, 2017-06-27:
@mc-android-developer sorry for the trouble, I just missed that part of your previous comment:
> I use Ubuntu Linux. ROS1 lunar (from binaries) and ROS2 beta1 (compiled from sources) with default rmw (fastrtps):

The support of overlay ros1 workspace has been fixed recently in the bridge: https://github.com/ros2/ros1_bridge/pull/67 so it's not part of beta1

mikaelarguedas, 2017-07-25:
@mc-android-developer I saw on other issues that you were working with the master branch these days. If you have the opportunity, can you try to pass custom messages using the latest version and report the outcome here ? Thanks!

mc-android-developer, 2017-07-26:
Hi @mikaelarguedas 
I checked that on master branch. It works fine. The same way it worked before in ROS2 beta1.
Here is the test project I use to test it: https://drive.google.com/open?id=0B68BPOrWQeULcmVXNDdTdTRYcFE
Please check README.md file. It has detailed instruction how to run it.

mikaelarguedas, 2017-07-26:
Glad it works for you, I'm going to close this issue then

closed 2017-07-26 by mikaelarguedas

whyscience, 2018-01-05:
Thank you too.