ros2/ros1_bridge #78: Unexpected AF_INET Problems

allenh1, 2017-06-27:
Running the bridge with both fastRTPS and connexts leads to the same behavior after a large amount of messages (consistently ~950 for connext, ~1100 for fastRTPS):

```
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
```

This is built on top of master.

dirk-thomas, 2017-06-27:
Please provide reproducible steps as well as more information about your network setup.

allenh1, 2017-06-27:
@dirk-thomas Sorry, I was trying to reduce it to the smallest case I could find.

Network Info:
===========
```
ROS_MASTER_URI=http://machina.local:11311
ROS_HOSTNAME=machina
```

To reproduce:
============
On ROS 1 Side:
-----------------------
```
$ roscore
```

```
$ rosrun roscpp_tutorials talker
```

On ROS 2 Side:
----------------------
```
$ ros2 run ros1_bridge dynamic_bridge
```

```
$ ros2 run demo_nodes_cpp listener
```

allenh1, 2017-06-27:
The following network info also fails after a large amount of messages:
```
ROS_MASTER_URI=http://localhost:11311
ROS_HOSTNAME=localhost
```

dirk-thomas, 2017-06-27:
Can you reproduce the problem when using an IP address instead of the hostname and not setting the `ROS_HOSTNAME`?

allenh1, 2017-06-27:
It seems to not crash if you use an IP address. It's about 2000 messages in now and has yet to crash.

allenh1, 2017-06-27:
Update: New problem with the IP address one, we are leaking a file descriptor somewhere. I'll look into it.

```
[ERROR] [1498606076.523479788]: getifaddr() failed
[ERROR] [1498606076.523514564]: socket() failed with error [Too many open files]
Failed to connect to 127.0.0.1:52001
[ERROR] [1498606076.523994787]: getifaddr() failed
[ERROR] [1498606076.524024879]: socket() failed with error [Too many open files]
Failed to connect to 127.0.0.1:46055
[ERROR] [1498606076.524405673]: getifaddr() failed
[ERROR] [1498606076.524433704]: socket() failed with error [Too many open files]
Failed to connect to 127.0.0.1:46055
[ERROR] [1498606076.524794888]: getifaddr() failed
[ERROR] [1498606076.524822730]: socket() failed with error [Too many open files]
```

allenh1, 2017-06-28:
OK, I found the file descriptor leak and fixed it (giving it a long time to crash with a very small number of fd's -- past the 2000). PR to come.

allenh1, 2017-06-28:
PR for file descriptor leak: #79

dirk-thomas, 2017-06-28:
Has this been fixed for you by #79?

allenh1, 2017-06-28:
This was indeed fixed by #79, I'm happy to say!

closed 2017-06-28 by allenh1

loulansuiye, 2018-08-24:
@allenh1 Hi, have you solved that problem ?I met the simillar question too...confusing........................

allenh1, 2018-08-24:
@loulansuiye this was fixed by the PR I mentioned above. It would be easier to tell what's wrong for you if you could post some output (though it is very unlikely that this would regress).

In the meantime, you can play around with your `ulimit -n` settings (the number of open file descriptors you can have open at once -- usually this defaults to~1024, if memory serves me) to see if this is the root of the problem.

danghuybk, 2018-10-31:
I meet the same problem. I dont know how to fix it :(

allenh1, 2018-10-31:
> I meet the same problem. I dont know how to fix it :(

@danghuybk could you please provide your ROS 2 version, OS, and a reproducible testcase?