ros2/ros1_bridge #82: dynamic_bridge: No executable found

volkandre, 2017-07-24:
After a fresh ros2 beta2 source install I tried `ros2 run ros1_bridge dynamic_bridge`, but got a `No executable found` error.
So I tried to compile ros1_bridge manually, by typing: `src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install -j1 --only ros1_bridge` (as suggested in the ros1_bridge Readme file). I sourced my ROS1 environment beforehand.
But ament did not create any executables, although there was no error in the output: 

<pre>Process package 'ros1_bridge' with context:
--------------------------------------------------------------------------------
 source_space => /home/username/ros2_ws/src/ros2/ros1_bridge
  build_space => /home/username/ros2_ws/build/ros1_bridge
install_space => /home/username/ros2_ws/install
   make_flags => -j1
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'ros1_bridge'
==> '. /home/username/ros2_ws/build/ros1_bridge/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/username/ros2_ws/build/ros1_bridge'
==> '. /home/username/ros2_ws/build/ros1_bridge/cmake__build.sh && /usr/bin/make -j1' in '/home/username/ros2_ws/build/ros1_bridge'
+++ Installing 'ros1_bridge'
==> '. /home/username/ros2_ws/build/ros1_bridge/cmake__install.sh && /usr/bin/make install' in '/home/username/ros2_ws/build/ros1_bridge'
Install the project...
-- Install configuration: ""
-- Execute custom install script
-- Symlinking: /home/username/ros2_ws/install/share/ament_index/resource_index/package_run_dependencies/ros1_bridge
-- Symlinking: /home/username/ros2_ws/install/share/ament_index/resource_index/parent_prefix_path/ros1_bridge
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/environment/ament_prefix_path.sh
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/environment/path.sh
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/local_setup.bash
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/local_setup.sh
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/local_setup.zsh
-- Symlinking: /home/username/ros2_ws/install/share/ament_index/resource_index/packages/ros1_bridge
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/cmake/ros1_bridgeConfig.cmake
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/cmake/ros1_bridgeConfig-version.cmake
-- Symlinking: /home/username/ros2_ws/install/share/ros1_bridge/package.xml
-- [ament] Deploying: /home/username/ros2_ws/install/local_setup.bash
-- [ament] Deploying: /home/username/ros2_ws/install/local_setup.sh
-- [ament] Deploying: /home/username/ros2_ws/install/local_setup.zsh
-- [ament] Deploying: /home/username/ros2_ws/install/setup.bash
-- [ament] Deploying: /home/username/ros2_ws/install/setup.sh
-- [ament] Deploying: /home/username/ros2_ws/install/setup.zsh
-- [ament] Deploying: /home/username/ros2_ws/install/_order_packages.py</pre>

I had to remove `--build-tests` from the ament command to make ament create the executables successfully (`src/ament/ament_tools/scripts/ament.py build --symlink-install -j1 --only ros1_bridge`). After this successful build, ament also worked with `--build-tests`.

I do not know, if this strange behavior is due to a bug in the CMakeLists.txt file or in ament or if it has something to do with my machine. I just wanted to let you know.

mikaelarguedas, 2017-07-24:
Thanks for reporting the issue,
Can you try to remove your build/ros1_bridge folder and run ament again ?
```
rm -r build/ros1_bridge
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install -j1 --only ros1_bridge
```
My guess would be that the first time you ran ament you didn't pass the `--skip-packages ros1_bridge` argument. If so I'll update the bridge README to make sure that the folder doesn't exist before trying to build it.

dirk-thomas, 2017-07-24:
If you didn't pass `--skip-packages ros1_bridge` on the first build invocation you have to pass `--force-cmake-configure` the second time when building the bridge since otherwise (as seen in your output) CMake is not being run again and doesn't try to find ROS 1 and therefore doesn't build the bridge.

volkandre, 2017-07-24:
Thanks for your replies. Your guesses were right. Thanks for the help and the Readme fix.

closed 2017-07-24 by volkandre

Dikshuy, 2021-02-12:
while running this command.
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
 I am getting this warning, 
CMake Warning at CMakeLists.txt:34 (message):
  Failed to find ROS 1 roscpp, skipping...

After doing this, I ran the following command
 ros2 run ros1_bridge dynamic_bridge

No executable found
but it shows this above error. I have done the above-mentioned things but still unable to resolve this issue. can you please help me out

jacobperron, 2021-03-12:
@Dikshuy Asking questions on old, closed threads on GitHub aren't likely to receive much attention. I suggest you ask your question on [ROS Answers](https://answers.ros.org/).

lwbdegit, 2022-08-29:
> while running this command. colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure I am getting this warning, CMake Warning at CMakeLists.txt:34 (message): Failed to [...]

I've met the same error, did you solve it?

Dikshuy, 2022-08-29:
no, i haven't

karanchahal, 2023-04-01:
I have the same issue

hsurya08, 2023-06-08:
Hi @Dikshuy,
I am facing the same issue. Can you please let me know if you are able to find a solution. Thanks!

Dikshuy, 2023-07-03:
not yet :/