ros2/ros1_bridge #84: Assess how much memory is needed to build the bridge

mikaelarguedas, 2017-07-24:
Right now the bridge instructions tell to build with `-j2`. Since the bridge has been refactored to have a separate compilation unit per message package, it would be valuable to assess how much memory is needed (at most) to build a message package. We could then update the build instructions accordingly for users to decide which flag is the most adapted to their setup.

dirk-thomas, 2017-07-24:
Does it make sense to spend time on this? The information will also get outdated over time.

I would suggest to instead mention the memory load in the readme and advice the user to specific `-jN` when needed.

mikaelarguedas, 2017-07-24:
That's a fair point, I put the `help wanted` tag because I didnt expected us to spend time on it in the near future but could be useful if anyone in the community is interested in doing this benchmarking. I'm fine with the suggested alternative, feel free to close this and submit the corresponding PR to the readme

dirk-thomas, 2017-07-24:
I don't mind who will be working on this. I just don't think any memory analysis will be a long term "solution".

See #85.

mikaelarguedas, 2017-07-24:
Sounds good to me

PR #85 (2017-07-24): note about memory
Fixes #84.
closed 2017-07-24 by dirk-thomas