ros2/ros1_bridge #88: ROS1 package that causes ros1_bridge build to fail in find_ros1_package

deng02, 2017-09-01:
**Required Info:**
- Operating System:: Ubutnu 16.04
- Installation: Using ROS Kinetic and ROS2 from CI server (http://ci.ros2.org/view/packaging/job/packaging_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2) Sept 1 4:20pm EST

**Reproduction steps:**
1.  mkdir -p ~/test_ws/src && cd ~/test_ws/src
2. git clone https://github.com/clearpathrobotics/dynamic_gazebo_models
4.  cd .. && source /opt/kinetic/setup.bash && catkin_make
5.  mkdir -p ~/bridge_ws/src && cd ~/bridge_ws/src
6. git clone http://github.com/ros2/ros1_bridge
7. cd .. && source /opt/ros2-linux/setup.bash && ament build -j1

The build will fail with the message:
```
--   Found actionlib_msgs, version 1.12.5
-- Checking for module 'bond'
--   Found bond, version 1.7.19
-- Checking for module 'diagnostic_msgs'
--   Found diagnostic_msgs, version 1.12.5
-- Checking for module 'dynamic_gazebo_models'
--   Found dynamic_gazebo_models, version 0.1.0
CMake Error at cmake/find_ros1_package.cmake:63 (message):
  pkg-config module 'dynamic_gazebo_models' failed to find library
  'dynamic_gazebo_models'
Call Stack (most recent call first):
  CMakeLists.txt:47 (find_ros1_package)

-- Configuring incomplete, errors occurred!
See also "/home/deng/bridge_ws/build/ros1_bridge/CMakeFiles/CMakeOutput.log".

<== Command '. /home/deng/bridge_ws/build/ros1_bridge/cmake__build.sh && /usr/bin/cmake /home/deng/bridge_ws/src/ros1_bridge -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/deng/bridge_ws/install' failed in '/home/deng/bridge_ws/build/ros1_bridge' with exit code '1'
<== Command '. /home/deng/bridge_ws/build/ros1_bridge/cmake__build.sh && /usr/bin/cmake /home/deng/bridge_ws/src/ros1_bridge -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/deng/bridge_ws/install' failed in '/home/deng/bridge_ws/build/ros1_bridge' with exit code '1'
```
We're unsure what it is about the package that needs to be fixed so that ros1_bridge can be built with a workspace that contains this.

dirk-thomas, 2017-09-01:
The package exports a library named `${PROJECT_NAME` (https://github.com/clearpathrobotics/dynamic_gazebo_models/blob/83269550e3646a712f380eb4fc381b3e02af460c/CMakeLists.txt#L22) aka `dynamic_gazebo_models` which is why it is being listed in the generated `pkg-config` file. But since the package doesn't install the library anywhere it is unavailable when the bridge tried to find that library.

closed 2017-09-01 by dirk-thomas

deng02, 2017-09-01:
@dirk-thomas  Thanks!