ros2/ros1_bridge #103: how to add my message type  to ros1_bridge ?

lxbeyond, 2018-02-12:
when i make my package in ROS2 ,I want to use message or receive topic  type that defined by myself in ROS1 , how to build this message type into ros1_bridge？ I did as https://github.com/ros2/ros1_bridge/blob/master/README.md, it said "This can be done by adding explicit dependencies on the message/service packages to the package.xml of the bridge", but it did not work.

mikaelarguedas, 2018-02-12:
@lxbeyond this is not enough information to help you. Please fill out the issue template when you open issues without the exact steps you followed and a way to reproduce the issue it's hard for us to help you figure out what didnt go as expected.

Make sure you do the following: 
- provide a ROS 1 message package with your message definition in your ROS 1 workspace
- provide a ROS 2 message package with your message definition in your ROS 2 workspace (they should have the same name and the same message name in ROS 1 and ROS 2)

- build your ROS 1 workspace with only `/opt/ros/kinetic/setup/bash` sourced
- build your ROS 2 workspace with only `/opt/ros/ardent/setup.bash` sourced
- In a new shell:
```
# Source your ROS 2 installation:
. <install-space-with-ros2>/local_setup.bash
# And if you have a ROS 1 overlay workspace, something like:
# . <install-space-to-ros1-overlay-ws>/setup.bash
# And if you have a ROS 2 overlay workspace, something like:
# . <install-space-to-ros2-overlay-ws>/local_setup.bash
```
```
ament build --build-tests --symlink-install --only ros1_bridge --force-cmake-configure
```

lxbeyond, 2018-02-13:
thanks a lot @mikaelarguedas , I did as your suggestion, and it works .  a more question , if package and message file name are different in ros1 and ros2, but data structure of message is the same , how to make they communicate with each other ?

mikaelarguedas, 2018-02-13:
You can define a "mapping rule" to tell the bridge the mapping between the ROS 1 package/msg name and the ROS 2 one (https://github.com/ros2/geometry2/blob/ros2/tf2_msgs/tfmessage_bridge_mapping_rule.yaml). Then this mapping rule file need to be exported in the package.xml of your ROS 2 package (https://github.com/ros2/geometry2/blob/8ad4f892889fd2bfa92459413b75124be14ed026/tf2_msgs/package.xml#L28)

mikaelarguedas, 2018-02-13:
Thanks for confirming that it's fixed.
As the original issue is resolved I'm going to close this. If you face any problem using custom messages with the bridge, feel free to open another ticket with all the information required to reproduce the issue

closed 2018-02-13 by mikaelarguedas

tanyouliang95, 2019-01-14:
Hi I have the same issue with using `ros1_bridge` with custom message, i tried to source these in sequence:
```
source /opt/ros/kinetic/setup/bash
source <install-space-with-ros2>/local_setup.bash
source ~/catkin_ws/devel/setup.bash
colcon build --build-tests --symlink-install --only ros1_bridge --force-cmake-configure
```

Then i got a compilation shortly after colcon build:
```
Failed to call rosmsg/rossrv
Call Stack (most recent call first):
  CMakeLists.txt:43 (find_ros1_interface_packages)

---
Failed   <<< ros1_bridge	[ Exited with code 1 ]
```
This seems to be related to source in ros1. Would kindly need some help with this, Thanks!!!!

dirk-thomas, 2019-01-14:
@tanyouliang95 this is a closed ticket. Please ask your question on answers.ros.org instead since it doesn't sounds like a bug in this package. Please make sure to include now information, e.g. the sources of the packages you are trying to build.