ros2/ros1_bridge #104: Bridge appears incapable of bridging ROS1 /clock to ROS2

matt-attack, 2018-02-14:
In ROS1 the rosgraph_msgs/Clock message consists of a time type internally, but in ROS2 the /clock topic is a builtin_interfaces/Time. It does not appear that the bridge can handle bridging this topic from ROS1 to ROS2.

Is there a way to force the bridge to handle this kind of conversion or should a special case be added to the bridge to allow for sim time and bag file playback?

gavanderhoorn, 2018-02-14:
Could this perhaps be solved by specifying a [custom mapping rule](https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst#how-can-i-specify-custom-mapping-rule-for-messages)?

matt-attack, 2018-02-14:
Where could I put the remapping? It appears it would need to be added to the builtin_interfaces package for the bridge to accept it otherwise I get an error similar to this one:
`Ignoring rule which affects a different ROS 2 package (builtin_interfaces) then the one it is defined in (my_msgs)`

dirk-thomas, 2018-03-26:
The problem is not related to the differences in the `duration` or `time` types. Those are already being handled in the `ros`_bridge`: see https://github.com/ros2/ros1_bridge/blob/03f95fea565c2eb28057df8f7b1595bdeac995bc/include/ros1_bridge/convert_builtin_interfaces.hpp

The bridge simply can't bridge a message definition which is not available on both side. And `Clock` doesn't exist in ROS 2 atm. Please see ros2/common_interfaces#50 which add the `Clock` message for ROS 2. With that being present bridging the message will work.

dirk-thomas, 2018-05-14:
This is not a bug on the `ros1_bridge` anymore. In ROS 1 the `clock` topic is documented to have the type `rosgraph_msgs/Clock` (see http://wiki.ros.org/Clock#Clock_Message).

@tfoote Can you please follow up on this since you added the clock topic to `rclcpp`.

dirk-thomas, 2018-05-14:
I created ros2/rclcpp#473 to track this.

closed 2018-05-14 by dirk-thomas

dirk-thomas, 2018-05-14:
Yes, you can temporarily work around it by defining a [remapping rule](https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst#how-can-i-specify-custom-mapping-rule-for-messages).