ros2/ros1_bridge #108: [field mapping] A missing field from 'fields_1_to_2' will silently not be bridged and be zero-initialized instead

mikaelarguedas, 2018-03-28:
If one need to specify the mapping from one field in ROS 1 that's different in ROS 2, you currently need to explicitly list all the field of the message even if they are identical.
If some fields are ommited from the rules, the bridge will not complain, it will just not populate these fields and pass fields zero initialized instead.

Is it intentional? to e.g. allow messages that have more fields in one ROS version to be passed on the other side of the bridge?

dirk-thomas, 2018-03-28:
> Is it intentional?

Yes, that is intentional. If not listed fields would still be mapped "automatically" how would the user be able to choose *not* to map those?

mikaelarguedas, 2018-03-28:
> If not listed fields would still be mapped "automatically" how would the user be able to choose not to map those?

Maybe by providing an option to opt-out of passing specific fields rather than opting in ? Providing a way to specify "custom mapping of specific fields" and "ignoring specific fields".
As this is not documented (AFAICT) and silently applied by the bridge, it is pretty misleading. The documentation says a "custom mapping" can be applied while 7 of them are identical. So if I need a custom handling for a specific field I would expect to provide only the custom mapping of that field and not expect the other fields to be impacted or ignored.

In this case I bridge a [`DisparityImage`](http://docs.ros.org/jade/api/stereo_msgs/html/msg/DisparityImage.html), and only a single field that needs to be camelCased, I need to provide manually 8 fields "custom mapping" to get the message across.

I don't know what use case we expect to be the most common (not bridging specific ROS1 field v.s. specifying custom mapping for a specific field) but I think both should be supported with minimal effort on the user side.

dirk-thomas, 2018-03-28:
> Maybe by providing an option to opt-out of passing specific fields rather than opting in ? Providing a way to specify "custom mapping of specific fields" and "ignoring specific fields".

That would be possible. In my opinion it makes the configuration more complicated though that is why I used this simple mapping.

> As this is not documented (AFAICT) and silently applied by the bridge, it is pretty misleading.

#109 adds a sentence to clarify the current need for explicitly mapping all fields.

> The documentation says a "custom mapping" can be applied while 7 of them are identical.

Can you post a link to the doc you refer. I didn't find on quick look.

mikaelarguedas, 2018-03-28:
> That would be possible. In my opinion it makes the configuration more complicated though that is why I used this simple mapping.

That's a fair point, maybe we should revisit this if we decide not to relax the field naming constraints in ROS 2.

> #109 adds a sentence to clarify the current need for explicitly mapping all fields.

:+1: 

> Can you post a link to the doc you refer. I didn't find on quick look.

this is a combination of: "fields are associated with each other. If one of the two associated messages has fields which is are not part of the other message they are being ignored. " and "Additional mapping rules can be provided by a ROS 2 package using a yaml file.". This implies that the mappings are additionals not replacements. So I would expect to provide only "additional" rules and not have to replace them all

dirk-thomas, 2018-03-28:
> So I would expect to provide only "additional" rules and not have to replace them all

Please feel free to improve the docs in any way which help to avoid the confusion.

mikaelarguedas, 2018-08-10:
Documentation clarified this in https://github.com/ros2/ros1_bridge/pull/109, closing

closed 2018-08-10 by mikaelarguedas