ros2/ros1_bridge #111: Bidirectional bridge duplicates messages in ROS1

Ekci, 2018-04-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - 0.4.0
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
I have generated a bidirectional bridge using parameter_bridge to send a std_msgs/Float32MultiArray messages from ROS1 to ROS2. It works but I have just realized that when I use ```rostopic hz /topic``` I get twice the frequency I set in my code. Turning off bridge make ```rostopic hz /topic``` show correct value. Using ```rostopic info /topic``` shows:

```Publishers: 
 * /start_controller (http://xxx:44737/)
 * /ros_bridge (http://xxx:38531/)

Subscribers: 
 * /rostopic_28342_1522934521725 (http://xxx:46713/)
 * /ros_bridge (http://xxx:38531/)
```

Inspecting ROS1 topic clearly shows that every message is duplicated. This makes me think that parameter_bridge sends ROS1 message to ROS2 and then from ROS2 back to ROS1. Unfortunately it's not looping back and doesn't crash: messages on ROS2 side are not duplicated.

dirk-thomas, 2018-04-05:
Duplicate of #43.

closed 2018-04-05 by dirk-thomas

dhood, 2018-04-05:
@Ekci thanks for reporting this. As mentioned, there's useful info in #43, including pointers if you are interested in working on a fix for the issue yourself (in which case we would welcome a pull request)