ros2/ros1_bridge #120: Packaging jobs failing to find 'builtin_interfaces'

dhood, 2018-06-15:
Our linux packaging jobs (the only ones that build the bridge) failed overnight with the following CMake error:

```
Starting >>> ros1_bridge
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found rmw: 0.4.0 (/home/rosbuild/ci_scripts/ws/install/share/rmw/cmake)
-- Found PythonInterp: /home/rosbuild/ci_scripts/venv/bin/python3 (found suitable version "3.6.5", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home/rosbuild/ci_scripts/venv/bin/python3
-- Found rclcpp: 0.4.0 (/home/rosbuild/ci_scripts/ws/install/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.4.0 (/home/rosbuild/ci_scripts/ws/install/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rmw_fastrtps_cpp: 0.4.0 (/home/rosbuild/ci_scripts/ws/install/share/rmw_fastrtps_cpp/cmake)
-- Found fastrtps_cmake_module: 0.4.0 (/home/rosbuild/ci_scripts/ws/install/share/fastrtps_cmake_module/cmake)
-- Found FastRTPS: /home/rosbuild/ci_scripts/ws/install/include  
-- Found std_msgs: 0.4.0 (/home/rosbuild/ci_scripts/ws/install/share/std_msgs/cmake)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'roscpp'
--   Found roscpp, version 1.14.2
-- Checking for module 'std_msgs'
--   Found std_msgs, version 0.5.11
-- Checking for module 'actionlib_msgs'
--   Found actionlib_msgs, version 1.12.6
-- Checking for module 'builtin_interfaces'
--   No package 'builtin_interfaces' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  cmake/find_ros1_package.cmake:29 (pkg_check_modules)
  CMakeLists.txt:47 (find_ros1_package)

-- Configuring incomplete, errors occurred!
```

Four builds failed the same way: (linux, arm) x (xenial, bionic).
An observation that I've made so far is that the four all used the docker cache for each step of the docker build process. The cache invalidation via the `today_str` injection doesn't appear to be in the packaging jobs. I'll add that and hope that that's enough to fix it (maybe some outdated deps), and if not we can investigate the actual cause.

mikaelarguedas, 2018-06-15:
Packaging job link: https://ci.ros2.org/view/packaging/job/packaging_linux/1107

dhood, 2018-06-15:
Invalidating the docker cache (https://github.com/ros2/ci/pull/194) didn't fix it [image] (was worth a shot 😉) so we will have to find the actual problem.

dhood, 2018-06-15:
Update.. the cache invalidation must have just needed to be higher up, because since I deployed a different change that added a build arg to the dockerfile, it works:  [image].

I'll run a few more jobs to check it's fixed for other configurations.

dhood, 2018-06-15:
arm bionic: [image]
arm xenial: [image]

linux bionic: [image]
linux xenial: [image]

so this looks fixed!

closed 2018-06-15 by dhood

dirk-thomas, 2018-06-15:
Do we know which exact line needed to be re-run and how the Dockerfile should be updated to catch this in the future?

dhood, 2018-06-15:
Not with any confidence (my focus was to get it working again). I can't say which line was the issue, or even that using the cache was the cause (maybe something else happened at the same time e.g. PR merged). If it was the dockerfile then I think it would be from something above [this line](https://github.com/ros2/ci/blob/f47f0ef8a6aae0faad63f7e93f71ca63e62c3599/linux_docker_resources/Dockerfile#L94) but nothing in there stands out as "reasonable". We could move the dockerfile invalidation higher to be conservative I suppose; can't hurt.

dhood, 2018-06-15:
this issue had a similar error message: https://github.com/ros2/ros1_bridge/issues/88
if anyone knows of anything that might relate that to builtin_interfaces then that would help narrow it down, otherwise I am at a loss