ros2/ros1_bridge #125: Opensplice service server does not create bridge

ironmig, 2018-06-22:
## Bug report

When dynamic_bridge is run using rmw_opensplice_cpp, a bridge is not created when a service server is run on the ros2 side and service client is run on ros1 side. This behavior works as expected with rmw_fastrtps_cpp and rmw_connext_cpp.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source
- Version or commit hash:
  -  master
- DDS implementation:
  - opensplice
- Client library (if applicable):
  - Either rclpy or rclpp

#### Steps to reproduce issue
_replace source files with your workspace directories_
**Terminal 1**
```
. ~/ros2_ws/install/setup.bash
.  /opt/ros/melodic/setup.bash
roscore
```
**Terminal 2**
```
. ~/ros2_ws/install/setup.bash
.  /opt/ros/melodic/setup.bash
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run ros1_bridge dynamic_bridge
```
**Terminal 3**
```
. ~/ros2_ws/install/setup.bash
RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run demo_nodes_cpp add_two_ints_server
```
**Terminal 4**
```
.  /opt/ros/melodic/setup.bash
rosrun roscpp_tutorials add_two_ints_client
```

#### Expected behavior
* dyanmic_bridge prints ```Create 1 to 2 bridge for service /add_to_ints
* Client prints ```Sum: 9```

#### Actual behavior
Nothing is printed by dynamic_bridge and service server, service client prints ```[ERROR] [time]: Failed to call service add_two_ints```

tfoote, 2018-06-22:
I've been able to reproduce this. It is specific to Opensplice and also specific to a 1 to 2 bridge. The 2 to 1 bridge works. (ros1 server ros2 client)

When using FastRTPS the service shows in `rosservice list` 

```
^Ctfoote@snowman:~/work/workspaces/ros2 Last: [0] (278s Seconds)
$ rosservice list
/add_two_ints
/ros_bridge/get_loggers
/ros_bridge/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
tfoote@snowman:~/work/workspaces/ros2 Last: [0] (0s Seconds)
$ rosservice info /add_two_ints
Node: /ros_bridge
URI: rosrpc://snowman:42874
Type: roscpp_tutorials/TwoInts
Args: a b
tfoote@snowman:~/work/workspaces/ros2 Last: [0] (0s Seconds)
```

with the Fast RTPS bridge
```
tfoote@snowman:~ Last: [0] (289s Seconds)
$ ros2 run ros1_bridge dynamic_bridge
Created 1 to 2 bridge for service /add_two_ints

```

FastRTPS server
```
tfoote@snowman:~/work/workspaces/ros2 Last: [0] (12s Seconds)
$ ros2 run demo_nodes_cpp add_two_ints_server
```

Versus with OpenSplice
bridge
```
tfoote@snowman:~ Last: [0] (102s Seconds)
$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run ros1_bridge dynamic_bridge
========================================================================================
Report      : WARNING
Date        : Fri Jun 22 15:35:27 PDT 2018
Description : dlopen of auto-translated name 'libdurability.so' failed: libdurability.so: cannot open shared object file: No such file or directory
Node        : snowman
Process     : dynamic_bridge <14282>
Thread      : spliced 7f79d8917700
Internals   : 6.7.180404OSS///os_libraryOpen/os_library.c/59/0/1529706927.187517531/84
========================================================================================
Report      : ERROR
Date        : Fri Jun 22 15:35:27 PDT 2018
Description : dlopen of 'durability' failed: durability: cannot open shared object file: No such file or directory
Node        : snowman
Process     : dynamic_bridge <14282>
Thread      : spliced 7f79d8917700
Internals   : 6.7.180404OSS///os_libraryOpen/os_library.c/70/0/1529706927.187736874/84
========================================================================================
Report      : ERROR
Date        : Fri Jun 22 15:35:27 PDT 2018
Description : Problem opening 'durability'

Node        : snowman
Process     : dynamic_bridge <14282>
Thread      : spliced 7f79d8917700
Internals   : 6.7.180404OSS///OpenSplice domain service/spliced.c/633/0/1529706927.187753481/84
========================================================================================
Report      : WARNING
Date        : Fri Jun 22 15:35:27 PDT 2018
Description : Could not start service durability as thread
Node        : snowman
Process     : dynamic_bridge <14282>
Thread      : spliced 7f79d8917700
Internals   : 6.7.180404OSS///OpenSplice domain service/spliced.c/706/0/1529706927.187763655/84
========================================================================================
Report      : WARNING
Date        : Fri Jun 22 15:35:27 PDT 2018
Description : using network interface enp0s25 (172.23.3.99) selected arbitrarily from: enp0s25, wlp3s0, docker0

Node        : snowman
Process     : dynamic_bridge <14282>
Thread      : ddsi2 7f79c52dc700
Internals   : 6.7.180404OSS///ddsi2/q_nwif.c/988/0/1529706927.189869238/84

```

server
```
tfoote@snowman:~/work/workspaces/ros2 Last: [0] (25s Seconds)
$ RMW_IMPLEMENTATION=rmw_opensplice_cpp ros2 run demo_nodes_cpp add_two_ints_server
========================================================================================
Report      : WARNING
Date        : Fri Jun 22 15:36:28 PDT 2018
Description : dlopen of auto-translated name 'libdurability.so' failed: libdurability.so: cannot open shared object file: No such file or directory
Node        : snowman
Process     : add_two_ints_server <14965>
Thread      : spliced 7f2293329700
Internals   : 6.7.180404OSS///os_libraryOpen/os_library.c/59/0/1529706988.800043655/84
========================================================================================
Report      : ERROR
Date        : Fri Jun 22 15:36:28 PDT 2018
Description : dlopen of 'durability' failed: durability: cannot open shared object file: No such file or directory
Node        : snowman
Process     : add_two_ints_server <14965>
Thread      : spliced 7f2293329700
Internals   : 6.7.180404OSS///os_libraryOpen/os_library.c/70/0/1529706988.800253473/84
========================================================================================
Report      : ERROR
Date        : Fri Jun 22 15:36:28 PDT 2018
Description : Problem opening 'durability'

Node        : snowman
Process     : add_two_ints_server <14965>
Thread      : spliced 7f2293329700
Internals   : 6.7.180404OSS///OpenSplice domain service/spliced.c/633/0/1529706988.800268320/84
========================================================================================
Report      : WARNING
Date        : Fri Jun 22 15:36:28 PDT 2018
Description : Could not start service durability as thread
Node        : snowman
Process     : add_two_ints_server <14965>
Thread      : spliced 7f2293329700
Internals   : 6.7.180404OSS///OpenSplice domain service/spliced.c/706/0/1529706988.800287470/84
========================================================================================
Report      : WARNING
Date        : Fri Jun 22 15:36:28 PDT 2018
Description : using network interface enp0s25 (172.23.3.99) selected arbitrarily from: enp0s25, wlp3s0, docker0

Node        : snowman
Process     : add_two_ints_server <14965>
Thread      : ddsi2 7f228dff8700
Internals   : 6.7.180404OSS///ddsi2/q_nwif.c/988/0/1529706988.802620998/84

```

ROS1 
```
tfoote@snowman:~/work/workspaces/ros2 Last: [0] (0s Seconds)
$ rosservice list
/ros_bridge/get_loggers
/ros_bridge/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level

```

I don't seem to have `libdurability.so` on my system.

tfoote, 2018-06-22:
It appears that the factory is not being generated for `roscpp_tutorials`

```
$ grep -rI get_service_factory_ros*
ros1_bridge/generated/rosgraph_msgs_factories.cpp:get_service_factory_rosgraph_msgs(const std::string & ros_id, const std::string & package_name, const std::string & service_name)
ros1_bridge/generated/get_factory.cpp:  factory = get_service_factory_rosgraph_msgs(ros_id, package_name, service_name);
ros1_bridge/generated/rosgraph_msgs_factories.hpp:get_service_factory_rosgraph_msgs(const std::string & ros_id, const std::string & package_name, const std::string & service_name);
```
I've confirmed that the package is in my workspace where I'm building the bridge.

dirk-thomas, 2018-06-22:
> It is specific to Opensplice and also specific to a 1 to 2 bridge.

Please try this with ros2/rmw_opensplice#234 applied.

tfoote, 2018-06-23:
I confirmed that the factory is not matching due to the types: 
```
Returning nullptr..... ros_id ros2, package_name example_interfaces, service_name Sample_AddTwoInts
```

And using ros2/rmw_opensplice#234 resolves this issue.

tfoote, 2018-06-23:
ros2/rmw_opensplice#234 merged

closed 2018-06-23 by tfoote