ros2/ros1_bridge #127: Failed to build bridge using melodic and bouncy

sloretz, 2018-07-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - bouncy
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
. /opt/ros/melodic/setup.bash
. /opt/ros/bouncy/local_setup.bash
colcon build --packages-select ros1_bridge --cmake-force-configure
```

#### Expected behavior

The ROS 1 bridge should succeed.

#### Actual behavior

```
developer@74e0cb101752:~/workspaces/bridge_ws$ . /opt/ros/melodic/setup.bash
developer@74e0cb101752:~/workspaces/bridge_ws$ . /opt/ros/bouncy/local_setup.bash
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
developer@74e0cb101752:~/workspaces/bridge_ws$ colcon build --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
--- stderr: ros1_bridge                         
Traceback (most recent call last):
  File "/home/developer/workspaces/bridge_ws/build/ros1_bridge/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/developer/workspaces/bridge_ws/build/ros1_bridge/devel/env.sh')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/environment_cache.py", line 62, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
  File "/usr/lib/python3.6/ast.py", line 48, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.6/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    ROS_DISTRO was set to 'bouncy' before. Please make sure that the environment does not mix paths from different distributions.
                 ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3
  "/home/developer/workspaces/bridge_ws/build/ros1_bridge/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/all.cmake:198 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:12 (find_package)
  /opt/ros/melodic/share/genmsg/cmake/genmsgConfig.cmake:197 (include)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgs-extras.cmake:2 (find_package)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgsConfig.cmake:197 (include)
  CMakeLists.txt:77 (find_package)

---
Failed   <<< ros1_bridge        [ Exited with code 1 ]

Summary: 0 packages finished [4.98s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

[... 82 lines cut ...]
developer@74e0cb101752:~/workspaces/bridge_ws$ . /opt/ros/bouncy/local_setup.bash 
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
developer@74e0cb101752:~/workspaces/bridge_ws$ echo $CMAKE_PREFIX_PATH
/opt/ros/melodic
developer@74e0cb101752:~/workspaces/bridge_ws$ unset CMAKE_PREFIX_PATH
developer@74e0cb101752:~/workspaces/bridge_ws$ colcon build --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
--- stderr: ros1_bridge                         
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find FastRTPS (missing: FastRTPS_INCLUDE_DIR FastRTPS_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /opt/ros/bouncy/share/fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake:95 (find_package_handle_standard_args)
  /opt/ros/bouncy/share/rmw_fastrtps_cpp/cmake/rmw_fastrtps_cpp-extras.cmake:20 (find_package)
  /opt/ros/bouncy/share/rmw_fastrtps_cpp/cmake/rmw_fastrtps_cppConfig.cmake:30 (include)
  /opt/ros/bouncy/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  CMakeLists.txt:208 (call_for_each_rmw_implementation)

---
Failed   <<< ros1_bridge        [ Exited with code 1 ]

Summary: 0 packages finished [2.57s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

dirk-thomas, 2018-07-06:
> The ROS_DISTRO warning appears to be the problem.

Please try to unset `ROS_DISTRO` between sourcing ROS 1 and ROS 2.

> Therefore the CMAKE_PREFIX_PATH must not contain paths from ROS 1 which would overlay ROS 2 packages.

This should better be phrased as: `The CMAKE_PREFIX_PATH must not have the ROS 1 paths prepended as they would overlay ROS 2 packages.`

The workspaces need to be sourced in the described order: first ROS 1, second ROS 2 (so that CMake finds ROS 2 packages in case of naming collisions). The `CMAKE_PREFIX_PATH` could be reset after sourcing ROS 1 since neither other ROS 1 CMake config files is needed - but keeping them (at the end of the environment variable) doesn't harm either.

sloretz, 2018-07-06:
Same error during build when unsetting `ROS_DISTRO` between sourcing ROS 1 and ROS 2

```
developer@74e0cb101752:~/workspaces/bridge_ws$ . /opt/ros/melodic/setup.bash
developer@74e0cb101752:~/workspaces/bridge_ws$ unset ROS_DISTRO
developer@74e0cb101752:~/workspaces/bridge_ws$ . /opt/ros/bouncy/local_setup.bash
developer@74e0cb101752:~/workspaces/bridge_ws$ colcon build --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
--- stderr: ros1_bridge                         
Traceback (most recent call last):
  File "/home/developer/workspaces/bridge_ws/build/ros1_bridge/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/developer/workspaces/bridge_ws/build/ros1_bridge/devel/env.sh')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/environment_cache.py", line 62, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
  File "/usr/lib/python3.6/ast.py", line 48, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.6/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    ROS_DISTRO was set to 'bouncy' before. Please make sure that the environment does not mix paths from different distributions.
                 ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3
  "/home/developer/workspaces/bridge_ws/build/ros1_bridge/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/all.cmake:198 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:12 (find_package)
  /opt/ros/melodic/share/genmsg/cmake/genmsgConfig.cmake:197 (include)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgs-extras.cmake:2 (find_package)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgsConfig.cmake:197 (include)
  CMakeLists.txt:77 (find_package)

---
Failed   <<< ros1_bridge        [ Exited with code 1 ]

Summary: 0 packages finished [5.05s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

If I also edit `/opt/ros/melodic/share/catkin/cmake/templates/env.sh.in` to contain `unset ROS_DISTRO` below the check for CLI arguments then I get a failure to find `FastRTPS`, just like I did when I cleared `CMAKE_PREFIX_PATH`. I'm starting to think the failure to find `FastRTPS` is a separate issue.

```
$ colcon build --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
--- stderr: ros1_bridge                         
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find FastRTPS (missing: FastRTPS_INCLUDE_DIR FastRTPS_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /opt/ros/bouncy/share/fastrtps_cmake_module/cmake/Modules/FindFastRTPS.cmake:95 (find_package_handle_standard_args)
  /opt/ros/bouncy/share/rmw_fastrtps_cpp/cmake/rmw_fastrtps_cpp-extras.cmake:20 (find_package)
  /opt/ros/bouncy/share/rmw_fastrtps_cpp/cmake/rmw_fastrtps_cppConfig.cmake:30 (include)
  /opt/ros/bouncy/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:61 (find_package)
  CMakeLists.txt:208 (call_for_each_rmw_implementation)

---
Failed   <<< ros1_bridge        [ Exited with code 1 ]
                                
Summary: 0 packages finished [1.94s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

> The CMAKE_PREFIX_PATH could be reset after sourcing ROS 1 since neither other ROS 1 CMake config files is needed - but keeping them (at the end of the environment variable) doesn't harm either.

I'm not sure what's happening yet. If the ROS 1 path in `CMAKE_PREFIX_PATH` is harmless, why do both clearing `CMAKE_PREFIX_PATH` and alternatively unsetting `ROS_DISTRO` in `env.sh.in` allow the the build to reach the point where `FastRTPS` cannot be found?

mikaelarguedas, 2018-07-06:
Does it work with ament_tools ?

dirk-thomas, 2018-07-06:
The warning message is there to warn "humans" - in this case this is what we want - mixing distributions. Maybe the [warning message](https://github.com/ros/ros_environment/blob/e1f2583934ddc52da57d7fba732ff4a67f9a1745/env-hooks/1.ros_distro.sh.in#L4) should be printed to `stderr` and when being invoked only `stdout` should be considered. That should fix the syntax error.

> ... the point where FastRTPS cannot be found

Maybe related to this question: https://answers.ros.org/question/296462/compilation-error-building-against-binary-bouncy-could-not-find-fastrtps/

mikaelarguedas, 2018-07-06:
> > ... the point where FastRTPS cannot be found

> Maybe related to this question: https://answers.ros.org/question/296462/compilation-error-building-against-binary-bouncy-could-not-find-fastrtps/

Could be related to this as well https://github.com/colcon/colcon-ros/pull/21

I tried based on one of our docker images and I can successfully build the bridge with ament_tools but not with colcon. I don;t get the same error though but figured I would post it here anyway.

List of commands I ran:

<details>

```
docker run -it --rm osrf/ros2:bouncy-basic
apt update && apt upgrade -y && apt install -y git ros-bouncy-ament-tools python3-colcon-common-extensions python3-rosdep
apt install -y   ros-melodic-actionlib-msgs ros-melodic-catkin ros-melodic-class-loader ros-melodic-cpp-common ros-melodic-diagnostic-msgs ros-melodic-gencpp ros-melodic-geneus ros-melodic-genlisp ros-melodic-genmsg ros-melodic-gennodejs ros-melodic-genpy ros-melodic-geometry-msgs ros-melodic-message-generation ros-melodic-message-runtime ros-melodic-nav-msgs ros-melodic-pluginlib ros-melodic-ros-environment ros-melodic-rosbag ros-melodic-rosbag-migration-rule ros-melodic-rosbag-storage ros-melodic-rosbuild ros-melodic-rosclean ros-melodic-rosconsole ros-melodic-roscpp-serialization ros-melodic-roscpp-traits ros-melodic-rosgraph ros-melodic-rosgraph-msgs ros-melodic-roslib ros-melodic-roslz4 ros-melodic-rosmaster ros-melodic-rosmsg ros-melodic-rosout ros-melodic-rospack ros-melodic-rosparam ros-melodic-rospy ros-melodic-rostime ros-melodic-rosunit ros-melodic-sensor-msgs ros-melodic-shape-msgs ros-melodic-std-msgs ros-melodic-std-srvs ros-melodic-stereo-msgs ros-melodic-topic-tools ros-melodic-trajectory-msgs ros-melodic-visualization-msgs ros-melodic-xmlrpcpp
source /opt/ros/bouncy/setup.bash 
source /opt/ros/melodic/setup.bash 
mkdir -p ~/ros2_ws/src && cd ~/ros2_ws/src
git clone https://github.com/ros2/ros1_bridge.git && cd ~/ros2_ws
export ROSDISTRO_INDEX_URL='https://raw.githubusercontent.com/ros2/rosdistro/ros2/index.yaml'
rosdep init
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro bouncy -y
```

Build with ament_tools:
```
rm -rf build install
ament build --symlink-install --only ros1_bridge
```

Build with colcon:
```
rm -rf build install
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
```
Colcon console output
```
Starting >>> ros1_bridge
-- The C compiler identification is GNU 7.3.0   
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done         
-- Detecting C compile features
-- Detecting C compile features - done          
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done       
-- Detecting CXX compile features
-- Detecting CXX compile features - done        
-- Found rmw: 0.5.0 (/opt/ros/bouncy/share/rmw/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.5", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rclcpp: 0.5.1 (/opt/ros/bouncy/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.5.0 (/opt/ros/bouncy/share/rmw_implementation_cmake/cmake)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'roscpp'
--   Found roscpp, version 1.14.2
-- Checking for module 'std_msgs'               
--   Found std_msgs, version 0.5.11
-- Checking for module 'actionlib_msgs'         
--   Found actionlib_msgs, version 1.12.6
-- Checking for module 'diagnostic_msgs'        
--   Found diagnostic_msgs, version 1.12.6
-- Checking for module 'geometry_msgs'
[... 31 lines cut ...]
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON              
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /root/ros2_ws/build/ros1_bridge/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.5") 
-- Looking for pthread.h
-- Looking for pthread.h - found                
-- Looking for pthread_create
-- Looking for pthread_create - not found       
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests
-- catkin 0.7.14                                
Failed   <<< ros1_bridge        [ Exited with code -11 ]
```
</details>

Looking into the colcon failure now.

sloretz, 2018-07-06:
>Does it work with ament_tools ?

Yes, a build with `ament build` succeeds in the above Dockerfile. No need to unset `CMAKE_PREFIX_PATH`, and no linker errors.

```
developer@b766ee2aeb4d:~/bridge_ws$ . /opt/ros/melodic/setup.bash 
developer@b766ee2aeb4d:~/bridge_ws$ . /opt/ros/bouncy/setup.bash 
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributio
ns.
developer@b766ee2aeb4d:~/bridge_ws$ ament build
# Topological order
 - ros1_bridge

# .... a lot of output, the build succeeds
```

dirk-thomas, 2018-07-06:
colcon/colcon-ros#23 should address the problem with the `CMAKE_PREFIX_PATH`.

dirk-thomas, 2018-07-06:
This should be fixed with the latest release of `colcon-ros` 0.2.9.

sloretz, 2018-07-06:
Thanks @dirk-thomas and @mikaelarguedas ! Closing.

closed 2018-07-06 by sloretz