ros2/ros1_bridge #131: ros1_bridge crashes when a rospy ROS 1 subscriber that is subscribing to a ROS 2 topic disconnects when bridge is not using `--bridge-all-topics`

dljsjr, 2018-07-31:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 16.04
- Installation type:
Binaries
- Version or commit hash:
Ardent
- DDS implementation:
Fast-RTPS
- Client library (if applicable):
N/A

#### Steps to reproduce issue
Create a ROS 2 Publisher, start this publisher, then start the `ros1_bridge dynamic_bridge` without any bridging arguments.

Create a ROS 1 python script using rospy, and subscribe to this topic. Then either call `unregister()` or simply terminate the node.

#### Expected behavior
The ROS 1 node should shut down and the bridge should be unaffected.

#### Actual behavior
The bridge exits, sometimes with no printout but often with a `what()` printout indicating an Illegal Argument.

#### Additional information
If `--bridge-all-topics` is passed to the bridge, it does not crash when the subscriber exits.

We discovered this while working on #130 and #129. I'm hoping to be able to provide more information for debugging once I get a debug build of our bridge (as indicated in #130) up and running, which should be this week.

dljsjr, 2018-08-03:
I was able to create some python scripts that doing talking and listening on either end of the bridge and I'm not able to repro the issue with simple self contained scripts. I'm working on a test environment that uses our full controller and API in simulation plus a version of the bridge with debug symbols to see if I can repro that way.

dljsjr, 2018-08-06:
@dirk-thomas @mikaelarguedas I'm not able to reproduce this issue using simulations with our full controller API. It seems to be endemic to the physical robot.

I can get you whatever information you need to help debug this, if you can let me know what you think is a good place to start.

dirk-thomas, 2018-08-06:
If we can't reproduce the problem we would need at least a stacktrace when the problem happens. Hopefully from that we can conclude what the problem might be.

dljsjr, 2018-08-06:
Sure thing, that'll be the first thing I try is running the bridge with GDB.

dirk-thomas, 2019-03-01:
Closing due to no response. Please feel free to comment on the closed ticket with the requested information and the issue can be reopened if necessary.

closed 2019-03-01 by dirk-thomas