ros2/ros1_bridge #133: rviz2 tf message drop

llllIIIIllll, 2018-08-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - release-latest/ros2.repos
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

## Step 1:
### terimal 1:

setup ros1 environment(kinetic) and run 
```bash
source <path-to-ros1>/setup.bash 
roscore
```
### terminal 2:

build [rotors_simulator](https://github.com/ethz-asl/rotors_simulator) with ros1
```bash
cd ~/catkin_ws/src
git clone https://github.com/ethz-asl/mav_comm
git clone https://github.com/ethz-asl/rotors_simulator
touch ~/catkin_ws/src/rotors_simulator/rotors_hil_interface/CATKIN_IGNORE
cd ~/catkin_ws && catkin_make
```
run the `mav_with_joy.launch`
```
source ~/catkin_ws/devel/setup.bash
roslaunch rotors_gazebo mav_with_joy.launch 
```
it will start gazebo, push the start button at gazebo

## Step 2:
### terminal 3
make the drone flying
```bash
source ~/catkin_ws/devel/setup.bash
rostopic pub /firefly/joy sensor_msgs/Joy "header: 
  seq: 0                                           
  stamp:                                             
    secs: 0
    nsecs: 0
  frame_id: ''
axes:
- 0
buttons:
- 3" 
```
## Step 3:
### terminal 4:
run the ros1_bridge 
```bash
source <path-to-ros1>/setup.bash 
source <path-to-ros2>/local_setup.bash 
export ROS_MASTER_URI=http://localhost:11311
ros2 run ros1_bridge dynamic_bridge
```
### terminal 5:
start rviz and rviz2
```bash
source <path-to-ros2>/local_setup.bash 
rviz2
```
add the tf in rviz2
### terminal 6:
```bash
source <path-to-ros1>/setup.bash 
rviz
```
add the tf in rviz

#### Expected behavior
the tf frame should be same smooth between rviz and rviz2

#### Actual behavior
rviz2 is obvious lag and these are some tf messages lost

#### Additional information
If whe add up the `Frame rate` in rviz2 the display will more smooth

the `ros2 run tf2_ros tf2_monitor`
information is 
```
All Broadcasters:
Node: <no authority available> 260.361 Hz, Average Delay: 1.53372e+09 Max Delay: 1.53372e+09

```
mem (7.7 GiB)

cpu (Intel® Core™ i7-8550U CPU @ 1.80GHz × 8 ) information
```
29750 wbl       20   0 4235388 366848 144508 R 105.3  4.6  30:30.85 gzclient                                                                             
29788 wbl       20   0 3563996 151204  82460 S  37.2  1.9  10:09.64 gzserver                                                                             
 1007 root      20   0  564680  87252  55464 S  11.6  1.1  11:14.80 Xorg                                                                                 
17930 wbl       20   0 1050108  87516  78152 S  11.0  1.1   0:21.32 dynamic_bridge                                                                       
25865 wbl       20   0 2113812 130572  99360 S  11.0  1.6   0:09.23 rviz                                                                                 
16677 wbl       20   0 1408952 132244 102928 S  10.3  1.6   0:21.86 rviz2                                                                                
 1906 wbl       20   0 1598140 149496  57432 S   6.6  1.9   8:05.13 compiz                                                                               
29602 wbl       20   0  881344  88116  34936 S   2.7  1.1   0:43.62 python                                                                               
 2120 wbl       20   0  644320  42284  15164 S   2.3  0.5   2:25.98 gnome-terminal-                                                                      
29503 wbl       20   0  617432  53792   7344 S   2.3  0.7   0:23.69 rosmaster                                                                            
29586 wbl       20   0  462444  14940  13212 S   2.3  0.2   0:36.95 robot_state_pub                                                                      
 1034 root     -51   0       0      0      0 S   1.7  0.0   1:44.17 irq/136-nvidia                                                                       
 1751 wbl       20   0 1316128 657092   8404 S   1.3  8.2  18:50.75 hud-service                                                                          
29568 wbl       20   0  346236  11832  10872 S   1.3  0.1   0:25.30 roll_pitch_yawr                                                                      
29516 wbl       20   0  399680  10108   9204 S   0.7  0.1   0:11.72 rosout                                                                               
29551 wbl       20   0  399736   9832   8932 S   0.7  0.1   0:11.05 joy_node                                                                             
29552 wbl       20   0  408040   9956   9052 S   0.7  0.1   0:11.20 rotors_joy_inte                                                                      
    8 root      20   0       0      0      0 I   0.3  0.0   0:08.23 rcu_sched                                                                            
 1775 wbl       20   0  659660  31372  15064 S   0.3  0.4   0:42.02 unity-panel-ser                                                                      
 2137 wbl       20   0  977208  97308  44136 S   0.3  1.2   0:18.31 fcitx-qimpanel                                                                       
23480 wbl       20   0   49568   3792   3064 R   0.3  0.0   0:00.16 top                                                                                  
    1 root      20   0  185704   4492   3168 S   0.0  0.1   0:01.98 systemd                                                                              
    2 root      20   0       0      0      0 S   0.0  0.0   0:00.01 kthreadd                                                                             
    4 root       0 -20       0      0      0 I   0.0  0.0   0:00.00 kworker/0:0H                                                                         
    6 root       0 -20       0      0      0 I   0.0  0.0   0:00.00 mm_percpu_wq                                                                         
    7 root      20   0       0      0      0 S   0.0  0.0   0:00.85 ksoftirqd/0                                                                          
    9 root      20   0       0      0      0 I   0.0  0.0   0:00.00 rcu_bh                                                                               
   10 root      rt   0       0      0      0 S   0.0  0.0   0:00.08 migration/0                                                                          
   11 root      rt   0       0      0      0 S   0.0  0.0   0:00.02 watchdog/0                                                                           
   12 root      20   0       0      0      0 S   0.0  0.0   0:00.00 cpuhp/0                                                                              
   13 root      20   0       0      0      0 S   0.0  0.0   0:00.00 cpuhp/1                                                                              
   14 root      rt   0       0      0      0 S   0.0  0.0   0:00.02 watchdog/1                                                                           
   15 root      rt   0       0      0      0 S   0.0  0.0   0:00.08 migration/1                                                                          
   16 root      20   0       0      0      0 S   0.0  0.0   0:00.88 ksoftirqd/1                                                                          
   18 root       0 -20       0      0      0 I   0.0  0.0   0:00.00 kworker/1:0H                                                                         
   19 root      20   0       0      0      0 S   0.0  0.0   0:00.00 cpuhp/2 
```

----
## Feature request

#### Feature description

#### Implementation considerations

calvertdw, 2018-08-08:
Have you compared the performance to OpenSplice? (`RMW_IMPLEMENTATION=rmw_opensplice_cpp`)

Are you using BEST_EFFORT or RELIABLE QoS settings for the tf in rviz2?

Since you are building from source, you might have to rebuild to use OpenSplice, following instructions at https://github.com/ros2/ros2/wiki/Linux-Development-Setup#install-more-dds-implementations-optional

llllIIIIllll, 2018-08-09:
I use the default Qos settings in rviz2 `RMW_QOS_POLICY_RELIABILITY_RELIABLE`, I have tried change it, but the problem still exist. 
I also try the opensplice, it has the same problem.

sloretz, 2018-09-14:
This looks like it is an issue higher level than the bridge, possible tf2 or RViz2.

When I run `ros2 topic echo /tf tf2_msgs/TFMessage` and `rostopic echo /tf` in side-by-side terminals the timestamps appear to be the same. If I kill and restart the transform publisher the last transform displayed by each is usually the same message (there is a delay of about 0.5 seconds between starting the ros 1 tf publisher and receiving a message on the ROS 2 side). Further tf2_monitor shows very little delay

```
$ ros2 run tf2_ros  tf2_monitor 

RESULTS: for all Frames

Frames:
Frame: base_link, published by <no authority available>, Average Delay: 0.00051274, Max Delay: 0.00609875

All Broadcasters:
Node: <no authority available> 100.101 Hz, Average Delay: 0.00051274 Max Delay: 0.00609875
```

sloretz, 2018-09-17:
@llllIIIIllll I believe the issue is resolved by the referenced PRs. Would you mind checking using your setup with the latest code?

closed 2018-09-17 by sloretz