ros2/ros1_bridge #134: Support field mapping for packages/messages having different names?

mikaelarguedas, 2018-08-10:
Currently the docs example for field mapping rules gives the following example:
```
-
  ros1_package_name: 'pkg_name'
  ros1_message_name: 'msg_name'
  ros2_package_name: 'pkg_name'
  ros2_message_name: 'msg_name'
  fields_1_to_2:
    foo: 'foo'
    ros1_bar: 'ros2_bar'
```

Is it intentional that we do not support defining mapping with either different pkg_name, different message names or both ?

In other words should the following be valid?

```
-
  ros1_package_name: 'ros1_pkg_name'
  ros1_message_name: 'ros1_msg_name'
  ros2_package_name: 'ros2_pkg_name'
  ros2_message_name: 'ros2_msg_name'
  fields_1_to_2:
    foo: 'foo'
    ros1_bar: 'ros2_bar'
```

dirk-thomas, 2018-08-10:
> In other words should the following be valid?

I would expect any kind of mapping to work.

mikaelarguedas, 2018-08-10:
Ok I'll label this as a bug and look into it then :+1:

dirk-thomas, 2018-08-10:
Is there any report that it doesn't work?

mikaelarguedas, 2018-08-10:
Yeah there was a [question on answers](https://answers.ros.org/question/298876/ros1_bridge-custom-message-pairing-failed-if-those-messages-have-non-identical-names-in-ros1-and-ros2/) that I've been looking into and narrowed it down to that scenario. I asked the question here to be able to refer to it.

Now that I know it should be supported I'll provide a PR as well as example packages to reproduce

mikaelarguedas, 2018-08-10:
addressed in #135

closed 2018-08-10 by mikaelarguedas