ros2/ros1_bridge #139: No executables in ros1_bridge

m4xst3r, 2018-10-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - from source
- Version or commit hash:
  - https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
- DDS implementation:
  - None
- Client library (if applicable):
  - ros1_bridge

#### Steps to reproduce issue

```
1.: . <install-space-with-ros2>/local_setup.bash

2.: colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure

3.: # Shell A:
. /opt/ros/melodic/setup.bash
roscore

4.: # Shell B:
. /opt/ros/melodic/setup.bash
. <install-space-with-bridge>/setup.bash
export ROS_MASTER_URI=http://localhost:11311
ros2 run ros1_bridge dynamic_bridge
```

#### Expected behavior
Should start the executable dynamic_bridge

#### Actual behavior
No executable found

#### Additional information

---- The colcon build works fine without any Errors or Warnings.
No response of the command: ros2 pkg executables | grep ros1_bridge

dirk-thomas, 2018-10-16:
In the steps to reproduce you didn't mention sourcing ROS 1 when building the `ros1_bridge` package. In that case the bridge simply wouldn't be built.

You could try the latest state from master of this repository. Then when missing to source ROS 1 you will see a warning message.

Please double check to follow the instructions exactly in order to also source ROS 1.

dirk-thomas, 2018-10-22:
I will go ahead and close the ticket for now due to no response.

closed 2018-10-22 by dirk-thomas