ros2/ros1_bridge #148: Error building ros1_bridge latest upstream master

TSC21, 2018-11-28:
## Bug report

Trying to build `ros1_bridge` from source results on a compile error (using `colcon build`).

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS
- ROS versions installed:
  - ROS2 Bouncy and ROS Melodic

#### Steps to reproduce issue

- Clone the `ros1_bridge` into a workspace
- Sourcing first ROS1 and then ROS2 env `setup.bash`
- Build using `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+`

#### Expected behavior

Successful build of the `ros1_bridge` package.

#### Actual behavior

Fails with the following errors:
```
In file included from /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/generated/get_factory.cpp:3:0:
/home/nuno/PX4/px4_ros_com_ros2/src/ros1_bridge/include/ros1_bridge/factory.hpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::ros2_callback(typename ROS2_T::SharedPtr, const rmw_message_info_t&, ros::Publisher, const string&, const string&, rclcpp::PublisherBase::SharedPtr)’:
/home/nuno/PX4/px4_ros_com_ros2/src/ros1_bridge/include/ros1_bridge/factory.hpp:166:85: error: request for member ‘str’ in ‘rcutils_get_error_string()’, which is of non-class type ‘const char*’
         auto msg = std::string("Failed to compare gids: ") + rmw_get_error_string().str;
                                                                                     ^~~
In file included from /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/generated/diagnostic_msgs_factories.hpp:3:0,
                 from /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/generated/diagnostic_msgs_factories.cpp:4:
/home/nuno/PX4/px4_ros_com_ros2/src/ros1_bridge/include/ros1_bridge/factory.hpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::ros2_callback(typename ROS2_T::SharedPtr, const rmw_message_info_t&, ros::Publisher, const string&, const string&, rclcpp::PublisherBase::SharedPtr)’:
/home/nuno/PX4/px4_ros_com_ros2/src/ros1_bridge/include/ros1_bridge/factory.hpp:166:85: error: request for member ‘str’ in ‘rcutils_get_error_string()’, which is of non-class type ‘const char*’
         auto msg = std::string("Failed to compare gids: ") + rmw_get_error_string().str;
                                                                                     ^~~
In file included from /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/generated/lifecycle_msgs_factories.hpp:3:0,
                 from /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/generated/lifecycle_msgs_factories.cpp:4:
/home/nuno/PX4/px4_ros_com_ros2/src/ros1_bridge/include/ros1_bridge/factory.hpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::ros2_callback(typename ROS2_T::SharedPtr, const rmw_message_info_t&, ros::Publisher, const string&, const string&, rclcpp::PublisherBase::SharedPtr)’:
/home/nuno/PX4/px4_ros_com_ros2/src/ros1_bridge/include/ros1_bridge/factory.hpp:166:85: error: request for member ‘str’ in ‘rcutils_get_error_string()’, which is of non-class type ‘const char*’
         auto msg = std::string("Failed to compare gids: ") + rmw_get_error_string().str;
                                                                                     ^~~
```

Thanks in advance!

TSC21, 2018-11-28:
Seems to be a related issue: https://github.com/ros2/ros1_bridge/issues/142. Last time I tried to build from source was around one month ago and I was able to build it successfully.

clalancette, 2018-11-28:
The problem in this case (and likely all of the cases you link to) is that you are trying to build ros1_bridge master against Bouncy.  We've changed API in-between Bouncy and Crystal, so this isn't going to work (for a number of other packages as well).  Your options here are:

1.  Build *everything* from master, following the instructions at https://index.ros.org/doc/ros2/Linux-Development-Setup/ .  I'd only recommend this if you know you need a new feature or fix from ros1_bridge.
2.  Make sure to check out the bouncy branch on the ros1_bridge repository, and build that (which should succeed when building against the rest of Bouncy).

I'm going to close this out for now, but feel free to reopen if that doesn't solve your problem.

closed 2018-11-28 by clalancette

TSC21, 2018-11-28:
Checking out the `bouncy` branch works. I just wanted to understand why does a small change have to break downward compatibility with Bouncy or Ardent. Architecturally speaking, it doesn't seem a scalable solution, unless there are other features on the current upstream code that are not compatible with versions bellow Crystal.