ros2/ros1_bridge #159: failed to create 2to1 bridge for topic '/rosout' with ROS 2 type

qingmingjie, 2019-01-16:
failed to create 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/Log' and ROS 1 type 'rosgraph_msgs/Log': No template specialization for the pair

dirk-thomas, 2019-01-18:
@nburek Since you added the `rcl_interfaces/Log` message in ros2/rcl_interfaces#53 can you maybe provide a pull request to also add a custom mapping file for it?

nburek, 2019-01-18:
Yes, I've added it to my team's backlog and we will schedule someone to start working on it next week. 

In the mean time, if you add the command line argument `__log_disable_rosout:=true` to the command line arguments when using `ros2 run` it will disable the rosout feature and the nodes should not create  the topic. This may be hard to verify with the command line tools as they spin up their own node to communicate with the service. I'm not sure if there is an easy way to include the disable flag when running that.

juanrh, 2019-01-25:
Hi @dirk-thomas, I'm taking a look to this. I'm trying to use a mapping file as suggested, but I have the problem that for obtaining a value for the field `stamp` of [`rcl_interfaces/Log`](https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/Log.msg) from a [`rosgraph_msgs/Log`](https://docs.ros.org/kinetic/api/rosgraph_msgs/html/msg/Log.html) message I need to access the value `stamp` nested in the field `header` with type [`std_msgs/Header`](https://docs.ros.org/kinetic/api/std_msgs/html/msg/Header.html). So I'd have to use a mapping file like:

```yaml
-
  ros1_package_name: 'rosgraph_msgs'
  ros1_message_name: 'Log'
  ros2_package_name: 'rcl_interfaces'
  ros2_message_name: 'Log'
  fields_1_to_2:
    header.stamp: 'stamp'
    level: 'level'
    name: 'name'
    msg: 'msg'
    file: 'file'
    function: 'function'
    line: 'line'
```

which is not valid because as far as I know projections cannot be specified in the mapping rule files. So this file is ignored during the build of `ros1_bridge install` with a message "A manual mapping refers to an invalid field 'header.stamp' in the ROS 1 message 'rosgraph_msgs/Log'". 

Is there a way to express these kind of projections for mapping files? Do you think I should do this with C++ instead, by adding another factory to [`builtin_interfaces_factories.hpp`](https://github.com/ros2/ros1_bridge/blob/master/include/ros1_bridge/builtin_interfaces_factories.hpp) instead? Any idea would be much appreciated.

Thanks

juanrh, 2019-01-31:
> I think it would be best to update the Python module to support specifying nested fields.

@dirk-thomas I've been taking a look to that, but I think that requires a significant refactor of the Python code and of the .em templates. The Python class `Mapping` should be changed to support tuples of Field in the keys for `fields_1_to_2` and values for `fields_2_to_1`. Also, in `determine_field_mapping` we only have the package name for `ros1_msg`, but not for it's nested fields, that can be in a different package like happends for `std_msgs/Header header` on `rosgraph_msgs/Log`. Finally, it would probably make sense to add support for selecting fields in ros 2 messages too. 

So I continued working on adding a custom mapping just for `rosgraph_msgs::Log` to `rcl_interfaces::msg::Log`. I think I'm close to completing that, but I'm having problems with serialization and deserialization of `rcl_interfaces::msg::Log` messages, I opened https://github.com/ros2/rcl_interfaces/issues/61 for the deserialization issue.

dirk-thomas, 2019-01-31:
> in `determine_field_mapping` we only have the package name for `ros1_msg`, but not for it's nested fields, that can be in a different package like happends for `std_msgs/Header header` on `rosgraph_msgs/Log`.

The mapping rule in yaml only needs to specify the package name and message name in both ROS versions. From there all types of (nested) fields can be determined by just parsing the corresponding message definition. Making the necessary information available in the various locations in the code would be part of the refactoring.

> it would probably make sense to add support for selecting fields in ros 2 messages too.

Absolutely.

juanrh, 2019-02-01:
Ok, I’ll then work on extending the code generation to support subfield selection. Thanks for the advice

juanrh, 2019-03-16:
Hi, 

@dirk-thomas  @gonzodepedro I have a draft for this on https://github.com/aws-robotics/ros1_bridge/commits/issue-159-py, so for only adds selection to ROS 1 fields, but it would be easy to add it to ROS 2 fields too. However I'm having problems testing this, I cannot use FastRTPS due to https://github.com/ros2/rcl_interfaces/issues/61, and I wasn't able to build this for OpenSplice or RTI Connext. For example for RTI Connext I get the following error when building the bridge on a container for `osrf/ros2:nightly`

```
source /opt/rti.com/rti_connext_dds-5.3.1/setenv_ros2rti.bash
source /opt/ros/melodic/setup.bash
source /opt/ros/crystal/setup.bash
rm -rf build/ros1_bridge install/ros1_bridge/
colcon build --symlink-install --packages-up-to  ros1_bridge --cmake-force-configure
...
Finished <<< rosidl_typesupport_cpp [1.21s]                                                                          
--- stderr: rosidl_generator_py                             
CMake Error at cmake/rosidl_generator_py_generate_interfaces.cmake:317 (add_dependencies):
  The dependency target
  "rosidl_generator_py_custom__rosidl_typesupport_opensplice_c" of target
  "rosidl_generator_py__rosidl_typesupport_opensplice_c__pyext" does not
  exist.
Call Stack (most recent call first):
  /opt/workspace/install/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:38 (include)
  /opt/workspace/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:280 (ament_execute_extensions)
  CMakeLists.txt:68 (rosidl_generate_interfaces)

---
Failed   <<< rosidl_generator_py	[ Exited with code 1 ]

Summary: 86 packages finished [43.7s]
  1 package failed: rosidl_generator_py
  4 packages had stderr output: rmw_opensplice_cpp rosidl_generator_py rosidl_typesupport_opensplice_c rosidl_typesupport_opensplice_cpp
  37 packages not processed
root@ub80c134adc2254:/opt/workspace# 
```

Any help would be much appreciated

gonzodepedro, 2019-03-20:
To circumvent ros2/rcl_interfaces#61 try running with the _log_disable_rosout:=true flag. 
`ros2 run ros2 run demo_nodes_cpp listener  _log_disable_rosout:=true`

To use RTI Coonext follow steps [here](https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/#install-more-dds-implementations-optional). 
Basic steps should be: 

1. Install DDS Connext
2. Run `cd /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts && source ./rtisetenv_x64Linux3gcc5.4.0.bash; cd -
3. Rebuild workspace
4. Set RMW_IMPLEMENTATION environment variable: `RMW_IMPLEMENTATION=rmw_connext_cpp`
5. Run node as usual
`

dirk-thomas, 2019-03-20:
> I have a draft for this on https://github.com/aws-robotics/ros1_bridge/commits/issue-159-py

Please create a pull request (possible marked as WIP) for the proposed changes.

gonzodepedro, 2019-03-21:
@juanrh 
Try the diff below and then run with __log_disable_rosout:=true 
`ros2 run ros1_bridge dynamic_bridge __log_disable_rosout:=true`

index 848188e..f6dffe8 100644
--- a/src/dynamic_bridge.cpp
+++ b/src/dynamic_bridge.cpp
@@ -455,14 +455,14 @@ int main(int argc, char * argv[])
     return 0;
   }
 
-  // ROS 1 node
-  ros::init(argc, argv, "ros_bridge");
-  ros::NodeHandle ros1_node;
-
   // ROS 2 node
   rclcpp::init(argc, argv);
   auto ros2_node = rclcpp::Node::make_shared("ros_bridge");
 
+  // ROS 1 node
+  ros::init(argc, argv, "ros_bridge");
+  ros::NodeHandle ros1_node;
+
   // mapping of available topic names to type names
   std::map<std::string, std::string> ros1_publishers;
   std::map<std::string, std::string> ros1_subscribers;

gonzodepedro, 2019-03-22:
The diff proposed above is part of PR #178

juanrh, 2019-03-26:
I updated PR #174 to support field selection for both ROS 1 and ROS 2. I have also created https://github.com/ros2/rcl_interfaces/pull/67 for the mapping file between ROS 1 and ROS 2 log messages

mlanting, 2019-04-02:
I gave this a try yesterday, and while the error messages about failing to create a bridge for the /rosout topic were gone, the bridge was crashing after trying to pass the first message:
`~/ros2_ws$ ros2 run ros1_bridge dynamic_bridge 
created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/Log' and ROS 1 type 'rosgraph_msgs/Log'
created 1to2 bridge for topic '/chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/String'
[INFO] [ros_bridge]: Passing message from ROS 1  �C to ROS 2  �C (showing msg only once per type)
realloc(): invalid pointer`

I was attempting to run the tutorial here: https://github.com/ros2/ros1_bridge/blob/master/README.md

I'm also getting garbled output in the [INFO] print messages, but that happens whether or not I run with @juanrh 's changes, so I will file a new issue about that.

mlanting, 2019-04-04:
Running with gdb, I can get the following stack trace after it crashes:

`created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/Log' and ROS 1 type 'rosgraph_msgs/Log'
created 1to2 bridge for topic '/chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/String'
[INFO] [ros_bridge]: Passing message from ROS 1 ����� to ROS 2 ����� (showing msg only once per type)
realloc(): invalid pointer

Thread 1 "dynamic_bridge" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff4b7c801 in __GI_abort () at abort.c:79
#2  0x00007ffff4bc5897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff4cf2b9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff4bcc90a in malloc_printerr (str=str@entry=0x7ffff4cf0efd "realloc(): invalid pointer") at malloc.c:5350
#4  0x00007ffff4bd4eba in __GI___libc_realloc (oldmem=0x5555565c9630, bytes=11) at malloc.c:3174
#5  0x00007ffff3b03e62 in rosidl_generator_c__String__assignn (str=0x5555565c9620, value=0x7fffffff9d50 "ros_bridge", n=10)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rosidl/rosidl_generator_c/src/string_functions.c:85
#6  0x00007fffe85d84fd in __cdr_deserialize (cdr=..., untyped_ros_message=0x5555565c9610)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/build/rcl_interfaces/rosidl_typesupport_fastrtps_c/rcl_interfaces/msg/dds_fastrtps_c/log__type_support_c.cpp:189
#7  0x00007fffe983caac in rmw_fastrtps_cpp::TypeSupport::deserializeROSmessage (this=0x555555d53f60, deser=..., ros_message=0x5555565c9610)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/type_support_common.cpp:88
#8  0x00007fffe920fe92 in rmw_fastrtps_shared_cpp::TypeSupport::deserialize (this=0x555555d53f60, payload=<optimized out>, data=0x7fffffffa140)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/TypeSupport_impl.cpp:100
#9  0x00007fffe8da4d90 in eprosima::fastrtps::SubscriberHistory::takeNextData (this=0x555556524560, data=data@entry=0x7fffffffa140, info=info@entry=0x7fffffffa150)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/eProsima/Fast-RTPS/src/cpp/subscriber/SubscriberHistory.cpp:429
#10 0x00007fffe8d9c4ec in eprosima::fastrtps::SubscriberImpl::takeNextData (this=<optimized out>, data=data@entry=0x7fffffffa140, info=info@entry=0x7fffffffa150)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/eProsima/Fast-RTPS/src/cpp/subscriber/SubscriberImpl.cpp:88
#11 0x00007fffe8d9c2f9 in eprosima::fastrtps::Subscriber::takeNextData (this=<optimized out>, data=data@entry=0x7fffffffa140, info=info@entry=0x7fffffffa150)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/eProsima/Fast-RTPS/src/cpp/subscriber/Subscriber.cpp:42
#12 0x00007fffe920d7ca in rmw_fastrtps_shared_cpp::_take (identifier=<optimized out>, subscription=<optimized out>, ros_message=<optimized out>, taken=0x7fffffffa21f, message_info=0x7fffffffb6d0)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_take.cpp:69
#13 0x00007ffff78944bf in rcl_take (subscription=0x555555d53760, ros_message=ros_message@entry=0x5555565c9610, message_info=message_info@entry=0x7fffffffb6d0)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rcl/rcl/src/rcl/subscription.c:251
#14 0x00007ffff7b290c2 in rclcpp::executor::Executor::execute_subscription (subscription=std::shared_ptr<rclcpp::SubscriptionBase> (use count 3, weak count 1) = {...})
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:324
#15 0x00007ffff7b2a1b5 in rclcpp::executor::Executor::execute_any_executable (this=this@entry=0x7fffffffcbd0, any_exec=...)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:274
#16 0x00007ffff7b2db6b in rclcpp::executor::Executor::spin_once (this=0x7fffffffcbd0, timeout=...) at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:242
#17 0x00007ffff7b28691 in rclcpp::executor::Executor::spin_node_once_nanoseconds (this=0x7fffffffcbd0, 
    node=std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface> (use count 6, weak count 1) = {...}, timeout=...)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:185
#18 0x000055555555ba1a in rclcpp::executor::Executor::spin_node_once<rclcpp::Node, std::ratio<1l, 1000l> > (timeout=..., node=..., this=0x7fffffffcbd0)
    at /home/mlanting/workspaces/ros1_bridge_pr_ws/install/include/rclcpp/executor.hpp:176
#19 main (argc=<optimized out>, argv=<optimized out>) at /home/mlanting/workspaces/ros1_bridge_pr_ws/src/ros2/ros1_bridge/src/dynamic_bridge.cpp:765'`

If I run with using opensplice instead of fastRTPS, I do not get the crash and the bridge seem to work as it should for the tutorial demo.

gonzodepedro, 2019-04-04:
Bug report created on ros2/rmw_fastrtps#265

rubenanapu, 2019-04-16:
Does anybody know whether there is any workaround for this issue? 

I've tried using ` __log_disable_rosout:=true` but the problem didn't go away.

I compiled `ros1_bridge` from source on tag `0.6.1` (because I had an old installation with that version that works) but the problem is still there.

Since the problem seemed to be introduced in ros2/rcl_interfaces#53 I've compiled that package from source as well (on `tags/0.6.1`) but the problem didn't go away.

Did anybody here managed to make it work? If so, how?

Thanks in advance for any help.

mlanting, 2019-04-16:
@rubenanapu As far as I can tell, everything seems to work fine, apart from that one topic. If you can tolerate the message spam I don't think it will stop you from using the bridge.

rubenanapu, 2019-04-16:
Hm, I'll try just to redirect the error message then for the moment.

Thanks @mlanting.

closed 2019-05-16 by dirk-thomas

nburek, 2019-08-08:
For anyone who is still running into this issue when they use launch files instead of `ros2 run`, here is how you can apply the same work around. 

When using ros2 run:
`ros2 run ros1_bridge dynamic_bridge __log_disable_rosout:=true`

In your launch file when using ros2 launch:
```
launch_ros.actions.Node(package='ros1_bridge`,
  node_executable='dynamic_bridge',
  arguments=['__log_disable_rosout:=true'])
```

serge-nikulin, 2019-10-22:
@dirk-thomas, I see this issue in melodic/dashing when I rebuild ros1_bridge following instructions from https://github.com/ros2/ros1_bridge. 

See the `--print-pairs` output at the end of the message.

```
$ ros2 run ros1_bridge dynamic_bridge --bridge-all-topics --show-introspection
...
...
failed to create 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option
  ROS 1: /rosout (rosgraph_msgs/Log) [0 pubs, >0 subs]
  ROS 1: /rosout_agg (rosgraph_msgs/Log) [>0 pubs, 0 subs]

failed to create 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option
  ROS 2: /TPCK_DEMO_NODE_PUB_MSG (safeai_interfaces/msg/Pub) [1 pubs, 0 subs]
  ROS 2: /health_monitor_topic (diagnostic_msgs/msg/DiagnosticArray) [1 pubs, 0 subs]
  ROS 2: /parameter_events (rcl_interfaces/msg/ParameterEvent) [2 pubs, 2 subs]
  ROS 2: /rosout (rcl_interfaces/msg/Log) [1 pubs, 0 subs]
```

```sh
$ ros2 run ros1_bridge dynamic_bridge --print-pairs
Supported ROS 2 <=> ROS 1 message type conversion pairs:
  - 'actionlib_msgs/msg/GoalID' (ROS 2) <=> 'actionlib_msgs/GoalID' (ROS 1)
  - 'actionlib_msgs/msg/GoalStatus' (ROS 2) <=> 'actionlib_msgs/GoalStatus' (ROS 1)
  - 'actionlib_msgs/msg/GoalStatusArray' (ROS 2) <=> 'actionlib_msgs/GoalStatusArray' (ROS 1)
  - 'builtin_interfaces/msg/Duration' (ROS 2) <=> 'std_msgs/Duration' (ROS 1)
  - 'builtin_interfaces/msg/Time' (ROS 2) <=> 'std_msgs/Time' (ROS 1)
  - 'diagnostic_msgs/msg/DiagnosticArray' (ROS 2) <=> 'diagnostic_msgs/DiagnosticArray' (ROS 1)
  - 'diagnostic_msgs/msg/DiagnosticStatus' (ROS 2) <=> 'diagnostic_msgs/DiagnosticStatus' (ROS 1)
  - 'diagnostic_msgs/msg/KeyValue' (ROS 2) <=> 'diagnostic_msgs/KeyValue' (ROS 1)
  - 'geometry_msgs/msg/Accel' (ROS 2) <=> 'geometry_msgs/Accel' (ROS 1)
  - 'geometry_msgs/msg/AccelStamped' (ROS 2) <=> 'geometry_msgs/AccelStamped' (ROS 1)
  - 'geometry_msgs/msg/AccelWithCovariance' (ROS 2) <=> 'geometry_msgs/AccelWithCovariance' (ROS 1)
  - 'geometry_msgs/msg/AccelWithCovarianceStamped' (ROS 2) <=> 'geometry_msgs/AccelWithCovarianceStamped' (ROS 1)
  - 'geometry_msgs/msg/Inertia' (ROS 2) <=> 'geometry_msgs/Inertia' (ROS 1)
  - 'geometry_msgs/msg/InertiaStamped' (ROS 2) <=> 'geometry_msgs/InertiaStamped' (ROS 1)
  - 'geometry_msgs/msg/Point' (ROS 2) <=> 'geometry_msgs/Point' (ROS 1)
  - 'geometry_msgs/msg/Point32' (ROS 2) <=> 'geometry_msgs/Point32' (ROS 1)
  - 'geometry_msgs/msg/PointStamped' (ROS 2) <=> 'geometry_msgs/PointStamped' (ROS 1)
  - 'geometry_msgs/msg/Polygon' (ROS 2) <=> 'geometry_msgs/Polygon' (ROS 1)
  - 'geometry_msgs/msg/PolygonStamped' (ROS 2) <=> 'geometry_msgs/PolygonStamped' (ROS 1)
  - 'geometry_msgs/msg/Pose' (ROS 2) <=> 'geometry_msgs/Pose' (ROS 1)
  - 'geometry_msgs/msg/Pose2D' (ROS 2) <=> 'geometry_msgs/Pose2D' (ROS 1)
  - 'geometry_msgs/msg/PoseArray' (ROS 2) <=> 'geometry_msgs/PoseArray' (ROS 1)
  - 'geometry_msgs/msg/PoseStamped' (ROS 2) <=> 'geometry_msgs/PoseStamped' (ROS 1)
  - 'geometry_msgs/msg/PoseWithCovariance' (ROS 2) <=> 'geometry_msgs/PoseWithCovariance' (ROS 1)
  - 'geometry_msgs/msg/PoseWithCovarianceStamped' (ROS 2) <=> 'geometry_msgs/PoseWithCovarianceStamped' (ROS 1)
  - 'geometry_msgs/msg/Quaternion' (ROS 2) <=> 'geometry_msgs/Quaternion' (ROS 1)
  - 'geometry_msgs/msg/QuaternionStamped' (ROS 2) <=> 'geometry_msgs/QuaternionStamped' (ROS 1)
  - 'geometry_msgs/msg/Transform' (ROS 2) <=> 'geometry_msgs/Transform' (ROS 1)
  - 'geometry_msgs/msg/TransformStamped' (ROS 2) <=> 'geometry_msgs/TransformStamped' (ROS 1)
  - 'geometry_msgs/msg/Twist' (ROS 2) <=> 'geometry_msgs/Twist' (ROS 1)
  - 'geometry_msgs/msg/TwistStamped' (ROS 2) <=> 'geometry_msgs/TwistStamped' (ROS 1)
  - 'geometry_msgs/msg/TwistWithCovariance' (ROS 2) <=> 'geometry_msgs/TwistWithCovariance' (ROS 1)
  - 'geometry_msgs/msg/TwistWithCovarianceStamped' (ROS 2) <=> 'geometry_msgs/TwistWithCovarianceStamped' (ROS 1)
  - 'geometry_msgs/msg/Vector3' (ROS 2) <=> 'geometry_msgs/Vector3' (ROS 1)
  - 'geometry_msgs/msg/Vector3Stamped' (ROS 2) <=> 'geometry_msgs/Vector3Stamped' (ROS 1)
  - 'geometry_msgs/msg/Wrench' (ROS 2) <=> 'geometry_msgs/Wrench' (ROS 1)
  - 'geometry_msgs/msg/WrenchStamped' (ROS 2) <=> 'geometry_msgs/WrenchStamped' (ROS 1)
  - 'map_msgs/msg/OccupancyGridUpdate' (ROS 2) <=> 'map_msgs/OccupancyGridUpdate' (ROS 1)
  - 'map_msgs/msg/PointCloud2Update' (ROS 2) <=> 'map_msgs/PointCloud2Update' (ROS 1)
  - 'map_msgs/msg/ProjectedMap' (ROS 2) <=> 'map_msgs/ProjectedMap' (ROS 1)
  - 'map_msgs/msg/ProjectedMapInfo' (ROS 2) <=> 'map_msgs/ProjectedMapInfo' (ROS 1)
  - 'nav_msgs/msg/GridCells' (ROS 2) <=> 'nav_msgs/GridCells' (ROS 1)
  - 'nav_msgs/msg/MapMetaData' (ROS 2) <=> 'nav_msgs/MapMetaData' (ROS 1)
  - 'nav_msgs/msg/OccupancyGrid' (ROS 2) <=> 'nav_msgs/OccupancyGrid' (ROS 1)
  - 'nav_msgs/msg/Odometry' (ROS 2) <=> 'nav_msgs/Odometry' (ROS 1)
  - 'nav_msgs/msg/Path' (ROS 2) <=> 'nav_msgs/Path' (ROS 1)
[... 83 lines cut ...]
  - 'visualization_msgs/msg/InteractiveMarkerFeedback' (ROS 2) <=> 'visualization_msgs/InteractiveMarkerFeedback' (ROS 1)
  - 'visualization_msgs/msg/InteractiveMarkerInit' (ROS 2) <=> 'visualization_msgs/InteractiveMarkerInit' (ROS 1)
[... cut ...]
  - 'nav_msgs/srv/GetMap' (ROS 2) <=> 'nav_msgs/GetMap' (ROS 1)
  - 'nav_msgs/srv/GetPlan' (ROS 2) <=> 'nav_msgs/GetPlan' (ROS 1)
  - 'nav_msgs/srv/SetMap' (ROS 2) <=> 'nav_msgs/SetMap' (ROS 1)
  - 'sensor_msgs/srv/SetCameraInfo' (ROS 2) <=> 'sensor_msgs/SetCameraInfo' (ROS 1)
  - 'std_srvs/srv/Empty' (ROS 2) <=> 'std_srvs/Empty' (ROS 1)
  - 'std_srvs/srv/SetBool' (ROS 2) <=> 'std_srvs/SetBool' (ROS 1)
  - 'std_srvs/srv/Trigger' (ROS 2) <=> 'std_srvs/Trigger' (ROS 1)
  - 'tf2_msgs/srv/FrameGraph' (ROS 2) <=> 'tf2_msgs/FrameGraph' (ROS 1)
```

dirk-thomas, 2019-10-22:
@serge-nikulin When using the Debian packages running the `dynamic_bridge` with `--print-pairs` does include the following line which is necessary to bridge the log messages:

```
  - 'rcl_interfaces/msg/Log' (ROS 2) <=> 'rosgraph_msgs/Log' (ROS 1)
```

If your result varies when building from source the most likely assumption is that you are not building the latest version of each package.

serge-nikulin, 2019-10-22:
@dirk-thomas,

I use OSRF-pre-built Patch 3 binaries from https://github.com/ros2/ros2/releases, pre-built `melodic` (built on 9/27/2019?), and `ros1_bridge` sources from `dashing` branch (ros2.repos retrieved on 10/15/2019).

Unless this `ros1_bridge` does not correspond to Dashing Patch 3, I don't see how I can get de-synchronized sources.

dirk-thomas, 2019-10-22:
> I use OSRF-pre-built Patch 3 binaries from https://github.com/ros2/ros2/releases

Those binaries do show the necessary pair for me. 

> pre-built `melodic` (built on 9/27/2019?), and `ros1_bridge` sources from `dashing` branch (ros2.repos retrieved on 10/15/2019)

I can just speculate that you did something different / are using a different state of the source code. To trouble shoot I would suggest to first check the exact commit hashes used (e.g. of `ros1_bridge` as well as `rcl_interfaces`).

forgetwhatuwant, 2020-12-22:
I faced this problem when my ROS2 version is Crystal. But I changed the Crystal to the Dashing version, and then everything went well. By the way my ros1 version is melodic, I hope that will help!