ros2/ros1_bridge #161: Compile Error: Undefined reference to `rosidl_service_type_support_t'

gmanwal, 2019-01-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04-->
- Installation type:
  - Binaries  --> Crystal and Melodic
- Version or commit hash:
  - Master
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
<colcon build --packages-select ros1_bridge --cmake-force-configure>
```

```

#### Expected behavior
Successful build of ros1_bridge
#### Actual behavior
Error output:
```
<CMakeFiles/test_ros2_server_cpp.dir/test/test_ros2_server.cpp.o: In function `rclcpp::Service<diagnostic_msgs::srv::SelfTest>::Service(std::shared_ptr<rcl_node_t>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::AnyServiceCallback<diagnostic_msgs::srv::SelfTest>, rcl_service_options_t&)':
test_ros2_server.cpp:(.text._ZN6rclcpp7ServiceIN15diagnostic_msgs3srv8SelfTestEEC2ESt10shared_ptrI10rcl_node_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_18AnyServiceCallbackIS3_EER21rcl_service_options_t[_ZN6rclcpp7ServiceIN15diagnostic_msgs3srv8SelfTestEEC5ESt10shared_ptrI10rcl_node_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_18AnyServiceCallbackIS3_EER21rcl_service_options_t]+0xb0): undefined reference to `rosidl_service_type_support_t const* rosidl_typesupport_cpp::get_service_type_support_handle<diagnostic_msgs::srv::SelfTest>()'
collect2: error: ld returned 1 exit status
make[2]: *** [test_ros2_server_cpp] Error 1
make[1]: *** [CMakeFiles/test_ros2_server_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/test_ros2_client_cpp.dir/test/test_ros2_client.cpp.o: In function `rclcpp::Client<diagnostic_msgs::srv::SelfTest>::Client(rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&)':
test_ros2_client.cpp:(.text._ZN6rclcpp6ClientIN15diagnostic_msgs3srv8SelfTestEEC2EPNS_15node_interfaces17NodeBaseInterfaceESt10shared_ptrINS5_18NodeGraphInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEER20rcl_client_options_t[_ZN6rclcpp6ClientIN15diagnostic_msgs3srv8SelfTestEEC5EPNS_15node_interfaces17NodeBaseInterfaceESt10shared_ptrINS5_18NodeGraphInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEER20rcl_client_options_t]+0xc0): undefined reference to `rosidl_service_type_support_t const* rosidl_typesupport_cpp::get_service_type_support_handle<diagnostic_msgs::srv::SelfTest>()'
collect2: error: ld returned 1 exit status
make[2]: *** [test_ros2_client_cpp] Error 1
make[1]: *** [CMakeFiles/test_ros2_client_cpp.dir/all] Error 2
>
```
#### Additional information

dirk-thomas, 2019-02-05:
Building the `ros1_bridge` package works fine for me locally as well as for the nightly CI jobs. Please provide more information about your setup (e.g. exact version) as well as the exact sequence of command you invoke (including sourcing environments etc.).

dirk-thomas, 2019-03-01:
Closing due to no response. Please feel free to comment on the closed ticket with the requested information and the issue can be reopened if necessary.

closed 2019-03-01 by dirk-thomas