ros2/ros1_bridge #165: 1to2 Bridge CPU and Memory Usage Increasing on FastRTPS with default qos

nate-jackson, 2019-02-13:
## Bug report

**Required Info:**

- Operating System: Ubuntu 16.04
- Installation type: binary
- Version or commit hash: bouncy
- DDS implementation: Fast-RTPS
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
We are running the ros1_bridge dynamic bridge. In the 1 to 2 direction, we see the cpu and memory usage climb higher and higher until it hits a little over 100% and causes the dds traffic to get throttled significantly. Seems like a memory leak or something.

On one computer:
```
roscore &
ros2 run ros1_bridge dynamic_bridge -- --bridge-all-2to1-topics
```
We run this on one computer just to get something subscribing to every topic. This seems to make performance degrade faster.

On another computer
```
roscore &
ros2 run ros1_bridge dynamic_bridge -- --bridge-all-1to2-topics
```
and
```
rosbag play -l <bag>
```

#### Expected behavior
CPU and Memory should stay stable, DDS should keep up, for the bridge running in the 1to2 direction.

#### Actual behavior
CPU and Memory usage slowly climb, over several minutes, and dds performance suffers, for the bridge running in the 1to2 direction. We have not seen it recover memory or cpu usage even after 10's of minutes of running.

#### Additional information
We tried writing a custom bridge to only bridge certain topics. Our custom bridge also saw the issue, but only when publishing messages from the bag as default qos. When we switched to sensor data qos, the cpu and memory usage stayed stable. 

We only see it on fastrtps. On the other dds implementation that we tried, it had no issues with default qos. 

We are measuring dds performance with a hzer that we've implemented in c++.

We are replaying this bag:
[time_wrapped_V15_2019-02-08-17-55-58_40.bag.zip](https://github.com/ros2/ros1_bridge/files/2862528/time_wrapped_V15_2019-02-08-17-55-58_40.bag.zip)

The bag has lots of large messages (PointCloud2) messages being sent at high rates. This is what we were testing when we discovered this problem.

dirk-thomas, 2019-02-13:
Can you please try the same with the latest the tires Crystal to confirm that the problem still exists.

clalancette, 2019-02-14:
I'll point out that I ran into a similar bug on Bouncy (in my case, it was just an ordinary pub/sub between two ROS 2 nodes, though at a fairly high frequency).  That problem was definitely fixed in Crystal, so I'd echo @dirk-thomas' suggestion to try again with the newer code.

nate-jackson, 2019-02-14:
I've now tested with crystal and cpu still jumps to 108% and climbing, memory is also climbing. The frequency of the dds messages isn't getting degraded as quickly, but still degrading.

We are running this on two arm64 ubuntu 16.04 machines. I'm going to try on my x64 ubuntu1 16.04 machine and see if results are any different.

dirk-thomas, 2019-02-14:
What is the frequency of message coming in on the ROS 1 side and what is the frequency of the message coming out on the ROS 2 side in the phase where the memory usage is increasing?

dirk-thomas, 2019-02-14:
It sounds to me that you are publishing ROS 1 messages faster than the bridge can relay them and as a consequence (with the default QoS) it is starting to buffer the messages in memory. Once the memory is exhausted it is dropping some of the incoming messages. That would also describe why using the sensor QoS doesn't have that problem since it uses a maximum queue size of 1 and therefore drops any incoming message immediately if it can't keep up with the speed.

nate-jackson, 2019-02-14:
The message is the /right_forecast message from the bag. It is a PointCloud2 with 286 points. There is 770 messages in them and the bag is 10 seconds long, so I expect roughly 77 hz. When it first starts, I get 77 hz on the receiving side, and it stays that way for at least 5 minutes. It's run now for about 10 minutes, and I'm getting 60 hz on the receiving side. But it's now decreasing quite rapidly, so it seems to have hit some ceiling.

dirk-thomas, 2019-02-14:
> I expect roughly 77 hz. When it first starts, I get 77 hz on the receiving side, and it stays that way for at least 5 minutes.

And how is the memory consumption during that phase? Constant or increasing?

> It's run now for about 10 minutes, and I'm getting 60 hz on the receiving side.

How about the memory consumption now?

nate-jackson, 2019-02-14:
That idea about the buffer getting exhausted makes sense, except that memory usage is continually increasing. I haven't yet seen it clean itself up. I would think that would max out. I'm only using top to look at memory usage.

nate-jackson, 2019-02-14:
Memory consumption starts at about 0.5%, and is now at 2.1%. It is increasing more rapidly now as well. CPU usage has topped out at 108% and is staying there.

nate-jackson, 2019-02-14:
During the first part, the memory consumption increases slowly compared to cpu usage.

nate-jackson, 2019-02-14:
Performance doesn't seem to degrade if I only bridge the single topic. So it seems to be related to having lots of publishers. But I may just need to run my test for a lot longer.

nate-jackson, 2019-02-14:
I have tried a different dds implementation (coredx) and it has not had this problem. Just as another data point, and a reason why I'm suspicious that something else is going on instead of just too many messages for the bridge to handle.

dirk-thomas, 2020-01-09:
@nate-jackson Can you please check again if the problem persists in ROS Eloquent since a lot of improvements have been made in FastRTPS since this ticket was created.

dirk-thomas, 2020-05-06:
Closing due to no further response. Please feel free to comment with information for a newer ROS distro and if applicable this ticket can be reopened.

closed 2020-05-06 by dirk-thomas

foxydevelby, 2023-03-29:
Hi, had a similar issue. how to improve the performance of pointcloud? left(ros1) right(ros2)
[image]