ros2/ros1_bridge #168: ros1_bridge fails to build from source

TSC21, 2019-02-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - `ros1_bridge` from source. ROS Melodic and ROS2 Crystal by deb
- Version or commit hash:
  - `961ddee6b76bcb670ce3cb939d125044de820102`
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
1. Clone https://github.com/PX4/px4_ros_com `master` branch to a ROS2 workspace;
2. Clone https://github.com/PX4/px4_ros_com `ros1` to a ROS1 workspace;
3. Build both workspaces using `colcon`;
4. Clone the `ros1_bridge` to the ROS2 workspace;
5. Issue the following commands before building the `ros1_bridge`, by the same order:
```sh
unset ROS_DISTRO
source /opt/ros/melodic/setup.bash
source /opt/ros/crystal/setup.bash # or local_setup.bash

# source the ROS workspaces
source <ros1_ws_dir>/install/setup.bash
source <ros2_ws_dir>/install/local_setup.bash
```
6. Build the `ros1_bridge` using `colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure`.

#### Expected behavior
It's expected that the `ros1_bridge` builds successfully. A notice that the above works when using the ROS2 Bouncy release.

#### Actual behavior
Output from the console:
```sh
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
Starting >>> ros1_bridge
-- Found rmw: 0.6.1 (/opt/ros/crystal/share/rmw/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rosidl_adapter: 0.6.2 (/opt/ros/crystal/share/rosidl_adapter/cmake)
-- Found rclcpp: 0.6.2 (/opt/ros/crystal/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.6.1 (/opt/ros/crystal/share/rmw_implementation_cmake/cmake)
-- Found ament_lint_auto: 0.6.3 (/opt/ros/crystal/share/ament_lint_auto/cmake)
-- Added test 'copyright' to check for copyright in CMake / C / C++ / Python code
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Added test 'flake8' to check Python code syntax and style conventions
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'pep257' to check Python code against some of the style conventions in PEP 257
-- Added test 'uncrustify' to check C / C++ code style
-- Added test 'xmllint' to check XML markup files
-- Found px4_msgs: 0.1.0 (/home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs/share/px4_msgs/cmake)
-- Found action_msgs: 0.6.2 (/opt/ros/crystal/share/action_msgs/cmake)
-- Using CATKIN_DEVEL_PREFIX: /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/devel
-- Using CMAKE_PREFIX_PATH: /home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs;/home/nuno/PX4/px4_ros_com_ros2/install/px4_ros_com;/home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com;/home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs;/opt/ros/melodic;/opt/ros/crystal
-- This workspace overlays: /home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com;/home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs;/opt/ros/melodic
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON              
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.7.14                                
Traceback (most recent call last):              
  File "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/devel/env.sh')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/environment_cache.py", line 62, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
  File "/usr/lib/python3.6/ast.py", line 48, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.6/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    ROS_DISTRO was set to 'crystal' before. Please make sure that the environment does not mix paths from different distributions.
                 ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3
  "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/all.cmake:198 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:12 (find_package)
  /opt/ros/melodic/share/genmsg/cmake/genmsgConfig.cmake:197 (include)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgs-extras.cmake:2 (find_package)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgsConfig.cmake:197 (include)
  CMakeLists.txt:81 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/CMakeFiles/CMakeOutput.log".
See also "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/CMakeFiles/CMakeError.log".
--- stderr: ros1_bridge
Traceback (most recent call last):
  File "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/devel/env.sh')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/environment_cache.py", line 62, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
  File "/usr/lib/python3.6/ast.py", line 48, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.6/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    ROS_DISTRO was set to 'crystal' before. Please make sure that the environment does not mix paths from different distributions.
                 ^
SyntaxError: invalid syntax
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python3
  "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/all.cmake:198 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:12 (find_package)
  /opt/ros/melodic/share/genmsg/cmake/genmsgConfig.cmake:197 (include)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgs-extras.cmake:2 (find_package)
  /opt/ros/melodic/share/actionlib_msgs/cmake/actionlib_msgsConfig.cmake:197 (include)
  CMakeLists.txt:81 (find_package)

---
Failed   <<< ros1_bridge	[ Exited with code 1 ]
                                
Summary: 0 packages finished [1.09s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

TSC21, 2019-02-17:
@dirk-thomas any input on this? Thanks!

dirk-thomas, 2019-02-19:
The error message sounds like your environment is setup incorrectly. Please follow the standard steps to build the `ros1_bridge` and avoid reusing terminal between the builds / workspaces. Also avoid sourcing multiple setup files as described above.

Out of my head I would suggest:

* Terminal A:
  * `source /opt/ros/melodic/setup.bash`
  * build `<ros1_ws_dir>`
* Terminal B:
  * `source /opt/ros/crystal/setup.bash  # or local_setup.bash`
  * build `<ros2_ws_dir>`
* Terminal C:
  * `source <ros1_ws_dir>/install/setup.bash`
  * `source <ros2_ws_dir>/install/setup.bash`
    * make sure to use the latest Debian packages of Crystal to ensure this recursively sources `/opt/ros/crystal`
  * build `<ros1_bridge>`

TSC21, 2019-02-19:
I understand the concern but I am actually using this on an automated build script and for the Bouncy release I was always able to correctly build the ros1_bridge by following the above steps (most of them as I said are automated by script in the same terminal). So even if it is not recommended, this did work before, so my question is what did change on the ros1_bridge side/ROS side?

TSC21, 2019-02-19:
Btw, the automatiom script can be found here: https://github.com/PX4/px4_ros_com/blob/master/scripts/build_all.bash

dirk-thomas, 2019-02-19:
The referenced build script first sources the ROS 2 workspace and then the ROS 1 workspace (which is again not the recommended order) [[1](https://github.com/PX4/px4_ros_com/blob/a8e74d344af320ba859588565f58e985775da4ea/scripts/build_all.bash#L118-L133)]. The output shows this kind of warning which later seems to also be responsible to fail the build:

> ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.

>So even if it is not recommended, this did work before, so my question is what did change on the ros1_bridge side/ROS side?

I am not aware of any changes on the ROS side. But it seems that you have changed this specific order of sourcing the workspaces in your build scipt just two days ago: https://github.com/PX4/px4_ros_com/commit/fa7f6cbf9cb0323b75a17a18dcd3ea8c7a22ed43#diff-737734c8edcb8cec03dd6b034a6328c9

TSC21, 2019-02-20:
@dirk-thomas even using the reverse order as bellow:
```bash
unset ROS_DISTRO
if [ -z $ros1_path ]; then
  source /opt/ros/$ROS1_DISTRO/setup.bash
else
  source $ros1_path
fi
if [ -z $ros2_path ]; then
  source /opt/ros/$ROS2_DISTRO/local_setup.bash
else
  source $ros2_path
fi

# source the ROS workspaces
source $ROS1_WS_DIR/install/setup.bash
source $ROS2_WS_DIR/install/setup.bash
```
The result is the same. Though, after building the workspaces, if I actually try to build the `ros1_bridge` by just sourcing the workspaces `setup.bash`:

```bash
unset ROS_DISTRO
# if [ -z $ros1_path ]; then
#   source /opt/ros/$ROS1_DISTRO/setup.bash
# else
#   source $ros1_path
# fi
# if [ -z $ros2_path ]; then
#   source /opt/ros/$ROS2_DISTRO/local_setup.bash
# else
#   source $ros2_path
# fi

# source the ROS workspaces
source $ROS1_WS_DIR/install/setup.bash
source $ROS2_WS_DIR/install/setup.bash
```

It will eventually start building, but it fails, since it gets some symlink problems:
```console
[ 18%] Linking CXX executable test_ros2_server_cpp  
CMakeFiles/test_ros2_server_cpp.dir/test/test_ros2_server.cpp.o: In function `rclcpp::Service<diagnostic_msgs::srv::SelfTest>::Service(std::shared_ptr<rcl_node_t>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::AnyServiceCallback<diagnostic_msgs::srv::SelfTest>, rcl_service_options_t&)':
test_ros2_server.cpp:(.text._ZN6rclcpp7ServiceIN15diagnostic_msgs3srv8SelfTestEEC2ESt10shared_ptrI10rcl_node_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_18AnyServiceCallbackIS3_EER21rcl_service_options_t[_ZN6rclcpp7ServiceIN15diagnostic_msgs3srv8SelfTestEEC5ESt10shared_ptrI10rcl_node_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_18AnyServiceCallbackIS3_EER21rcl_service_options_t]+0xb0): undefined reference to `rosidl_service_type_support_t const* rosidl_typesupport_cpp::get_service_type_support_handle<diagnostic_msgs::srv::SelfTest>()'
collect2: error: ld returned 1 exit status
CMakeFiles/test_ros2_server_cpp.dir/build.make:162: recipe for target 'test_ros2_server_cpp' failed
make[2]: *** [test_ros2_server_cpp] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/test_ros2_server_cpp.dir/all' failed
make[1]: *** [CMakeFiles/test_ros2_server_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 20%] Linking CXX executable test_ros2_client_cpp  
CMakeFiles/test_ros2_client_cpp.dir/test/test_ros2_client.cpp.o: In function `rclcpp::Client<diagnostic_msgs::srv::SelfTest>::Client(rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&)':
test_ros2_client.cpp:(.text._ZN6rclcpp6ClientIN15diagnostic_msgs3srv8SelfTestEEC2EPNS_15node_interfaces17NodeBaseInterfaceESt10shared_ptrINS5_18NodeGraphInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEER20rcl_client_options_t[_ZN6rclcpp6ClientIN15diagnostic_msgs3srv8SelfTestEEC5EPNS_15node_interfaces17NodeBaseInterfaceESt10shared_ptrINS5_18NodeGraphInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEER20rcl_client_options_t]+0xc0): undefined reference to `rosidl_service_type_support_t const* rosidl_typesupport_cpp::get_service_type_support_handle<diagnostic_msgs::srv::SelfTest>()'
collect2: error: ld returned 1 exit status
CMakeFiles/test_ros2_client_cpp.dir/build.make:162: recipe for target 'test_ros2_client_cpp' failed
make[2]: *** [test_ros2_client_cpp] Error 1
CMakeFiles/Makefile2:700: recipe for target 'CMakeFiles/test_ros2_client_cpp.dir/all' failed
make[1]: *** [CMakeFiles/test_ros2_client_cpp.dir/all] Error 2
[ 26%] Linking CXX executable simple_bridge_1_to_2  
CMakeFiles/simple_bridge_1_to_2.dir/src/simple_bridge_1_to_2.cpp.o: In function `rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::Publisher(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_publisher_options_t const&, std::shared_ptr<std::allocator<std_msgs::msg::String_<std::allocator<void> > > > const&)':
simple_bridge_1_to_2.cpp:(.text._ZN6rclcpp9PublisherIN8std_msgs3msg7String_ISaIvEEES4_EC2EPNS_15node_interfaces17NodeBaseInterfaceERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERK23rcl_publisher_options_tRKSt10shared_ptrISaIS5_EE[_ZN6rclcpp9PublisherIN8std_msgs3msg7String_ISaIvEEES4_EC5EPNS_15node_interfaces17NodeBaseInterfaceERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERK23rcl_publisher_options_tRKSt10shared_ptrISaIS5_EE]+0x22): undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String_<std::allocator<void> > >()'
collect2: error: ld returned 1 exit status
CMakeFiles/simple_bridge_1_to_2.dir/build.make:158: recipe for target 'simple_bridge_1_to_2' failed
make[2]: *** [simple_bridge_1_to_2] Error 1
CMakeFiles/Makefile2:1281: recipe for target 'CMakeFiles/simple_bridge_1_to_2.dir/all' failed
make[1]: *** [CMakeFiles/simple_bridge_1_to_2.dir/all] Error 2
[ 28%] Linking CXX executable simple_bridge_2_to_1  
CMakeFiles/simple_bridge_2_to_1.dir/src/simple_bridge_2_to_1.cpp.o: In function `rclcpp::create_subscription_factory<std_msgs::msg::String_<std::allocator<void> >, void (&)(std::shared_ptr<std_msgs::msg::String_<std::allocator<void> > >), std::allocator<void>, std_msgs::msg::String [... cut ...] ence to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String_<std::allocator<void> > >()'
collect2: error: ld returned 1 exit status
make[2]: *** [simple_bridge_2_to_1] Error 1
CMakeFiles/simple_bridge_2_to_1.dir/build.make:158: recipe for target 'simple_bridge_2_to_1' failed
CMakeFiles/Makefile2:2187: recipe for target 'CMakeFiles/simple_bridge_2_to_1.dir/all' failed
make[1]: *** [CMakeFiles/simple_bridge_2_to_1.dir/all] Error 2
```

TSC21, 2019-02-20:
@dirk-thomas also if I try to follow your suggestion, regarding the Terminal C, it fails:
```sh
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
Starting >>> ros1_bridge
-- Found rmw: 0.6.1 (/opt/ros/crystal/share/rmw/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
Traceback (most recent call last):              
  File "/opt/ros/crystal/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at /opt/ros/crystal/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3
  /opt/ros/crystal/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/crystal/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:38 (include)
  /opt/ros/crystal/share/rmw/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  /opt/ros/crystal/share/rmw/cmake/rmwConfig.cmake:38 (include)
  CMakeLists.txt:13 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/CMakeFiles/CMakeOutput.log".
See also "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/CMakeFiles/CMakeError.log".
--- stderr: ros1_bridge
Traceback (most recent call last):
  File "/opt/ros/crystal/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at /opt/ros/crystal/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3
  /opt/ros/crystal/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/crystal/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:38 (include)
  /opt/ros/crystal/share/rmw/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  /opt/ros/crystal/share/rmw/cmake/rmwConfig.cmake:38 (include)
  CMakeLists.txt:13 (find_package)

---
Failed   <<< ros1_bridge	[ Exited with code 1 ]

Summary: 0 packages finished [0.21s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

TSC21, 2019-02-20:
Is this the paths I should expect:
```cmake
-- Using CMAKE_PREFIX_PATH: /home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs;/home/nuno/PX4/px4_ros_com_ros2/install/px4_ros_com;/home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com;/home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs;/opt/ros/melodic;/opt/ros/crystal;/home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge
-- This workspace overlays: /home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com;/home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs;/opt/ros/melodic
```

dirk-thomas, 2019-02-20:
With the steps I mentioned I above I only get that far:

* Terminal A in `<ros1_ws_dir>`:
  * `source /opt/ros/melodic/setup.bash`
  * `colcon build`
* Terminal B in `<ros2_ws_dir>`:
  * `source /opt/ros/crystal/setup.bash
    * `export ROS2_DISTRO=$ROS_DISTRO`  # this is necessary since `px4_ros_com` expects that custom env var
  * `colcon build`
    * Fails to build with `python import error:  No module named 'genmsg'`
      Your ROS 2 specific branch tries to use `genmsg` from ROS 1 - you simply can't rely on that package to be available.
      * A comment regarding the error message: you should print any kind of error message to `stderr` and not as currently done to `stdout`.
    * To get past this I sourced `/opt/ros/melodic/setup.bash` before the Crystal setup file (**which is not something you should require / rely on**).
    * The build fails with: `generate_microRTPS_bridge.py: error: argument -f/--fastrtpsgen-dir: expected one argument`
      * I don't think the Crystal binaries contain `fastrtpsgen`?
* Terminal C (after the above hack in terminal B):
  * *didn't get here*

In order to be able to help you with your problem I need to be able to easily reproduce the case following our recommended workflow and copy-n-paste steps. Currently that is not the case and it took me way too much time to even get until here. Therefore I will wait for more thorough steps before trying it again.

TSC21, 2019-02-20:
> * Fails to build with `python import error:  No module named 'genmsg'`
  Your ROS 2 specific branch tries to use `genmsg` from ROS 1 - you simply can't rely on that package to be available.

I get that but we are setting this on the same script so or I install `genmsg` using `apt` or `pip` or I rely on the ROS1 package.

> The build fails with: `generate_microRTPS_bridge.py: error: argument -f/--fastrtpsgen-dir: expected one argument`
* I don't think the Crystal binaries contain `fastrtpsgen`?

The only thing required is installing FastRTPS on your system by source or using a binary release. You should install release 1.7.

> In order to be able to help you with your problem I need to be able to easily reproduce the case following our recommended workflow and copy-n-paste steps. Currently that is not the case and it took [...]

Most of the steps are documented in https://dev.px4.io/en/middleware/micrortps.html. But part of it is already outdated given the restructure I did on `px4_msgs` and `px4_ros_com` packages. But I think that as soon as you install FastRTPS on your system, all should be good.

TSC21, 2019-02-20:
@dirk-thomas thanks for giving a try to this.

dirk-thomas, 2019-02-21:
I will try to capture the reasons for the various failures mentioned in this thread as well as how to fix them:

> ```
> Traceback (most recent call last):              
> ...
>     ROS_DISTRO was set to 'crystal' before. Please make sure that the environment does not mix paths from different distributions.
>                  ^
> SyntaxError: invalid syntax
> ```

When building a ROS 2 workspace the ROS_DISTRO environment variable must be set to the ROS 2 distro name. Otherwise the implicitly sourced environment will print a warning which isn't expected and results in that failue.

---

> ```
> if [ -z $ros1_path ]; then
>   source /opt/ros/$ROS1_DISTRO/setup.bash
> else
>   source $ros1_path
> fi
> if [ -z $ros2_path ]; then
>   source /opt/ros/$ROS2_DISTRO/local_setup.bash
> else
>   source [...]

Do not mix the order of ROS 1 and ROS 2 workspaces: always source all ROS 1 workspace first, then source all ROS 2 workspaces. That ensures that CMake finds the ROS 2 packages (pkg_config is used for ROS 1 packages).

---

> ```
> # if [ -z $ros1_path ]; then
> #   source /opt/ros/$ROS1_DISTRO/setup.bash
> # else
> #   source $ros1_path
> # fi
> # if [ -z $ros2_path ]; then
> #   source /opt/ros/$ROS2_DISTRO/local_setup.bash
> # else [...]

resulting in:

> ```
> CMake Warning at CMakeLists.txt:33 (message):
>   Failed to find ROS 1 roscpp, skipping...
> ```

This doesn't work since the `$ROS1_WS_DIR/install/setup.bash` script doesn't correctly source `/opt/ros/$ROS1_DISTRO/setup.bash`. As a consequence `PKG_CONFIG_PATH` doesn't contain `/opt/ros/$ROS1_DISTRO` and ROS 1 can't be found with pkg-config. This is a bug in the chaining when the underlay is coming from ROS 1 Debian packages and the overlay is a colcon-built workspace.

---

> ```
> /tmp/wsb/src/ros1_bridge/test/test_ros2_server.cpp:17:10: fatal error: diagnostic_msgs/srv/self_test.hpp: No such file or directory
>  #include "diagnostic_msgs/srv/self_test.hpp"
> /tmp/wsb/src/ros1_bridge/test/test_ros2_client.cpp:18:10: fatal error: diagnostic_msgs/srv/self_test.hpp: No such file or directory
>  #include "diagnostic_msgs/srv/self_test.hpp"
> ```

or

> ```
> /opt/ros/crystal/include/rclcpp/subscription_factory.hpp:108: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String_<std::allocator<void> > >()'
> /opt/ros/crystal/include/rclcpp/publisher.hpp:173: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String_<std::allocator<void> > >()'
> /opt/ros/crystal/include/rclcpp/subscription_factory.hpp:108: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String_<std::allocator<void> > >()'
> /opt/ros/crystal/include/rclcpp/publisher.hpp:173: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String_<std::allocator<void> > >()'
> ```

This happens when the ROS 2 package `diagnostic_msgs` isn't being found and the ROS 1 package is selected instead. See #169 for an improved error message in that case.

---

With all this I was finally able to build the bridge.

dirk-thomas, 2019-02-21:
> Most of the steps are documented

This will get off-topic:

Documentation is certainly good but ROS packages are standardized for a reason: so that you can process them in the same way. If I want to build *N* ROS packages I don't want to read potentially *N* custom instruction documents on how each of them wants to be treated. The ROS ecosystem is relying on that foundation to efficiently modularize complex applications. Therefore I would highly suggest to consider using a standardized approach which doesn't rely on customizations of the process. Also as soon as you want your packages to be processed on any ROS infrastructure (for CI or release) it will likely not work with such custom requirements.

dirk-thomas, 2019-02-26:
Since there is no further comment I assume the problem was somewhere in the process and you were able to resolve it based on the provided information.

closed 2019-02-26 by dirk-thomas

TSC21, 2019-02-26:
@dirk-thomas I am actually still trying to find a way of having this done in a propper way and still maintain flexibility. Regarding the `genmsg` problem: I do see it happen, so my question is - how do we solve it without sourcing the ROS1 environment? Installing `python-genmsg` did not solve the problem.

TSC21, 2019-02-26:
Also, even after building both workspaces in separate windows and having `ros1_bridge` to be built in another terminal as well, I am still not able to build it:
```sh
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
Starting >>> ros1_bridge
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done          
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done        
-- Found rmw: 0.6.1 (/opt/ros/crystal/share/rmw/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rosidl_adapter: 0.6.3 (/opt/ros/crystal/share/rosidl_adapter/cmake)
-- Found rclcpp: 0.6.3 (/opt/ros/crystal/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.6.1 (/opt/ros/crystal/share/rmw_implementation_cmake/cmake)
-- Found std_msgs: 0.6.1 (/opt/ros/crystal/share/std_msgs/cmake)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'roscpp'
--   No package 'roscpp' found
CMake Warning at CMakeLists.txt:33 (message):
  Failed to find ROS 1 roscpp, skipping...

-- Configuring done                             
-- Generating done
-- Build files have been written to: /home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge
Install the project...                          
-- Install configuration: ""
-- Execute custom install script
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ament_index/resource_index/package_run_dependencies/ros1_bridge
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ament_index/resource_index/parent_prefix_path/ros1_bridge
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/environment/ament_prefix_path.sh
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/environment/path.sh
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/local_setup.bash
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/local_setup.sh
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/local_setup.zsh
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ament_index/resource_index/packages/ros1_bridge
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/cmake/ros1_bridgeConfig.cmake
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/cmake/ros1_bridgeConfig-version.cmake
-- Symlinking: /home/nuno/PX4/px4_ros_com_ros2/install/ros1_bridge/share/ros1_bridge/package.xml
--- stderr: ros1_bridge
CMake Warning at CMakeLists.txt:33 (message):
  Failed to find ROS 1 roscpp, skipping...

---
Finished <<< ros1_bridge [1.30s]
                     
Summary: 1 package finished [1.38s]
  1 package had stderr output: ros1_bridge
```
The process, in the same terminal:
1. `source $ROS1_WS_DIR/install/setup.bash`
2. `source $ROS2_WS_DIR/install/setup.bash`
3. `colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+`

TSC21, 2019-02-26:
A further note: I am building the ROS1 workspace using `colcon`. I don't know if that matters or not.

dirk-thomas, 2019-02-26:
> Regarding the `genmsg` problem: I do see it happen, so my question is - how do we solve it without sourcing the ROS1 environment?

If you want to use a ROS 1 package you should source a ROS 1 environment. Anything else is a hack.

> Installing `python-genmsg` did not solve the problem.

If you are referring to the package provided by Debian upstream you shouldn't mix those packages with packages provided by the OSRF apt repo.

> CMake Warning at CMakeLists.txt:33 (message):
>   Failed to find ROS 1 roscpp, skipping...

Please re-read my previous comment where I specifically mention this case and that you need to manually source the ROS 1 underlay (before sourcing the ROS 1 overlay).

TSC21, 2019-02-26:
> Please re-read my previous comment where I specifically mention this case and that you need to manually source the ROS 1 underlay (before sourcing the ROS 1 overlay).

Alright so that did solve the issue. But now, I am facing the problem with `diagnostic_msgs`, which was supposed to be solved in https://github.com/ros2/ros1_bridge/pull/169. And I am cloning `ros1_bridge` master. Is there something I am missing here?

TSC21, 2019-02-26:
OK, I just saw that that PR was just to induce an error message when the package is not found. So in that sense, I am still blocked since I am getting into the same trouble as before.

> Please re-read my previous comment where I specifically mention this case and that you need to manually source the ROS 1 underlay (before sourcing the ROS 1 overlay).

I did this and I am facing the same problem as before.

TSC21, 2019-02-26:
@dirk-thomas from the https://github.com/ros2/ros1_bridge/issues/168#issuecomment-465838155 comment, it's not clear what steps did you actually take to make this work. Would you mind writing them so I can properly reproduce a working setup? Thanks!

dirk-thomas, 2019-02-26:
> I am still blocked since I am getting into the same trouble as before.

This thread describes numerous different problems. So this is simply not enough information for me to give you any feedback.

> it's not clear what steps did you actually take to make this work. Would you mind writing them so I can properly reproduce a working setup?

I have reproduced your setup multiple times and tried to comment on each and every problem discovered in this thread and how I got around them. At the end it built successfully for me - with the steps described before.

As mentioned before: if you would like help to debug a problem you can't resolve please include all exact steps you are doing and the resulting error as well as what you have already tried to resolve the problem.

TSC21, 2019-02-26:
Ok, so there's not misunderstand here:
- Terminal 1: (ROS2 ws)
1. `source /opt/ros/melodic/setup.bash` because of `genmsg`
2. `source /opt/ros/crystal/setup.bash`
3. `export ROS2_DISTRO=crystal` (will be fixed so to not be required)
4. `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+`

- Terminal 2: (ROS1 ws)
1. `source /opt/ros/melodic/setup.bash`
2. `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+`

- Terminal 3: (ros1_bridge)
1. `source /opt/ros/melodic/setup.bash`
2. `source px4_ros_com_ros1/install/setup.bash`
3. `source px4_ros_com_ros2/install/setup.bash`
4. `colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+`

In terminal 3, this results in:
```
CMake Error at CMakeLists.txt:67 (message):     
  Failed to find ROS 2 package 'diagnostic_msgs'

-- Configuring incomplete, errors occurred!
See also "/home/nuno/PX4/px4_ros_com_ros2/build/ros1_bridge/CMakeFiles/CMakeOutput.log".
--- stderr: ros1_bridge
CMake Error at CMakeLists.txt:67 (message):
  Failed to find ROS 2 package 'diagnostic_msgs'

---
Failed   <<< ros1_bridge	[ Exited with code 1 ]

Summary: 0 packages finished [6.86s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

dirk-thomas, 2019-02-26:
> ... as well as what you have already tried to resolve the problem

I assume you have double checked that the ROS 2 package `diagnostic_msgs` is available? Have you checked the various `*_PREFIX_PATH` environment variables? What is there content - is ROS 1 anywhere in front of ROS 2?

TSC21, 2019-02-26:
> I assume you have double checked that the ROS 2 package `diagnostic_msgs` is available?

Yes I have and it is available.

> Have you checked the various `*_PREFIX_PATH` environment variables? What is their content - is ROS 1 anywhere in front of ROS 2?

How do I effectively check that?

TSC21, 2019-02-26:
@dirk-thomas is the order of the setup I wrote in https://github.com/ros2/ros1_bridge/issues/168#issuecomment-467618941 is correct or not? Cause that's what I am trying to figure out. I still cannot be sure I am following the right procedure and move on to other kind of debug if I am not even sure I am following the correct order of the setup.

TSC21, 2019-02-26:
Update: New attempt, now without `--symlink-install`. Same issue.

dirk-thomas, 2019-02-26:
> > Have you checked the various `*_PREFIX_PATH` environment variables? What is their content - is ROS 1 anywhere in front of ROS 2?

> How do I effectively check that?

`env | grep _PREFIX_PATH`

> is the order of the setup I wrote in [#168 (comment)](https://github.com/ros2/ros1_bridge/issues/168#issuecomment-467618941) correct or not?

It looks fine.

> Cause that's what I am trying to figure out. I still cannot be sure I am following the right procedure and can't move on to other kind of debug if I am not even sure I am following the correct order of the setup.

When you get to the error message `Failed to find ROS 2 package 'diagnostic_msgs'` the immediate next questions are:
* is it there (you confirmed yes)
* how it is being found: using the `CMAKE_PREFIX_PATH`, hence double checking the value of these environment variables

TSC21, 2019-02-27:
@dirk-thomas:
```bash
$ env | grep _PREFIX_PATH
AMENT_PREFIX_PATH=/home/nuno/PX4/px4_ros_com_ros2/install/px4_ros_com:/home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs:/opt/ros/crystal
CMAKE_PREFIX_PATH=/home/nuno/PX4/px4_ros_com_ros2/install/px4_ros_com:/home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs:/home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com:/home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs:/opt/ros/melodic
COLCON_PREFIX_PATH=/home/nuno/PX4/px4_ros_com_ros2/install:/home/nuno/PX4/px4_ros_com_ros1/install:/opt/ros/crystal
```
What output should I expect? Since `/opt/ros/crystal` is not on the `CMAKE_PREFIX_PATH`, I assume that's the issue.

TSC21, 2019-02-27:
> > ... as well as what you have already tried to resolve the problem
> 
> I assume you have double checked that the ROS 2 package `diagnostic_msgs` is available? Have you checked the various `*_PREFIX_PATH` environment variables? What is their content - is ROS 1 anywhere in front of ROS 2?

When you mean in front, do you mean, in the `CMAKE_PREFIX_PATH`, the ROS1 one paths appear before or after the ROS2 paths, considering the left to right reading (starting on `=`)?

dirk-thomas, 2019-02-27:
> When you mean in front, do you mean, in the `CMAKE_PREFIX_PATH`, the ROS1 paths appear before or after the ROS2 paths, considering the left to right reading (starting on `=`)? And, for this to work, should they appear first or after?

Yes, the order of paths in `CMAKE_PREFIX_PATH` determines in which locations CMake looks first. As mentioned in the instructions of the `ros1_bridge` as well as this thread the ROS 2 paths need to appear *before* the ROS 1 paths.

> I think the problem here is that `source px4_ros_com_ros2/install/setup.bash` is not adding `/opt/ros/crystal` to the `CMAKE_PREFIX_PATH`.

`/opt/ros/crystal` not being present is obviously the reason for CMake not being able to find the ROS 2 package `diagnostic_msgs`.

Please check my comment from above (https://github.com/ros2/ros1_bridge/issues/168#issuecomment-465244081):

> `source <ros2_ws_dir>/install/setup.bash`
> * make sure to use the latest Debian packages of Crystal to ensure this recursively sources `/opt/ros/crystal`

What version of `ros-crystal-ros-workspace` do you have installed?

TSC21, 2019-02-27:
> What version of `ros-crystal-ros-workspace` do you have installed?

`0.6.1-0bionic.20190208.212300`

TSC21, 2019-02-27:
I suppose we are expecting https://github.com/ros2/ros2/issues/653 fix, meaning https://github.com/ros2/ros_workspace/pull/11, to be included, but from what I can see, https://discourse.ros.org/t/patch-release-and-new-packages-for-ros-2-crystal-clemmys-2019-02-14/7867 is out but my system states it's up to date.

TSC21, 2019-02-27:
http://repo.ros2.org/status_page/ros_crystal_default.html?q=ros_workspace. It seems to be the latest. So I suppose it includes the patch as well, meaning that this still doesn't seem to be the problem.

dirk-thomas, 2019-02-27:
> So I suppose `2019-02-14` deb packages are still not out.

Just because the announcement was sent on Feb 14th doesn't imply that all Debian packages have been rebuilt on that date. Actually it is very common that the rebuild happens some time before and the resulting Debian packages are being checked / tested before being announced.

The second patch release announcement does contain this version bump:

> ros-crystal-ros-workspace: 0.6.0-1 -> 0.6.1-0

And you already confirmed that you do have version `0.6.1` installed. So that seems to be just fine. So when you source `/opt/ros/crystal` you should have `COLCON_PREFIX_PATH` set to that path, right? And when you source `px4_ros_com_ros2/install` you should have that path as well as `/opt/ros/crystal` in the `COLCON_PREFIX_PATH`, right?

I have to revised my previous comment:

> `/opt/ros/crystal` not being present is obviously the reason for CMake not being able to find the ROS 2 package `diagnostic_msgs`.

As long as `/opt/ros/crystal` is in the `COLCON_PREFIX_PATH` `colcon` will add it to the `CMAKE_PREFIX_PATH`. 

When your build in the 3rd terminal failed the environment was:

```shell
CMAKE_PREFIX_PATH=
  /home/nuno/PX4/px4_ros_com_ros2/install/px4_ros_com:
  /home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs:
  /home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com:
  /home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs:
  /opt/ros/melodic
COLCON_PREFIX_PATH=
  /home/nuno/PX4/px4_ros_com_ros2/install:
  /home/nuno/PX4/px4_ros_com_ros1/install:
  /opt/ros/crystal
```

While `/opt/ros/crystal` from the `COLCON_PREFIX_PATH` should be added to the `CMAKE_PREFIX_PATH` it might happen in the wrong order. Can you check how the variable is reported in the first line of the file `log/latest_build/ros1_bridge/command.log`?

If it does read something like this: `CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/opt/ros/crystal` the resulting order is incorrect. As a temporary workaround you can try to either remove `/opt/ros/melodic` from `CMAKE_PREFIX_PATH` or inserting `/opt/ros/crystal` before the melodic path.

dirk-thomas, 2019-03-01:
> If it does read something like this: `CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/opt/ros/crystal` the resulting order is incorrect. As a temporary workaround you can try to either remove `/opt/ros/melodic` from `CMAKE_PREFIX_PATH` or inserting `/opt/ros/crystal` before the melodic path.

The latest release of `colcon` should make this work out-of-the-box (see colcon/colcon-ros#53).

TSC21, 2019-03-04:
> The latest release of `colcon` should make this work out-of-the-box (see [colcon/colcon-ros#53](https://github.com/colcon/colcon-ros/issues/53)).

@dirk-thomas so should I wait for a new ROS patch release?

TSC21, 2019-03-04:
@dirk-thomas by manually exporting the `CMAKE_PREFIX_PATH` with your suggestion:
```sh
export CMAKE_PREFIX_PATH=/home/nuno/PX4/px4_ros_com_ros2/install/px4_ros_com:/home/nuno/PX4/px4_ros_com_ros2/install/px4_msgs:/home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com:/home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs:/opt/ros/crystal/setup.bash:/opt/ros/melodic
```
It seems I was able to build the bridge, even though with some warnings regarding `liburdfdom`:
```sh
CMake Warning at CMakeLists.txt:117 (add_executable):
  Cannot generate a safe runtime search path for target dynamic_bridge
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:177 (custom_executable)

CMake Warning at CMakeLists.txt:117 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:163 (custom_executable)

CMake Warning at CMakeLists.txt:146 (add_library):
  Cannot generate a safe runtime search path for target ros1_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:117 (add_executable):
  Cannot generate a safe runtime search path for target parameter_bridge
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/crystal/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:170 (custom_executable)
```
This is however outputting as a `stderr`. So, 2 questions:
1. Are those warning something I should worry about?
2. Is there a way of improving the verbosity of `colcon` for situations like this, where warnings are exported as errors?

Also, I suppose that soon as the new release of `colcon` comes with a ROS patch, this won't be necessary, but as a temporary fix this works for now. Thanks for all your time and dedication on helping solving this.

dirk-thomas, 2019-03-04:
> > The latest release of colcon should make this work out-of-the-box (see colcon/colcon-ros#53).

> @dirk-thomas so should I wait for a new ROS patch release?

Python packages like `colcon` are not bound to a specific ROS distribution and therefore not subjects to syncs / releases of individual distros. The latest version of `colcon` packages should already be available.

dirk-thomas, 2019-03-04:
> It seems I was able to build the bridge, even though with some warnings regarding `liburdfdom`:
> This is however outputting as a `stderr`.
> Are those warning something I should worry about?

I haven't seen the warnings during my build (if I remember correctly). I general I wouldn't recommend to ignore such warnings. The [packaging_linux](https://ci.ros2.org/view/packaging/job/packaging_linux/1386/consoleFull) job doesn't show that warning either but it is also using a much simpler setup than your builds. I am not sure what exact part causes the warnings in your case.

> Is there a way of improving the verbosity of colcon for situations like this, where warnings are exported as errors?

CMake outputs warnings to `stderr` which seems reasonable. `colcon` by default doesn't show any output until a package has finished and in case of any `stderr` output shows that. You can configure `colcon` to not show `stderr` output in that case (`--event-handlers console_stderr-`) but that doesn't seem like a good idea since it would also hide any other `stderr` output.

TSC21, 2019-03-04:
> Python packages like `colcon` are not bound to a specific ROS distribution and therefore not subjects to syncs / releases of individual distros. The latest version of `colcon` packages should [...]

Yep it seems like I don't require to manually extend `CMAKE_PREFIX_PATH` to have it build. Thanks!

> I haven't seen the warnings during my build (if I remember correctly). I general I wouldn't recommend to ignore such warnings. The [...]

Do you have any tip on what may be causing this? I only get this on `urdfdom`.

> CMake outputs warnings to `stderr` which seems reasonable. `colcon` by default doesn't show any output until a package has finished and in case of any `stderr` output shows that. You can configure [...]

Yes that seems reasonable. Thanks!

dirk-thomas, 2019-03-04:
> Do you have any tip on what may be causing this? I only get this on `urdfdom`.

Some ROS 1 package(s) is linking against it (expecting the libraries to come from `/usr/lib`) and some ROS 2 package(s) is linking against it (expecting the libraries to come from `/opt/ros/crystal/lib`). Since there can be only one order in `LD_LIBRARY_PATH` it needs to be ensured that both are using ABI compatible versions of `urdfdom` otherwise the ROS 1 package(s) getting the ROS 2 libs might fail / break / behave badly.

TSC21, 2019-03-04:
@dirk-thomas I do see another thing appearing in the terminal where I am building the `ros1_bridge`.

```sh
[0.179s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/nuno/PX4/px4_ros_com_ros1/install/px4_ros_com' in the environment variable CMAKE_PREFIX_PATH seems to be a catkin workspace but it doesn't contain any 'local_setup.*' files. Maybe the catkin version is not up-to-date?
[0.180s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/nuno/PX4/px4_ros_com_ros1/install/px4_msgs' in the environment variable CMAKE_PREFIX_PATH seems to be a catkin workspace but it doesn't contain any 'local_setup.*' files. Maybe the catkin version is not up-to-date?
[0.180s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/opt/ros/melodic' in the environment variable CMAKE_PREFIX_PATH seems to be a catkin workspace but it doesn't contain any 'local_setup.*' files. Maybe the catkin version is not up-to-date?
```

dirk-thomas, 2019-03-04:
> I do see another thing appearing in the terminal where I am building the ros1_bridge.

Currently the ROS 1 overlay built with `colcon` can't source the ROS 1 underlay coming from Debian packages. Simply because `catkin` doesn't have a way to only source one specific workspace. The problem was mentioned before:

> This doesn't work since the `$ROS1_WS_DIR/install/setup.bash` script doesn't correctly source `/opt/ros/$ROS1_DISTRO/setup.bash`. ... This is a bug in the chaining when the underlay is coming from ROS 1 Debian packages and the overlay is a colcon-built workspace.

ros/catkin#993 is introducing these `local_setup` files for `catkin`. Though that change has not been released yet. Until that is the case you can continue to explicitly source the ROS 1 underlay.

TSC21, 2019-03-04:
> ros/catkin#993 is introducing these local_setup files for catkin. Though that change has not been released yet. Until that is the case you can continue to explicitly source the ROS 1 underlay.

Alright makes sense. Thanks! Although it's still not clear to me how to solve the `urdfdom` issue.

slmat27, 2019-03-07:
@dirk-thomas I have been facing similar issue, exactly similar to the below comment, where I get the same error in Terminal B and can't find a way to solve it to reach Terminal C. Using Ubuntu 18.04, ROS1 Melodic, and ROS2 Crystal. Your feedback would be much appreciated on how did you resolve this issue and was able to build ros1_bridge accordingly in the the third terminal. Thank you!

> With the steps I mentioned I above I only get that far:
> 
>     * Terminal A in `<ros1_ws_dir>`:
>       
>       * `source /opt/ros/melodic/setup.bash`
>       * `colcon build`
> 
>     * Terminal B in `<ros2_ws_dir>`:
>       
>       * `source /opt/ros/crystal/setup.bash
>         
>         * `export ROS2_DISTRO=$ROS_DISTRO`  # this is necessary since `px4_ros_com` expects that custom env var
>       * `colcon build`
>         
>         * Fails to build with `python import error:  No module named 'genmsg'`
>           Your ROS 2 specific branch tries to use `genmsg` from ROS 1 - you simply can't rely on that package to be available.
>           
>           * A comment regarding the error message: you should print any kind of error message to `stderr` and not as currently done to `stdout`.
>         * To get past this I sourced `/opt/ros/melodic/setup.bash` before the Crystal setup file (**which is not something you should require / rely on**).
>         * The build fails with: `generate_microRTPS_bridge.py: error: argument -f/--fastrtpsgen-dir: expected one argument`
>           
>           * I don't think the Crystal binaries contain `fastrtpsgen`?
> 
>     * Terminal C:
>       
>       * _didn't get here_
>

dirk-thomas, 2019-03-07:
@slmat27 Please ask your question on answers.ros.org and include enough information to reproduce your problem.

> how did you resolve this issue and was able to build ros1_bridge accordingly in the the third terminal.

For this generic question my simple answer will be: as described in the above thread.

slmat27, 2019-03-11:
@dirk-thomas Thank you for your feedback, I came directly to you instead of ROS Answers, since your problem was exactly like mine and I read all the messages in the thread above, but none was explaining how did you solve the problem.

I am using Ubuntu 18.04, with ROS Melodic and ROS2 Crystal, followed these setup steps:

1. Clone https://github.com/PX4/px4_ros_com `master` branch to a ROS2 workspace;
2. Clone https://github.com/PX4/px4_msgs `master` branch to the same ROS2 workspace;
3. Clone https://github.com/PX4/px4_ros_com `ros1` to a ROS1 workspace;
4. Clone https://github.com/PX4/px4_msgs `ros1` branch to the same ROS1 workspace;
5. Installed Fast RTPS using the steps over here https://dev.px4.io/en/setup/fast-rtps-installation.html

Steps to reproduce issue:

- Terminal A in `<ros1_ws_dir>`:
  * `source /opt/ros/melodic/setup.bash`
  * `colcon build`

- Terminal B in `<ros2_ws_dir>`:
  * `source /opt/ros/melodic/setup.bash`
  * `source /opt/ros/crystal/setup.bash`
  * `export ROS2_DISTRO=$ROS_DISTRO`
  * `colcon build`
    * `px4_msgs` build is completed successfully
    * `px4_ros_com` build fails and trigger the below error
      * `generate_microRTPS_bridge.py: error: argument -f/--fastrtpsgen-dir: expected one argument`

which based on the above thread, it was the same error you got and @TSC21 replied saying:

> The only thing required is installing FastRTPS on your system by source or using a binary release. You should install release 1.7.

which I already have it installed, but yet the error persists and I was simply wondering if you can help me fixing the error you managed to fix, since my ultimate goal later is to build `ros1_bridge` package and use it, but I just got stuck in the same step you did.

Thanks once again!

TSC21, 2019-03-11:
@slmat27 if you have doubts regarding how to properly set px4_ros_com, move your questions to an issue on px4_ros_com instead. What was debated here was a completely different matter. Thanks