ros2/ros1_bridge #181: Garbled text output in [INFO] messages

mlanting, 2019-04-02:
- Operating System: Ubuntu 18.04
- Installation type: Source install of Crystal, binary install of Melodic
- Version or commit hash:
  - Output of git rev-parse HEAD: 01e9a68220aada84b3b5111829ec206b3d98352b
- DDS implementation: fastRTPS

#### Steps to reproduce issue
Run the ros1_bridge tutorial Example 1 as described here: https://github.com/ros2/ros1_bridge/blob/master/README.md

#### Expected behavior
Output should be readable.

#### Actual behavior
Output contains garbled characters or is sometimes incomplete, e.x.:
    `~/ros2_ws$ ros2 run ros1_bridge dynamic_bridge 
    created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/Log' and ROS 1 type 'rosgraph_msgs/Log'
    created 1to2 bridge for topic '/chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/String'
     (showing msg only once per type)ge from ROS 1  �P`
or
    `~/ros2_ws$ ros2 run ros1_bridge dynamic_bridge 
    created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/Log' and ROS 1 type 'rosgraph_msgs/Log'
    created 1to2 bridge for topic '/chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/String'
    [INFO] [ros_bridge]: Passing message from ROS 1  �C to ROS 2  �C (showing msg only once per type)`

mlanting, 2019-04-03:
My co-worker helped identify the issue - when the logging macros are called in factory.hpp, the message type name variables passed in need to have .data() added, otherwise it just tries to print the address.

Is the appropriate fix to just add the .data() to string types passed into the macro, or should the logging macros be modified to accept std::string types without the need for the data() function to be called?

mlanting, 2019-04-03:
I've tested the patch and I'm getting the correct output now.

paulbovbel, 2019-04-03:
Thank you for the fix @dirk-thomas

PR #182 (2019-04-04): fix log messages
Fixes #181 which is a regression of #167.

[image]
closed 2019-04-04 by dirk-thomas