ros2/ros1_bridge #187: use .idl files for the ROS 2 interfaces

dirk-thomas, 2019-04-18:
Otherwise interfaces with `.idl` extensions won't be picked up by the bridge.

ksuszka, 2020-11-04:
Isn't enough to change  `${${package_name}_INTERFACE_FILES}` in line https://github.com/ros2/ros1_bridge/blob/596eb40ff81bc0bed4badcd5dc301fe47955a4d5/CMakeLists.txt#L102 to `${${package_name}_IDL_FILES}` ? If I'm not mistaken, rosidl generates idl files for msg anyway, so there should be no change for msg files, but interfaces with only idl files can be supported.

At the moment we have this issue, that we cannot compile ros1_bridge if ros-dashing-autoware-auto-msgs package is installed as in the latest version it has only idl files and current ros1_bridge adds those messages to factory files but doesn't compile them so there are linker errors.

clalancette, 2020-11-18:
@wjwwood Is this solved by https://github.com/ros2/ros1_bridge/pull/296 ?

wjwwood, 2020-11-19:
> @wjwwood Is this solved by #296 ?

Yup!

@ksuszka it's quite a big more involved than that, since the object in Python you get from processing an msg file directly (which is what the `ros1_bridge` was doing) is different from the object you get from processing the idl file instead.

closed 2020-11-24 by wjwwood