ros2/ros1_bridge #195: realloc(): invalid pointer causing SIGABRT after recent updates

cottsay, 2019-05-17:
After pulling `ros1_bridge` changes this morning on Dashing, I'm seeing:
```
[INFO] [ros_bridge]: Passing message from ROS 1 geometry_msgs/Twist to ROS 2 geometry_msgs/msg/Twist (showing msg only once per type)
realloc(): invalid pointer
```
It's not always the same message type, but it always happens after only a few seconds bridging about 10 topics. Here is a backtrace:
```
#0  0x00007ffff012657f in raise () at /lib64/libc.so.6
#1  0x00007ffff0110895 in abort () at /lib64/libc.so.6
#2  0x00007ffff01699c7 in __libc_message () at /lib64/libc.so.6
#3  0x00007ffff01702cc in  () at /lib64/libc.so.6
#4  0x00007ffff01755ba in realloc () at /lib64/libc.so.6
#5  0x00007ffff6df9d31 in rosidl_generator_c__String__assignn () at /opt/ros/dashing/lib/librosidl_generator_c.so
#6  0x00007ffff6df9dc8 in rosidl_generator_c__String__assign () at /opt/ros/dashing/lib/librosidl_generator_c.so
#7  0x00007ffff74ee42f in _Log__cdr_deserialize () at /opt/ros/dashing/lib/librcl_interfaces__rosidl_typesupport_fastrtps_c.so
#8  0x00007fffd90a03aa in rmw_fastrtps_cpp::TypeSupport::deserializeROSmessage(eprosima::fastcdr::Cdr&, void*) () at /opt/ros/dashing/lib/librmw_fastrtps_cpp.so
#9  0x00007fffd9026ad4 in rmw_fastrtps_shared_cpp::TypeSupport::deserialize(eprosima::fastrtps::rtps::SerializedPayload_t*, void*) ()
    at /opt/ros/dashing/lib/librmw_fastrtps_shared_cpp.so
#10 0x00007fffefb2c58c in eprosima::fastrtps::SubscriberHistory::takeNextData(void*, eprosima::fastrtps::SampleInfo_t*) () at /opt/ros/dashing/lib/libfastrtps.so.1
#11 0x00007fffd901e9ad in rmw_fastrtps_shared_cpp::_take(char const*, rmw_subscription_t const*, void*, bool*, rmw_message_info_t*, rmw_subscription_allocation_t*) ()
    at /opt/ros/dashing/lib/librmw_fastrtps_shared_cpp.so
#12 0x00007fffd901ee0a in rmw_fastrtps_shared_cpp::__rmw_take_with_info(char const*, rmw_subscription_t const*, void*, bool*, rmw_message_info_t*, rmw_subscription_allocation_t*) () at /opt/ros/dashing/lib/librmw_fastrtps_shared_cpp.so
#13 0x00007fffd909ff6f in rmw_take_with_info () at /opt/ros/dashing/lib/librmw_fastrtps_cpp.so
#14 0x00007ffff79d2c8e in rmw_take_with_info () at /opt/ros/dashing/lib/librmw_implementation.so
#15 0x00007ffff7a586a6 in rcl_take () at /opt/ros/dashing/lib/librcl.so
#16 0x00007ffff7e0272b in rclcpp::executor::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>) () at /opt/ros/dashing/lib/librclcpp.so
#17 0x00007ffff7e02160 in rclcpp::executor::Executor::execute_any_executable(rclcpp::executor::AnyExecutable&) () at /opt/ros/dashing/lib/librclcpp.so
#18 0x00007ffff7e01ec7 in rclcpp::executor::Executor::spin_once(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () at /opt/ros/dashing/lib/librclcpp.so
#19 0x00007ffff7e01933 in rclcpp::executor::Executor::spin_node_once_nanoseconds(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () at /opt/ros/dashing/lib/librclcpp.so
#20 0x000000000043856a in void rclcpp::executor::Executor::spin_node_once<rclcpp::Node, long, std::ratio<1l, 1000l> >(std::shared_ptr<rclcpp::Node>, std::chrono::duration<long, std::ratio<1l, 1000l> >) ()
#21 0x000000000043112c in main ()
```

dirk-thomas, 2019-05-17:
Can you please add debug information for the backtrace, e.g. what the parameters the `rosidl_generator_c__String__assign` / `rosidl_generator_c__String__assignn` calls have.

hidmic, 2019-05-17:
This is the same issue I'm hunting down in https://github.com/ros2/build_cop/issues/192.

dirk-thomas, 2019-05-17:
Closing this duplicate.

closed 2019-05-17 by dirk-thomas

khssnv, 2019-10-18:
Launching [example](https://github.com/ros2/ros1_bridge#example-1a-ros-1-talker-and-ros-2-listener) from readme I experiencing the same with ros1_bridge built from [latest commit](https://github.com/ros2/ros1_bridge/tree/e8568f19c85c70fe677a0470c4281e411f338311) in master branch.

dirk-thomas, 2019-10-18:
CI is currently passing for the master branch and I just tried to reproduce it locally and it was working for me:

```
created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
created 1to2 bridge for topic '/chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/msg/String'
[INFO] [ros_bridge]: Passing message from ROS 1 std_msgs/String to ROS 2 std_msgs/msg/String (showing msg only once per type)
[INFO] [ros_bridge]: Passing message from ROS 2 rcl_interfaces/msg/Log to ROS 1 rosgraph_msgs/Log (showing msg only once per type)
removed 1to2 bridge for topic '/chatter'
removed 2to1 bridge for topic '/rosout'
```

So it is likely something specific to your environment / steps. If you would like help troubleshooting what it is please include more information: on what platforms, what version of ROS 1 / 2 you are using, which binaries / which commit hashed if you are building from source, the exact steps you followed, etc.

khssnv, 2019-10-21:
### Platform
Ubuntu 18.04.3 LTS bionic, zsh shell
### ROS1 distro
Melodic
### ROS2 distro
Dashing
### ros1_bridge
Commit [e8568f19c85c70fe677a0470c4281e411f338311](https://github.com/ros2/ros1_bridge/tree/e8568f19c85c70fe677a0470c4281e411f338311)
### Building
Built as it guided by [README](https://github.com/ros2/ros1_bridge#building-the-bridge-from-source).
### Failure
Example failed to run: [Example 1a: ROS 1 talker and ROS 2 listener](https://github.com/ros2/ros1_bridge#example-1a-ros-1-talker-and-ros-2-listener)

```console
╭─khassanov@32vd ~/Workspace/ros2_ws
╰─$ ros2 run ros1_bridge dynamic_bridge
created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
created 1to2 bridge for topic '/chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/msg/String'
[INFO] [ros_bridge]: Passing message from ROS 1 std_msgs/String to ROS 2 std_msgs/msg/String (showing msg only once per type)
realloc(): invalid pointer
```

### More details
Please find platform data source and ros1_bridge backtrace below.

<details>

```console
╭─khassanov@32vd ~
╰─$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.3 LTS"
╭─khassanov@32vd ~
╰─$ . /opt/ros/melodic/setup.zsh
╭─khassanov@32vd ~
╰─$ env | grep ROS
ROS_DISTRO=melodic
ROS_ETC_DIR=/opt/ros/melodic/etc/ros
ROS_PACKAGE_PATH=/opt/ros/melodic/share
ROS_PYTHON_VERSION=2
ROS_VERSION=1
ROS_ROOT=/opt/ros/melodic/share/ros
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
╭─khassanov@32vd ~
╰─$ . /opt/ros/dashing/setup.zsh
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
╭─khassanov@32vd ~
╰─$ env | grep ROS
ROS_DISTRO=dashing
ROS_ETC_DIR=/opt/ros/melodic/etc/ros
ROS_PACKAGE_PATH=/opt/ros/melodic/share
ROS_PYTHON_VERSION=3
ROS_VERSION=2
ROS_ROOT=/opt/ros/melodic/share/ros
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
```
</details>

dirk-thomas, 2019-10-22:
If you are using the Debian packages of Dashing why are you building the bridge from source? I assume you want support for additional custom messages.

> `ros1_bridge` Commit e8568f19c85c70fe677a0470c4281e411f338311

The commit is from the `master` branch which is targeting Eloquent - not Dashing. You should use the latest version from the `dashing` branch instead (atm 46baa889bcb9808b5b461f98faee520a90d57347).