ros2/ros1_bridge #205: ros1_bridge dynamic_bridge missing libroscpp.so

dario-pedro, 2019-06-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 running docker ROS2 nightly
- Installation type:
  - docker ROS2 nightly

When I run `docker ROS2 nightly` I get /opt/ros/dashing/lib/ros1_bridge/dynamic_bridge: error while loading shared libraries: libroscpp.so: cannot open shared object file: No such file or directory

Any idea how to solve it ?

dirk-thomas, 2019-07-15:
Have you sourced ROS 1?

kk2491, 2019-07-17:
@dirk-thomas @dario-pedro I was getting the same error. So I sourced both ```dashing``` and ```melodic```, now I am getting the below error. Could you please let me know what the issue is.
```
system@d1dbb9f4c6b7:/module$ ros2 run ros1_bridge dynamic_bridge 
Failed to load entry point 'info': /opt/ros/dashing/lib/librosbag2_storage.so: undefined symbol: _ZNK12class_loader11ClassLoader15isLibraryLoadedEv
The C extension '/opt/ros/dashing/lib/python3.6/site-packages/rosbag2_transport/_rosbag2_transport_py.cpython-36m-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Segmentation fault
systeml@d1dbb9f4c6b7:/module$ 
```

dirk-thomas, 2019-07-17:
> I sourced both `dashing` and `melodic`

You need to source them in the other order - ROS 1 first, then ROS 2 - as described in the instructions.

dirk-thomas, 2019-07-27:
Assuming the previous comments have resolved the issue I will go ahead an close this ticket. Please feel free to continue commenting and if necessary the ticket can be reopened.

closed 2019-07-27 by dirk-thomas

potentialdiffer, 2020-09-24:
I am running into a similar problem. (Using Debian 9, Kinetic and Dashing from source) What I've done so far: 

- rebuild ROS1 and ROS2 workspace in "clean" terminal
- compiled `ros1_bridge` with first sourcing ROS1 and then ROS2 workspace 

Building `ros1_bridge` succeeds, however running the bridge results in the error, which was previously mentioned:

```
$ source /home/xxx/ros1_kinetic/install/setup.bash
$ source /home/xxx/ros2_dashing/install/setup.bash
$ ros2 run ros1_bridge dynamic_bridge                                                        
/home/xxx/ros2_dashing/install/ros1_bridge/lib/ros1_bridge/dynamic_bridge: symbol lookup error: /home/xxx/ros1_kinetic/install/navfn/lib/libnavfn.so: undefined symbol: _ZN12class_loader20class_loader_private30getCurrentlyLoadingLibraryNameB5cxx11Ev
```

I checked `libnavfn.so`:

```
$ ldd /home/xxx/ros1_kinetic/install/navfn/lib/libnavfn.so | grep class_loader
libclass_loader.so => /home/xxx/ros2_dashing/install/class_loader/lib/libclass_loader.so (0x00007f538c5cf000)
```

The linker uses the ROS2 libraries for ROS1, which is obviously wrong. Actually, a lot more libraries get linked wrong, the `class_loader` is just the first, that causes the error. If only ROS1 workspace is sourced, the linker uses the right libraries:

```
$ source /home/xxx/ros1_kinetic/install/setup.bash
$ ldd /home/xxx/ros1_kinetic/install/navfn/lib/libnavfn.so | grep class_loader
libclass_loader.so => /home/xxx/ros1_kinetic/install/class_loader/lib/libclass_loader.so (0x00007f60fe015000)
```

I have no further idea how to come around that problem, as I've already made a clean compilation of both workspaces, with double checking, that there was a clean environment. 

I appreciate any help or idea. Thanks in advance.

shlokgoel, 2021-01-07:
I am getting a similar error. 

I have sourced ROS Melodic first and then ROS Dashing inside the docker. I have followed instructions as provided in the ros2 GitHub wiki. 

[image]

I would appreciate any help or idea. Thank you!

jackxia123, 2022-03-11:
> I am getting a similar error.
> 
> I have sourced ROS Melodic first and then ROS Dashing inside the docker. I have followed instructions as provided in the ros2 GitHub wiki.
> 
> [image]
> 
> I would appreciate any help or idea. Thank you!

## set ros1 environment variables
source "/opt/ros/${ROS2_DISTRO}/setup.bash"
## set ros2 environment variables
source "/opt/ros2-lgsvl-bridge/install/setup.bash"
source "${LANE_FOLLOWING_ROOT_DIR}/ros2_ws/install/local_setup.bash"

only set ros2 environment topic , not set ros1

ruoxi521, 2022-07-03:
@ [jackxia123](https://github.com/jackxia123)  
I had the same error at the beginning, when I looked at the project's[ Readme](https://github.com/ros2/ros1_bridge/blob/dashing/README.md) file, I saw these, so I tried it and the error was resolved.
[image]

```
# Shell C:
. /opt/ros/melodic/setup.bash
# Or, on OSX, something like:
# . ~/ros_catkin_ws/install_isolated/setup.bash
rosrun rospy_tutorials talker
```

My operation is similar to yours, but it works fine.
[image]

After comparison, I feel that the problem seems to be the lack of the following command line
```
. <install-space-with-bridge>/setup.bash
```
I'm not sure it will work, but I think I can give it a try, hope it helps

ruoxi521, 2022-07-03:
[image]

The specific location is here in the readme file