ros2/ros1_bridge #206: Limitations of bridged messages

TheLaas, 2019-07-03:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Docker: Ubuntu 18.04, Ros2 Dashing, Ros melodic
- Docker: Ubuntu 14.04, Ros Indigo
- Installation type:
  - Ros2 dashing, binaries, ros:dashing-ros-base-bionic as base image
  - Ros melodic, binares, ros-melodic-ros-base=1.4.1-*
  - Ros Indigo, osrf/ros:indigo-desktop-full-trusty as base image.
- Version or commit hash:
  - 
- DDS implementation:
  - Default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

1. Replay data in ros indigo in container 1, sends out topic at rate ~100Hz
2. Start up the bridge in container 2 (`ros2 run ros1_bridge dynamic_bridge`)
3. Start subscriber to `nav_msgs/Odemetry` in container 2

#### Expected behavior
Read and print all published odometry messages in subscriber

#### Actual behavior
The messages gets through the bridge but they are received in batches. Get around 10 msgs consecutive then it takes around 20seconds until the next batch arrives.

```
---
header:
  stamp:
    sec: 1562149700 <-----------
    nanosec: 454283300
  frame_id: odom
child_frame_id: base
pose:
  pose:
    position:
      x: 9.417769857764657
      y: -17.242610332467184
      z: 0.0
    orientation:
      x: -0.0
      y: 0.0
      z: 0.3598343613819179
      w: -0.9330162015575438
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist:
  twist:
    linear:
      x: 4.697222222222222
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.009773843811168284
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header:
  stamp:
    sec: 1562149722 <-------------
    nanosec: 843339600
  frame_id: odom
child_frame_id: base
pose:
  pose:
    position:
      x: 31.292122668154306
      y: -23.24881794357751
      z: 0.0
    orientation:
      x: 0.0
      y: -0.0
      z: -0.9982219034870322
      w: -0.05960730994371486
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist:
  twist:
    linear:
      x: 1.238888888888889
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.018291050560900607
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

#### Additional information
I've checked the system the cpu load and it's around 30%.
I've echoed the messages at the ros1 side of the bridge container and all messages seem to arrive there.  

## 
I'm not sure if this is a docker problem or a bridge problem, but since the bridge container can print all the ros1 messages I'm thinking that it might be some limitation to the bridge? I've also tried to run the same setup but on a linux-computer. It gets better there but it still arrive in batches where it jumps 2 seconds instead of 20.

Should this be expected? I need to bridge more topics but if it's this slow with just one topic I'm not sure how it's going to be if I add more. How much load can the bridge be expected to handle?

TheLaas, 2019-07-04:
I moved over to using `static_bridge` instead. Now performance is as expected.

closed 2019-07-04 by TheLaas

yeo-menghan, 2023-06-09:
Have you tried multiple topics on static bridge? i.e. bridging all topics from ros1 to ros2 via static bridge? Is that even possible