ros2/ros1_bridge #210: Problem with parameter_server terminate called after throwing an instance of 'XmlRpc::XmlRpcException'

zxc455233, 2019-08-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04, ROS2 Dashing
- Installation type:
  - from source
- Version or commit hash:
  - 
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
This is my ros1 launch file to set topics for parameter_bridge
```xml
<launch>
<rosparam param="/topics">["/robot2/strategy/state", "/robot3/strategy/state"]</rosparam>
</launch>
```
And there is the output messages of ```rosparam get /topics```
```bash
- /robot2/strategy/state
- /robot3/strategy/state
```

But, there is unexpected problem when I try to run parameter_bridge
```bash
$ ros2 run ros1_bridge parameter_bridge
terminate called after throwing an instance of 'XmlRpc::XmlRpcException'
```

zxc455233, 2019-08-04:
Sorry, I noticed that I need to set "type", "queue_size", and "topic" in parameter server.
After I set up my ros parameter server, It works good.

closed 2019-08-04 by zxc455233

schornakj, 2019-10-10:
Just wanted to provide a more complete working example, since I was stuck on the XmlRpc formatting for a bit:

```
<launch>
<rosparam param="/topics">[{"topic": "/joint_states", "type": "sensor_msgs/msg/JointState", "queue_size": 10}]</rosparam>
</launch>
```

Note that `type` has to be the ROS2 type of the topic to be bridged.