ros2/ros1_bridge #218: Message frequency lower than expected.

Oofs, 2019-09-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18
- Installation type:
  -  binaries
- Version or commit hash:
  - dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Follow the steps in turotial, start from **Playing back topic data with rosbag and ROS 1 Bridge**
https://index.ros.org//doc/ros2/Tutorials/Rosbag-with-ROS1-Bridge/
#### Expected behavior
When I played .bag(ros1 bag) containing the PointCloud2 msgs, I expect to get them in ros2 with the same frequency as  which in ros1 bag.
#### Actual behavior
The frequency of PointCloud2 is about 9HZ, which is lower than original frequency (10 HZ).
When I do these steps on Xavier Jetson, the frequency is even lower (about 5HZ).

EgalYue, 2019-09-29:
Hi,
I also met the same problem. So i made a simple test without ros1_bridge:

**Running Env:**
* Ubuntu 18.04 in one Docker Container on PC and Ubuntu 18.04 on Xaviar
* ROS version: ROS1 melodic, ROS2 Dashing  
* PC and Xavier under same Network situation

**Testing process**
Create a simple node only used to publish different size of message, and observe the topic frequency in ROS1 and ROS2. The frequency of publishing the message is 10 Hz.

**Code**

ROS1.0

```
#include "ros/ros.h"
#include "std_msgs/String.h"
#include <sstream>
#include <fstream>
#include <string>

std::string read_datafile(std::string message_filename){  
  std::ifstream ifs(message_filename.c_str());
  if(ifs.fail()) {
	std::cerr << "data_*byte.txt do not exist.\n";
	exit(0);
  }

  std::string bytedata;
  getline(ifs, bytedata);

  return bytedata;
}

int main(int argc, char **argv)
{
  ros::init(argc, argv, "talker");

  ros::NodeHandle n;

  ros::Publisher chatter_pub = n.advertise<std_msgs::String>("chatter", 1000);

  ros::Rate loop_rate(10);

  int count = 0;

  std::string bytedata = read_datafile("/home/algo/TestCode/ros1_simple/src/simple_test_talker/src/byte_data/data_1Mbyte.txt");
  while (ros::ok())
  {
    std_msgs::String msg;
    std::stringstream ss;
    ss << bytedata << count;
    msg.data = ss.str();

    ROS_INFO("%s", msg.data.c_str());

    chatter_pub.publish(msg);
    ros::spinOnce();
    loop_rate.sleep();
    ++count;
  }

  return 0;
}
```

```
rostopic hz chatter
rostopic bw chatter
```

ROS 2.0 with the same idea, ignore code.

**Results**
In Docker Container on PC:
[image]

On Xavier:
[image]

**Question:**
1. ROS 2.0 with different DDS(fastetps, opensplice) has a lower bandwidth than ROS 1.0 ? (I know that, FastRTPS does not support large data because it is designed as a lightweight implementation for embedded systems)
2. Why the frequency and bandwidth have so big difference on PC and Xavier (both ROS1.0 and ROS2.0), even though under same network situation?

dirk-thomas, 2019-10-03:
Please double check the CPU usage and share if this is due to being CPU bound.

EgalYue, 2019-10-08:
Hi,
I double check the CPU usage, **I don't think, this is due to being CPU bound.**
Following is CPU usage results(I make the same test in docker container on PC and also on Xavier):

[image]
 

And here I post same screenshots of results as example:
On Xavier:
ROS 1.0, Message size 2M
[image]

On Xavier:
ROS 2.0, fastrtps, Message size 2M
[image]

On Xavier:
ROS 2.0, opensplice, Message size 2M
[image]

 In addition, I upload my test code:
https://github.com/EgalYue/ros_simple_publisher

qinyusen, 2019-10-09:
@dirk-thomas  
hi , same here, test between a pc and a xavier, and  test is under 1000mbps link networks env. , network test by iperf3,  and hardware situation using ethtool, and it shows the ethernet worked under 1000m full mode.

CPU load is under 15% on both side. SSD IO / memory IO  / network IO (file buffer etc.) have set to  unlimited, and should be good for the link.

Similar situation as description above persons, the bandwidth is block on 10MBps(i.e. 80Mbps), which is super wired.

what can I do , I would like to help to debug this ?

EgalYue, 2019-10-14:
@dirk-thomas
Hi, I make futher testing about this problem. I found that, if the communication is **between 2 Xavers or 2 PCs** under 1000mbps network env. , there is no more problem, the frequency is 10Hz as is expected and the bandwidth of topic can be 100MB/s. 
Ex:
On Xavier 01 / PC 01, run:
```
export ROS_DOMAIN_ID=23
source /opt/ros/dashing/setup.bash
ros2 run simple_test_talker simple_talker
```

On Xavier 02 / PC 02, run:
```
export ROS_DOMAIN_ID=23
source /opt/ros/dashing/setup.bash
ros2 topic hz /topic
```
**In this case, communication between 2 Xaviers or 2 PCs is no problem!**

**This problem occurs only  by localhost!**
Ex:
Using only one Xavier/PC,
 open terminal 1 run:
```
source /opt/ros/dashing/setup.bash
ros2 run simple_test_talker simple_talker
```
open terminal 2 run:
```
source /opt/ros/dashing/setup.bash
ros2 topic hz /topic
```
**In this case, the frequency of topic is lower than expected!**, which should be 10hz.

  | Xavier
-- | --
Message Size | frequency(Hz) | bandwidth(MB/s)
1M | 5.7 | 6.5
2M | 1.5 | 5-6
4M | 4 - 5 | 20
10M | 7.5 | 65 - 80

  | PC
-- | --
Message Size | frequency(Hz) | bandwidth(MB/s)
1M | 9.5 - 9.7 | 9.6 – 9.9
2M | 9.3 – 9.6 | 18 - 19
4M | 9.2 – 9.3 | 38
10M | 9.0 | 90

**So, I am confused about this. Is the communication on localhost still via port (TCP or UDP)?** 

**This problem is even same by using opensplice.**

Following are the versions, which i used:

SW | version
-- | --
ROS 2.0 | Dashing,  Installing via Debian Packages
  |  
fastrtps | 1.8.0
rmw_fastrtps_cpp | 0.7.5
rmw | 0.7.2
rmw_implementation | 0.7.1
rcl | 0.7.7
rclcpp | 0.7.10

In addition, I upload my test code:
https://github.com/EgalYue/ros_simple_publisher

dirk-thomas, 2020-01-15:
> Is the communication on localhost still via port (TCP or UDP)?

If the RMW impl. doesn't provide a special optimization (which neither FastRTPS nor OpenSplice does in the versions you used) then yes, the communication is still happening over UDP.

I can only suggest to try reproducing your problem with the latest release Eloquent.

Since there is currently not enough information for us to do anything specific about this ticket I will go ahead and close it. Please feel free to continue commenting and if the problem can be reproduced with the latest release the ticket can be reopened.

closed 2020-01-15 by dirk-thomas

grischi, 2021-03-26:
I'm suffering from this issue on ros2 foxy. When I play a ROS1 rosbag, rostopic hz outputs a certain message frequency which is constant at least up to one percent (ROS1). Running ros2 topic hz for the bridged topic outputs a message frequency which is lower in average by 5-50% (ROS2). Tested with some frequencies between 1 and 400 Hz by varying the playback rate of the rosbag. I'm running all of this on the same machine and CPU load is rather low.

gilad-lif, 2022-06-01:
have anyone solved this problem?

grischi, 2022-09-29:
I can just report that at least since ...

Operating System:
Ubuntu 20
Installation type:
binaries
Version or commit hash:
galactic
DDS implementation:
cyclonedds (https://docs.ros.org/en/rolling/How-To-Guides/DDS-tuning.html)

... I am not experiencing this issue any more.

IdanAviv89, 2023-05-06:
Hi,
have you managed to solve this?
I am facing the same issue on ros galactic version (Ubuntu 20). I use the python rosbag tool to convert the euroc mav dataset from ros1 to ros2. And while in ros1 I get that the imu topic is at 200Hz, at ros2 I get that the imu topic is at 120Hz.

gilad-lif, 2023-05-06:
> Hi, have you managed to solve this? I am facing the same issue on ros galactic version (Ubuntu 20). I use the python rosbag tool to convert the euroc mav dataset from ros1 to ros2. And while in ros1 I get that the imu topic is at 200Hz, at ros2 I get that the imu topic is at 120Hz.

Hello idan, i spent a while on the problem without success, and eventually solved the original problem without using the bridge. Send me an email, i will be happy to share my (small) knowledge, gilad.lifshitz.perfetto@gmail.com.