ros2/ros1_bridge #220: Ignoring errored service requests

machinekoder, 2019-10-10:
##  Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Build from source 
- Version or commit hash:
  46baa889bcb9808b5b461f98faee520a90d57347
- DDS implementation:
  FastRTPS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
* Create service in ROS1
* Create client in ROS2
* Call service in client producing an unhandled exception in the ROS1 service callback

#### Expected behavior
* `ros1_bridge` displays the error message but doesn't crash/stop.

#### Actual behavior
* `ros1_bridge` stops/crashes displaying the error message

#### Additional information
If this shouldn't be the default behavior for some reason, a command-line argument e.g. `--ignore-errors` would be great.

jacobperron, 2019-10-10:
Can you provide a [SSCCE](http://sscce.org/) for reproducing the error? Thanks

jacobperron, 2019-10-24:
@machinekoder Friendly ping. If you could provide reproducible example of a service and client that causes the error that would be great.

wjwwood, 2019-11-14:
We'd appreciate help implementing a fix/enhancement for this. For now I'm putting this in the backlog as we don't have time to address it.

dirk-thomas, 2019-11-14:
Instead of moving the ticket to the `backlog` this ticket needs `more information`.

dirk-thomas, 2020-01-31:
@machinekoder Friendly ping.

dirk-thomas, 2020-05-19:
Reproducible example - run the following:
* `roscore`
* `dynamic_bridge`
* ROS 1 `rospy_tutorials` `add_to_ints_server` with the server callback modified to raise an exception
* ROS 2 `demo_nodes_py` `add_to_ints_client`

Atm the `rclcpp` / `rcl` / `rm` API must return a response from a service call and there is no way to communicate a problem. Until that is resolved the bridge can't do anything but error in this case (or never return the service call which isn't a good option either).