ros2/ros1_bridge #223: ROS1_bridge error : tf/TFMessage'": No template specialization for the pair

amranov, 2019-10-21:
## Bug report

**Required Info:**

- Operating System:
ubuntu 18.04 
- Installation type:
source
- Version or commit hash:
  - 
- DDS implementation:
RTI Connext
- Client library (if applicable):
rclcpp, rclpy

#### Steps to reproduce issue
Hi Guys, We are trying to use ROS1_bridge to control base_robot first we souced ros1 and launch the driver.launch second shell source the ros2 (dashing) and run the NAV2 to bring our robot third shell we run the ros1_bridge (installed via debian package) ;

We get this error from ros1_bridge  
```

> failed to create 2o1 bridge for topic '/tf' with ROS 2 type 'tf2_msgs/msg/TFMessage' and ROS 1 type 'tf/tfMessage': No template specialization for the pair

```
Any suggestions?
#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

gavanderhoorn, 2019-10-21:
This would appear to be a cross-post of [ROS1_bridge error : tf/TFMessage'": No template specialization for the pair](https://answers.ros.org/question/335793) on ROS Answers.

dirk-thomas, 2019-10-22:
Please include which exact version you are building from source. If you have new enough versions of the `ros1_bridge` repo as well as the `geometry2` repo the bridging should work.

The fact that the pair isn't shown with `--print-pairs` is related to the fact that there are two mappings from the ROS 2 message `tf2_msgs/msg/TFMessage` to different ROS 1 types. #225 addressed showing both of them in `--print-pairs`.

dirk-thomas, 2020-01-09:
Closing due to no response.

closed 2020-01-09 by dirk-thomas