ros2/ros1_bridge #226: Feature Request: Support moveit_msgs

tobi-v, 2019-11-15:
## Feature request

#### Feature description

Hello there! Is it possible to include the moveit_msgs as a supported interface in the next release of ros1_bridge?
Since MoveIt is used by a lot of other packages, but moveit2 isn't ready yet, this would be a great possibility to use the functionalities without having to build everything from source.

dirk-thomas, 2019-11-15:
Until the maintainers of the `moveit_msgs` package release it into a ROS 2 distribution there is nothing to be done here. Please ask your question in the `moveit_msgs` repository instead.

closed 2019-11-15 by dirk-thomas

mjcarroll, 2023-03-22:
Probably better to ping current maintainers @methylDragon and @quarkytale