ros2/ros1_bridge #227: Building ROS2 from source necessary for custom messages?

mvoellmy, 2019-11-26:
Hi,

Quick question:
In the README it mentions, that the bridge only works with messages defined at compile time of the bridge.

Also it says:

> Building the bridge from source
> Before continuing you should have the prerequisites for building ROS 2 from source installed following these instructions.
> 

It is clear that the bridge has to be built from source for custom messages to work. Is it necessary that ROS2 is built from source as well or is this a typo?

Thanks,

Miro

clalancette, 2019-11-26:
> It is clear that the bridge has to be built from source for custom messages to work. Is it necessary that ROS2 is built from source as well or is this a typo?

Neither, really.  The sentence says (emphasis mine):

> Before continuing you should have the _prerequisites for building ROS 2 from source_ installed following these instructions.

I guess that could be clarified with a comma or emphasis (pull requests to do that welcome).

In any case, it means that you should setup as if you are going to build from source (to make sure you have all of the prerequisites, and then you just need to build the bridge from source.

I'm going to close this out, but as I mentioned do feel free to open a pull request to improve the documentation.  Thanks.

closed 2019-11-26 by clalancette

mvoellmy, 2019-11-26:
Alright! Thanks for the prompt response!