ros2/ros1_bridge #240: parameter_bridge breaks when converting pointcloud msgs

tanyouliang95, 2020-02-07:
## Bug report

### Environment
Tested on Ubuntu 18

Image pulled from osrf docker hub
```bash
docker pull osrf/ros:eloquent-ros1-bridge
docker run -it --net=host osrf/ros:eloquent-ros1-bridgedocker run -it --net=host osrf/ros:eloquent-ros1-bridge
> ros2 run ros1_bridge parameter_bridge
```
- Default DDS implementation

### Application
I was using the `ros1_bridge` to convert ros1 `sensor_msgs/PointCloud2` to `sensor_msgs/msg/PointCloud2` data. For my application, I was publishing high resolution 3D pointcloud data at around 5hz (specifically from a Kinect sensor). It works well, however often the `ros1_bridge` node will break after a short period of time.

### Error Msgs

These were the error msgs that i got in multiple instances of running the `rosbridge`
```bash
realloc(): invalid next size
malloc(): memory corruption
corrupted size vs. prev_size
realloc(): invalid pointer
double free or corruption (out)
```

Previously, I was using `ros1_bridge` for multiple applications with smaller data packets, and Iam glad that `ros1_bridge` was always able to work reliably. This current issue might be due to the larger packet size. I am also not sure whether is it due to the deficiency of the  default dds layer (Fastrtps). Also, will the `queue_size` in the yaml file (for the parameter_bridge) actually matters? 

Hope to get help on this issue. Thanks alot!

dirk-thomas, 2020-02-07:
Please provide reproducible steps including the involved ROS 1 and ROS 2 nodes.

tanyouliang95, 2020-02-07:
### Dependencies
```
docker pull osrf/ros:eloquent-ros1-bridge
```
The sample .yaml and .bag files are here: 
https://drive.google.com/drive/folders/1rN3PfcUrlBC-nwI1W5qxb-9IWJjQy_Nw?usp=sharing

### Run the code
- *Terminal 1*: `roscore`

- *Terminal 2*: Run sample rosbag
```bash
rosbag play kinect.bag -l 
```

- *Terminal 3*: Run parameter bridge
```bash
rosparam load rosbridge_topics.yaml
docker run -it --net=host osrf/ros:eloquent-ros1-bridge
> ros2 run ros1_bridge parameter_bridge
```

- *Terminal 4*: Check if pointcloud publishing in ros2 
```bash
docker ps
# copy the container id
docker exec -it <CONTAINER_ID>  /bin/bash
> source /opt/ros/eloquent/setup.bash
> ros2 topic echo /kinect31/qhd/points
```

Time after time, the parameter_bridge on terminal 3 will break.

ivanpauno, 2020-02-20:
@tanyouliang95 Can you add some example tracebacks of those errors?

dirk-thomas, 2020-03-03:
There has been a patch to Fast-RTPS recently which addressed a memory problem. The current Docker image contains `ros-eloquent-fastrtps` version 1.9.3-1. Can you please try to update the packages in the container (which should bump to version1.9.3-2) and double check if the issue persists or has been resolved?

dirk-thomas, 2020-05-06:
Closing due to no further reply. Please feel free to comment with the requested information and if necessary the ticket can be reopened.

closed 2020-05-06 by dirk-thomas

zentala, 2020-06-03:
I have this mentioned fixed version and it's crashing for me too.
```
$ apt-cache policy ros-eloquent-fastrtps
ros-eloquent-fastrtps:
  Installed: 1.9.3-2bionic.20200218.174738
  Candidate: 1.9.3-2bionic.20200218.174738
  Version table:
 *** 1.9.3-2bionic.20200218.174738 500
        500 http://packages.ros.org/ros2/ubuntu bionic/main amd64 Packages
        100 /var/lib/dpkg/status
$
```
In my case, I am bridging PCL from ROS Melodic to ROS Eloquent. When I am starting `ros2_object_analitycs` I got this:
```
$ /opt/robot_devkit/perception/object_analytics_node/lib/object_analytics_node/object_analytics_node --tracking --localization --ros-args -r /object_analytics/detected_objects:=/ros2_openvino_toolkit/detected_objects -r /object_analytics/rgb:=/camera/color/image_raw -r /object_analytics/pointcloud:=/camera/depth/color/points       
[INFO] [OA_Composition]: Load library libsegmenter_component.so
[INFO] [OA_Composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<object_analytics_node::segmenter::SegmenterNode>
[INFO] [OA_Composition]: Load library libtracking_component.so
[INFO] [OA_Composition]: Instantiate class rclcpp_components::NodeFactoryTemplate<object_analytics_node::tracker::TrackingNode>
[WARN] []: No objects to publish!
[1]    14536 segmentation fault (core dumped)   --tracking --localization --ros-args -r  -r  -r 
```
But when I am using realsense wrapper for ROS Eloquent this issue doesn't occur.

dirk-thomas, 2020-06-04:
> When I am starting ros2_object_analitycs I got this:

@zentala The segmentation fauly you posted is not happening in the `ros1_bridge` but in `object_analytics_node` which is not part of this repo. You might want to report the problem for that node in the repository where it is coming from.